springfall2008/batpred #935: Midnight slot not selected for car or iboost charging, despite being cheaper

mpartington, 2024-04-10:
**Describe the bug**
As discussed on FB, it selects the 23.30 slot in preference to the cheaper one at 00.00.  If you force idle the 23.30 slot, then it does select it.   Also tried changing the max charge unit cost to -1p, no effect.

**Expected behavior**
Choose the cheapest slot for car charging

**Predbat version**
v7.16.12

**Screenshots**
[image]

with slot forced idle
[image]

settings
[image]

[image]

**Log file**
Period of interest is around 9pm, where we were running different scenarios

mpartington, 2024-04-10:
Just added the requirement for another charge slot and it does select it, but still has a preference for the earlier cheaper slot
[image]

mpartington, 2024-04-11:
Similar tonight, but 23.30 slot this time. Have upped the daily target, but no change 

[image]

springfall2008, 2024-04-13:
Fix now released

closed 2024-04-13 by springfall2008