springfall2008/batpred #936: iBoost Forecast Using Current & Next Gas Rates

nbullus, 2024-04-11:
**Describe the bug**
The Octopus Gas Tracker tariff utilises two cost sensors:

sensor.octopus_energy_gas_XXXXXXXXXXXX_XXXXXXXXXX_current_rate
sensor.octopus_energy_gas_XXXXXXXXXXXX_XXXXXXXXXX_next_rate

But only the current rate is defined in the apps.yaml and used when determining whether gas is cheaper than electricity. Once tomorrows rate is published, the forecast on which slots will be targeted for iBoost will use the wrong price.

**Expected behavior**
When iBoost is determining whether gas is cheaper than electricity within the plan forecast, it should use the appropriate gas sensor based on whether it is forecasting for today or tomorrow.

The next rate will have a state of 'Unknown' until the price is published at 4pm. Therefore the plan should estimate tomorrows iBoost slots using current gas price until tomorrow's gas price is released.

**Predbat version**
 
v7.6.11

**Environment details**

- GivEnergy AIO
- Solis PV inverter
- Standard HAOS installer

**Screenshots**
Current and next rate sensors listed shown under **Gas Price** card.

[image]

apps.yaml:

[image]

**Log file**

Not applicable for this issue.

springfall2008, 2024-04-11:
In theory this should work, it will pull rate data from previous, current and next rate events.

Are you seeing that it doesn't work?

nbullus, 2024-04-11:
It looks like the current/next gas rate sensors don't work as I thought. The gas tracker price doesn't change at 4pm as originally thought, but actually changes at midnight and so there is only ever one rate to reference each day, i.e. the current rate. The next gas rate sensor remains as "Unknown", So when Predbat it is looking at tomorrows Agile slots to see if any are cheaper than gas, it has to compare with todays gas price  Any Agile slots tomorrow earmarked today for iBoost may then change at midnight once the market gas price changes. As the gas price may go up/down it has to re-calculate based on the the updated gas price. Therefore there should be no issue with how it is working and so close the report down.`