springfall2008/batpred #937: REST failed to setReserve

Surb109, 2024-04-11:
**Describe the bug**
Getting status message "Warn - Inverter 0 REST failed to setReserve to 100 got 4"

**Expected behavior**
Description of current action being performed

**Predbat version**
v7.16.12

**Environment details**

- Givenergy AIO 13.4kWH battery
- Running Home Assistant container in Docker
- Using GivTCP integration

**Screenshots**
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

RobinCu, 2024-04-11:
This should sort it: https://github.com/springfall2008/batpred/issues/879

Surb109, 2024-04-11:
Thanks Robin for the prompt response.

I have made the suggested change and it's now seems to be working correctly.

I'll try and do a better job of searching the issues before logging a ticket. I did try but didn't find anything similar 🙂

RobinCu, 2024-04-11:
No problem. Working out the correct search term can be a challenge in itself!

gcoan, 2024-04-11:
I plan to put together a 'quick reference' for different inverter types as to specific predbat configuration values to be set.  Try to make it easier for others to find these things than them being buried in the docs

closed 2024-09-01 by springfall2008