springfall2008/batpred #940: Pls add support for Foxess

rddelacosta, 2024-04-11:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
Please add support for Foxess

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

springfall2008, 2024-04-12:
Hi,

This is very likely possible, but I'm afraid I don't own a Fox so I can't just add support with no details.

Do you have the battery integration into home assistant and if so are the control documented?

nickgee31, 2024-04-12:
I'm about to set this integration up for a family member in the next few weeks: https://github.com/nathanmarlor/foxess_modbus - using the USB to RS485 adapter. 

Happy to help with testing when I've got it up and running for them.

rddelacosta, 2024-04-12:
> Hi,
> 
> This is very likely possible, but I'm afraid I don't own a Fox so I can't just add support with no details.
> 
> Do you have the battery integration into home assistant and if so are the control documented?

I integrated it via https://github.com/nathanmarlor/foxess_modbus. There is a similar integration for controlling charging but is better suited for octopus flux rather than octopus agile (https://github.com/nathanmarlor/foxess_em)

nickgee31, 2024-04-19:
I've had a go at setting this up with Predbat today but seem to be hitting a bit of a stumbling block. It starts the discharge from 23:10 and then appears to get stuck when trying to switch into charge mode at 23:30 with the error: Exception raised argument of type 'int' is not iterable.

I've attached the predbat.log and the apps.yaml that I've made which seems to work well. But clearly there is something wrong with what I've done! 

[predbat.log](https://github.com/springfall2008/batpred/files/15046111/predbat.log)
Apps.yaml - https://pastebin.com/LKPXYJHW

nickgee31, 2024-04-21:
I've got this up and running now - https://pastebin.com/qvru4Csf

Created two number helpers as it seems to need to control the charge and discharge rate. In reality these two created helpers do nothing but they're just there to make it work.

springfall2008, 2024-04-21:
If you set 'output_charge_control' to 'current' then it will control the charge/discharge using 

timed_charge_current/timed_discharge_current entities configured in apps.yaml

Or if you set it to 'none' then it won't do either of these

nickgee31, 2024-04-22:
Ah perfect will give that a go thanks! Not sure if this is specific to this or needs a general discussion but do you know how to stop the discharge during the day: 
[image]
I would think it would be better to leave the discharge as late as possible in case the battery is needed. I've attached a couple of predbat logs in case it gives any insight.
[predbat.log](https://github.com/springfall2008/batpred/files/15060019/predbat.log)
[predbat-1.log](https://github.com/springfall2008/batpred/files/15060020/predbat-1.log)

rddelacosta, 2024-05-05:
@nickgee31 @springfall2008 thanks for your work for potentially supporting foxess!

Did you manage to sort out the issue re the discharging during the day?

rddelacosta, 2024-06-25:
Hi both, thanks for these.

@springfall2008  I've managed to set this up - for the most part it works (will see if it charges actually), but I regularly encounter "Error: Exception raised 'NoneType' object has no attribute 'split'".

Pls see my setup here: https://pastebin.com/1UVtMwRB

Pls see log for predbat here: https://pastebin.com/hbbdjv5v

Hope you could please assist me on this.

Thanks

icornish72, 2024-06-25:
I have set this up as well, following the advise here.
 I get error:- 
2024-06-25 22:29:48.958387: Error: Exception raised can only concatenate str (not "NoneType") to str
2024-06-25 22:29:48.960012: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 11002, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 10029, in update_pred
    self.record_status(
  File "/config/predbat.py", line 1613, in record_status
    self.call_notify("Predbat status change to: " + message + extra)
  File "/config/predbat.py", line 109, in call_notify
    self.call_service_wrapper("notify/" + device, message=message)
TypeError: can only concatenate str (not "NoneType") to str

This was because I left Notify blank. Putting something in here (ie a valid Telegram Notifier), and it all seems to magically work (In Monitor Mode). I will leave it overnight and see what occurs...

gcoan, 2024-06-25:
Have you tried commenting the notify line out of apps.yaml rather than setting it to a blank entry?

icornish72, 2024-06-25:
I may try that tomorrow - I'm going to let it run overnight, as I kinda like the plan (and the charging toggle seems to work!). Getting Notifications keeps me aware of what it might be doing...

gcoan, 2024-06-25:
> I may try that tomorrow - I'm going to let it run overnight, as I kinda like the plan (and the charging toggle seems to work!). Getting Notifications keeps me aware of what it might be doing...
> 
> 

personally I too like the notifications so have never tried it without notifications off.  Good luck !

rddelacosta, 2024-06-25:
> I may try that tomorrow - I'm going to let it run overnight, as I kinda like the plan (and the charging toggle seems to work!). Getting Notifications keeps me aware of what it might be doing...

Have you tried charging it? Mine is charging at just 41w for some reason - dunno why.

rddelacosta, 2024-06-26:
> I have set this up as well, following the advise here. I get error:- 2024-06-25 22:29:48.958387: Error: Exception raised can only concatenate str (not "NoneType") to str 2024-06-25 22:29:48.960012: [...]

Mine is blank and it works - I get notifications from my mobile app

rddelacosta, 2024-06-26:
Have figured out that for the foxes modbus integration, I should specify the charge rate  (number.foxess_inverter_force_charge_power) and discharge rate (number.foxess_inverter_force_discharge_power). I've calculated mine using 20A x lower end of my battery's voltage (i.e., c. 2kw), but I think max ampere is like 50A and default is like 35A.
@springfall2008  if I want predbat to calculate the charging curve, should I use the charge rates above or should I use the actual historical charge rate (including data when battery was charged by the solar panel)? Thanks

letsstartbeinganon, 2024-08-21:
Is there any YAML file available that I can just copy and paste over to my own set up? The below Pastebin links now seem to have been removed.

> Hi both, thanks for these.
> 
> @springfall2008 I've managed to set this up - for the most part it works (will see if it charges actually), but I regularly encounter "Error: Exception raised 'NoneType' object has no attribute 'split'".
> 
> Pls see my setup here: https://pastebin.com/1UVtMwRB
> 
> Pls see log for predbat here: https://pastebin.com/hbbdjv5v
> 
> Also, for some reason when it force charged, only 41w was used to charge my batteries. Hence instead of c. 20% to c. 70%, my batteries were essentially uncharged even though it did charge. Any fix for this?
> 
> Hope you could please assist me on this.
> 
> Thanks

springfall2008, 2024-09-01:
I believe this is now documented, please review and raise a ticket for any new updates

closed 2024-09-01 by springfall2008