springfall2008/batpred #941: Forcing a discharge slot that is in progress to Idle leave inverter in Eco (Paused) Mode

mpartington, 2024-04-11:
**Describe the bug**
Today Predbat started a forced discharge earlier in the day than I would have liked, so I used the override.to select force idle.  It stopped the discharge,.but only later did I realise the battery was not supplying load.

The battery mode was Eco (Paused)

**Expected behavior**
The inverter to be fully reset to Eco mode and able to charge and discharge

I haven't tested cancelling a forced charge slot whilst in progress, but this should also be reset if necessary.

**Predbat version**
V7.16.12

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
[image]

**Log file**
Easy to replicate. Select.force discharge, then once discharging, select forced idle for the same slot.  Will go into eco paused mode and stop.supplying the house

springfall2008, 2024-04-11:
Eco (Paused) is when reserve is at 100%, this is a hold charge? What does the show?

mpartington, 2024-04-11:
> Eco (Paused) is when reserve is at 100%, this is a hold charge? What does the show?

Is this what you need?  Predbat says it was idle, but it's ECO (paused) in the Giv TCP modes. SOC 97%.  I'll grab the logs for you

[image]

mpartington, 2024-04-11:
Ah, you wanted reserve - thats at 4%

[image]

mpartington, 2024-04-11:
Might be because battery power mode was off? (not turned off by me)
[image]

gcoan, 2024-04-11:
I had something similar yesterday.  The battery was left in Eco (Paused) when Predbat said it should be in Idle.  Reserve I don't think was 100%, just mode was Eco Paused.

I manually changed it and the battery started discharging.

springfall2008, 2024-11-16:
Should be fixed now, let me know if not