springfall2008/batpred #943: Plan colours could be more gradiated to distinguish between cost/kwh in each plan segment

Jorthax, 2024-04-11:
**Is your feature request related to a problem? Please describe.**
It is difficult at a glance in this plan to see that electricity is 2p a unit after 14p a unit, before it drops to 0 and goes blue.

**Describe the solution you'd like**
It would be nice if the plan used Blue <0 , then Green->Yellow->Orange->Red gradiant for the other price slots.

**Describe alternatives you've considered**
Fixed cutoff points (user defined?) for each of the colour bands or %'s of the next 24 hours. e.g. Green <10% of the next 24hrs 'mean' cost.

**Additional context**
I would hope this would improve the 'at a glance' nature of the plan. Thank you for the consideration.

[image]

gcoan, 2024-04-11:
Have you seen Predbat table card?  Allows more customisation of the predbat plan https://github.com/pacemaker82/PredBat-Table-Card

Jorthax, 2024-04-11:
> Have you seen Predbat table card? Allows more customisation of the predbat plan https://github.com/pacemaker82/PredBat-Table-Card

I had not seen this! I will look into it. Thank you very much :)

closed 2026-09-18 by chalfontchubby