springfall2008/batpred #947: After update to 7.16.13 getting error message

BondBug, 2024-04-13:
**Describe the bug**
error message after update to 7.16.13 

**Expected behavior**
A clear and concise description of what you expected to happen.

7.16.13

2024-04-13 10:13:08.074741 WARNING pred_bat: ------------------------------------------------------------
2024-04-13 10:13:08.074852 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-04-13 10:13:08.075241 WARNING pred_bat: Worker Ags: {'id': 'ec38599479d949839075038b7d5acde0', 'name': 'pred_bat', 'objectid': 'b026101bfb334a71b317497c4c5fee9b', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7fcfc8c30e90>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-04-13 10:13:08.075427 WARNING pred_bat: ------------------------------------------------------------
2024-04-13 10:13:08.076488 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 13159, in update_time_loop
    raise e
  File "/config/apps/predbat.py", line 13154, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12166, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 10778, in calculate_plan
    self.publish_discharge_limit(self.discharge_window_best, self.discharge_limits_best, best=True)
  File "/config/apps/predbat.py", line 8174, in publish_discharge_limit
    state=discharge_end_str,
          ^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'discharge_end_in_minutes' where it is not associated with a value

2024-04-13 10:13:08.076587 WARNING pred_bat: ------------------------------------------------------------

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

gcoan, 2024-04-13:
I just upgraded to 7.16.13 and it installed and ran fine the first time for me

asplatinum2, 2024-04-13:
I have the same issue.

BondBug, 2024-04-13:
Just to add I am using the combined appdeamon- predbat install and running in a docker container on a Mac Mini

thevilla, 2024-04-13:
Similar error here after upgrade to 7.16.13.

2024-04-13 12:00:47.539421 INFO pred_bat: ERROR: Exception raised cannot access local variable 'discharge_end_in_minutes' where it is not associated with a value

Jorthax, 2024-04-13:
> Similar error here after upgrade to 7.16.13.
> 
> 2024-04-13 12:00:47.539421 INFO pred_bat: ERROR: Exception raised cannot access local variable 'discharge_end_in_minutes' where it is not associated with a value

Same error here.

springfall2008, 2024-04-13:
Sorry fixed in 7.16.14

thevilla, 2024-04-13:
Looking good for me.  Thank you for the very prompt fix and your continued support.

closed 2024-04-21 by springfall2008