springfall2008/batpred #951: iBoost Alternative Gas Price Source

dandwhelan, 2024-04-14:
**Is your feature request related to a problem? Please describe.**
I used iBoost in a backwards, non-traditional iBoost way. I want to manually enter a fake gas price with an input.number.namehere field. Predbat doesn't seem to allow me to use inputs and only accepts the sensor  "sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'"

I'd like to use ... 

  #metric_octopus_gas: 'input_number.gasprice'

**Describe alternatives you've considered**
Manual HA Automations

**Additional context**
Can you please also allow iBoost today accept multiple sources to populate the KWs used? Abit like how you can add more in 'car_charging_energy'

Thanks

dandwhelan, 2024-04-14:
So I use the use iboost when cheaper than gas. Another way you could do this is implement the car don't charge over x price into iboost.

springfall2008, 2024-04-20:
There is a version now on main where you can set the gas rate manually like you do electric e.g.

rates_gas:
  rate: 5.0

dandwhelan, 2024-04-20:
Nice one. Thanks

closed 2024-04-20 by dandwhelan