springfall2008/batpred #952: Solcast PV predication not showing

stokebob, 2024-04-14:
**Describe the bug**
Solcast is setup and the sensors are working in HA. However, predbat doesn't seem to be using any predicted solar. On the predbat plan the solar column is blank (no values). In the 'data prediction' chart the PV and PV actual figures are the same, and increase together as solar is generated (although I've only just installed predbat, so I unsure as the weather this maybe expected behaviour).

I've tried manually setting the sensor names in the apps.yaml file, this didn't change anything. Although when I check the logs predbat is pulling the values from the sensors as expected, so I don't think this is the issue.

**Expected behavior**
Predbat to use the predicted solar generation thoughout the day in it's calculations.

**Predbat version**
v7.16.15

**Environment details**

- Inverter and battery setup: GivEnergy AIO + 5kw Solar PV
- Standard HAOS installer or Docker: HAOS

**Screenshots**
[image]
[image]

**Log file**
[log.txt](https://github.com/springfall2008/batpred/files/14970047/log.txt)

gcoan, 2024-04-14:
That's weird

the logfile shows that predbat is picking up solcast forecast, but the actual forecast appears to be repeating:

```yaml
2024-04-13 23:55:03.742687 INFO pred_bat: PV Data for pv_forecast_today total 11.43 kWh
2024-04-13 23:55:03.759663 INFO pred_bat: PV Data for pv_forecast_tomorrow total 23.42 kWh
2024-04-13 23:55:03.774926 INFO pred_bat: PV Data for pv_forecast_d3 total 28.75 kWh
2024-04-13 23:55:03.792659 INFO pred_bat: PV Data for pv_forecast_d4 total 22.74 kWh
...
2024-04-13 23:55:06.853019 INFO pred_bat:      PV: [    8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2,     8.2]
```

Can you share what's in apps.yaml for the PV config and have a look at the solcast sensor values.

You could try reloading the solcast integration.  Assume you have got an automation running that retrieves the PV data from solcast into HA?

stokebob, 2024-04-14:
Yes, automation is setup to retrieve PV data each night.

I've have restarted HA with no effect, or did you mean uninstall/install the Solcast integration?

Although it looks as though the Solcast integration is working correctly. I've had a look at the Solcast sensors and they all appear to working as expected. I'm assuming batpred pulls the data it need from the attributes, looking there the data doesn't seem to repeat at all.

I installed batpred on Friday, it hasn't worked from the go. I left it a bit to see if it would click in , but nothing changed. Hopefully I've gotten the batpred settings right?

app.yaml PV settings:
  # Set these to match solcast sensor names
  # The regular expression (re:) makes the solcast bit optional
  # If these don't match find your own names in Home Assistant
  pv_forecast_today: re:(sensor.(solcast_|)(pv_forecast_|)forecast_today)
  pv_forecast_tomorrow: re:(sensor.(solcast_|)(pv_forecast_|)forecast_tomorrow)
  pv_forecast_d3: re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3))
  pv_forecast_d4: re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4))
  #pv_forecast_today: sensor.solcast_pv_forecast_forecast_today
  #pv_forecast_tomorrow: sensor.solcast_pv_forecast_forecast_tomorrow
  #pv_forecast_d3: sensor.solcast_pv_forecast_forecast_day_3
  #pv_forecast_d4: sensor.solcast_pv_forecast_forecast_day_4

Data pulled from Solcast sensor attribute:

DetailedForecast
- period_start: '2024-04-14T00:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T00:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T01:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T01:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T02:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T02:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T03:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T03:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T04:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T04:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T05:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T05:30:00+01:00'
pv_estimate: 0
[... 222 lines cut ...]
pv_estimate90: 1.3428
- period_start: '2024-04-14T19:00:00+01:00'
pv_estimate: 0.0926
pv_estimate10: 0.0563
pv_estimate90: 0.2245
- period_start: '2024-04-14T20:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T21:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T22:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-04-14T23:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
Dayname
Sunday
DataCorrect
true

RobinCu, 2024-04-14:
I saw something similar (no PV data in the plan) when I was messing about with the PV estimate in apps.yml. The section is further down the apps.yml, after the rate information (line 351 in mine, but it may be different in the latest default apps.yml on GitHub).

  # For pv estimate, leave blank for central estimate, or add 10 for 10% curve (worst case) or 90 or 90% curve (best case)
  # If you use 10 then disable pv_metric10_weight below
  # pv_estimate: 90

Where it says "leave blank", I found that leaving the value blank, i.e.:
   pv_estimate:
casued the issue you are seeing.

If I commented out the line:
 # pv_estimate: 90
I found it worked and picked up the correct central figure.

It could be unrelated, but I thought I'd mention it, as it may save some head scratching.

stokebob, 2024-04-14:
This fixed it! That's great, thanks.

> pv_estimate:

The pv_estimate line I'd left blank, commenting it out resolved the issue.

I'm guessing this is a bug that needs fixing, or just rewording the yaml file to comment out if not needed.

gcoan, 2024-04-14:
I’ll update the documentation to make it clearer

gcoan, 2024-04-14:
I've fixed the predbat code so if pv_estimate is set in apps.yaml, but no value is set, then the central estimate is used (per the documentation).

The original code was trying to find "pv_estimateNone" in the solcast forecast which of course failed.

It'll get included in the next code push I do

springfall2008, 2024-11-16:
Closing this old ticket, please open a new one if you have an up to date question