springfall2008/batpred #953: Historical plan view

dandwhelan, 2024-04-14:
**Is your feature request related to a problem? Please describe.**

No

**Describe the solution you'd like**

A feature to record the daily money saved/spent

**Describe alternatives you've considered**

Excel.

gcoan, 2024-04-14:
Try predbat.cost_today sensor which is the daily accumulated net cost as calculated by Predbat, or cost_today_import & cost_today_export if you want these broken out

There's also the Octopus integration daily_accumulated_cost sensor which does the same based on the Octopus meter data (need a Home Mini for this to give you today's data though)

bensebborn, 2024-04-18:
This would also be good, so we can check when first setting things up whether the battery did discharge or charge when it was expected? Unless there’s another way of doublechecking that

closed 2024-08-31 by dandwhelan