springfall2008/batpred #956: Incorrect energy requirement calculated for Predbat led Car Charging using Manual SOC

mpartington, 2024-04-14:
**Describe the bug**
It appears Predbat is incorrectly calculating car charging losses, leading to an incorrect number of kWh being scheduled 

For example, the screenshot shows I set a manual SoC level of 45.3kWh and I have hardcoded by battery capacity to 52kWh in apps.yaml.  Therefore I need 6.7kWh + losses

My losses are set to 5%, so:
6.7 x 1.05 = 7.04 kWh, but Predbat schedules 6.78 kWh (too low)

I then tried 20% losses
6.7 x 1.2 = 8.04 kWh, but Predbat schedules 8.38 (too high)

I then tried 0% losses and Predbat correctly schedules 6.7kWh

**Expected behavior**
Correctly calculate charge required

**Predbat version**

v7.16.15

**Environment details**

- Standard HAOS installer or Docker

**Screenshots**
[image]

[image]

**Log file**
[appdaemon.log](https://github.com/springfall2008/batpred/files/14972130/appdaemon.log)

mpartington, 2024-04-15:
Strangely the amount of kWh needed seems accurate on this trial (with 5% losses). However there seems to be an issue with the slot cost sorting. The cheaper slot should be more utilised
[image]

mpartington, 2024-04-15:
Another thing that seems to be happening almost every time (which didn't used to) is the first time you change the input number, it pumps back to the original value
[image]

springfall2008, 2024-04-20:
Fix now on main for testing

closed 2024-04-21 by springfall2008