springfall2008/batpred #957: No load sensor (Sunsynk)

bensebborn, 2024-04-15:
Hi

I am trying to set
Up my Sunsynk inverter, and most options I’ve managed to sort

However the load sensor on Sunsynk is nonsense and often doesn’t even add up in their own app. 

So I think I need to create my own sensor to calculate it?

I also have another Inverter (solis) that just does PV and isn’t connected to the Sunsynk

is this correct:

Current Load = (Grid import + Sunsynk AC output + Solis AC output)

bensebborn, 2024-04-15:
Saying that, how would it account for things like force discharging the battery? Don’t want that to appear as house load

gcoan, 2024-04-15:
I use a custom sensor for my GivEnergy inverters as the inverter ones are wrong. The template sensor code is in the apps.yaml part of the predbat documentation, but basically it's:

Load = import - export + pv generation - battery charge + battery discharge 

If you want to subtract EV charge you can or you can do this in predbat as part of the car charging configuration

Meatballs1, 2024-04-17:
I don't rely on the sunsynk data as I'm AC coupled so it can't really see my solar.

I have separate sensor for load (which goes negative for export) and my template is basically:

`Load_power = grid_sensor_power + solaredge_pv_power + [0,sunsynk_inverter_power]|max`

Using max avoids any issues with battery charging being seen as load of you need to exclude that

bensebborn, 2024-04-17:
Thanks a lot. Got it working with:

` {{ states('sensor.sunsynk_inverter_grid_power')|float(0)  + states('sensor.total_solar_output')|float(0) - states('sensor.sunsynk_inverter_battery_power')|float(0) }}`