springfall2008/batpred #958: Struggling to obtain battery charge curve depite following documentation

johnleeshore, 2024-04-15:
**Describe the bug**
Hi I am struggling to obtain the battery charge curve despite follow the instructions (I think)

**Expected behavior**
Create battery charge curve in predbat.log

**Predbat version**

7.16.15

**Environment details**

- GIV-HY3.6 Inverter
- Standard HAOS installer
- GivEnergy 9.5kwH battery

**Screenshots**
Relevant info in apps.yaml:

_ge_cloud_data: False
  ge_cloud_serial: '{geserial}'
  ge_cloud_key: 'xxxx'
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  #
  # Run balance inverters every N seconds (0=disabled) - only for multi-inverter
  balance_inverters_seconds: 60
  #
  # When set use the REST API rather than HA entity for control, should be more reliable/faster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
  givtcp_rest:
    - 'http://homeassistant.local:6345'
  #  - 'http://homeassistant.local:6346'

  # When enabled automatic restart will restart the add-on if communication fails
  # Example below is auto-restart for GivTCP add-on itself
  auto_restart:
    - shell: 'rm -rf /homeassistant/GivTCP/*.pkl'
    - service: hassio/addon_restart
      addon: a6a2857d_givtcp

  #  Example on how to restart the inverter via GivTCP
  #  - service: switch.turn_on
  #    entity_id: switch.givtcp_{geserial}_reboot_invertor

  # If not using REST then instead set the Control here (one for each inverter)
  # You should keep this section even when using REST as a fallback if it fails and for charge curve calculations
  charge_rate:
    - number.givtcp_{geserial}_battery_charge_rate
  #  - number.givtcp2_{geserial2}_battery_charge_rate
  discharge_rate:
    - number.givtcp_{geserial}_battery_discharge_rate
  #  - number.givtcp2_{geserial2}_battery_discharge_rate
  battery_power:
    - sensor.givtcp_{geserial}_battery_power
  # - sensor.givtcp2_{geserial2}_battery_power
  pv_power:
    - number.givtcp_{geserial}_pv_power
   # - number.givtcp2_{geserial2}_pv_power
  load_power:
    - number.givtcp_{geserial}_load_power
   # - number.givtcp2_{geserial2}_load_power
  soc_kw:
    - sensor.givtcp_{geserial}_soc_kwh
   # - sensor.givtcp2_{geserial2}_soc_kwh
  soc_max:
    - sensor.givtcp_{geserial}_battery_capacity_kwh
   # - sensor.givtcp2_{geserial2}_battery_capacity_kwh
  reserve:
    - number.givtcp_{geserial}_battery_power_reserve
   # - number.givtcp2_{geserial2}_battery_power_reserve
  inverter_mode:
    - select.givtcp_{geserial}_mode
   # - select.givtcp2_{geserial2}_mode
  inverter_time:
    - sensor.givtcp_{geserial}_invertor_time
  #  - sensor.givtcp2_{geserial2}_invertor_time
  charge_start_time:
    - select.givtcp_{geserial}_charge_start_time_slot_1
   # - select.givtcp2_{geserial2}_charge_start_time_slot_1
  charge_end_time:
    - select.givtcp_{geserial}_charge_end_time_slot_1
   # - select.givtcp2_{geserial2}_charge_end_time_slot_1
  charge_limit:
    - number.givtcp_{geserial}_target_soc
   # - number.givtcp2_{geserial2}_target_soc
  scheduled_charge_enable:
    - switch.givtcp_{geserial}_enable_charge_schedule
   # - switch.givtcp2_{geserial2}_enable_charge_schedule
  scheduled_discharge_enable:
    - switch.givtcp_{geserial}_enable_discharge_schedule
   # - switch.givtcp2_{geserial2}_enable_discharge_schedule
  discharge_start_time:
    - select.givtcp_{geserial}_discharge_start_time_slot_1
   # - select.givtcp2_{geserial2}_discharge_start_time_slot_1
  discharge_end_time:
    - select.givtcp_{geserial}_discharge_end_time_slot_1
  #  - select.givtcp2_{geserial2}_discharge_end_time_slot_1_

**Log file**

PS I have replaced inverter number with xxxxx

2024-04-15 10:14:38.999895 INFO pred_bat: Find charge curve with sensors sensor.givtcp_xxxxx_soc_kwh and number.givtcp_fa2317g610_battery_charge_rate and predbat.status and sensor.givtcp_xxxxx_battery_power
2024-04-15 10:14:39.254031 INFO pred_bat: Find charge curve has 15.0 days of data, max days 15
2024-04-15 10:14:39.290895 INFO pred_bat: Note: Can not find battery charge curve (no final curve), one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
2024-04-15 10:14:39.291964 INFO pred_bat: Find discharge curve with sensors sensor.givtcp_xxxx_soc_kwh and number.givtcp_fa2317g610_battery_discharge_rate and predbat.status and sensor.givtcp_xxxx_battery_power
2024-04-15 10:14:39.540414 INFO pred_bat: Find discharge curve has 15.0 days of data, max days 15
2024-04-15 10:14:39.581464 INFO pred_bat: Note: Can not find battery charge curve (no final curve), one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
2024-04-15 10:14:39.582767 INFO pred_bat:

johnleeshore, 2024-04-15:
Ah yes that could be it. It’s only been active for 6 days. Thanks for the prompt reply Geoffrey.

RobinCu, 2024-04-15:
That was my first thought. Strange that the log file says 15.0 days of charge curve data. 

Can I just check your apps.yml says (approx line 161):
  battery_charge_power_curve: auto
  
An interim solution would be to uncomment the default curve, which I found to be quite accurate prior to the auto feature being introduced:
battery_charge_power_curve: 
    91 : 0.91
    92 : 0.81
    93 : 0.71
    94 : 0.62
    95 : 0.52
    96 : 0.43
    97 : 0.33
    98 : 0.24
    99 : 0.24
    100 : 0.24

johnleeshore, 2024-04-15:
[image]

RobinCu, 2024-04-15:
Ah OK.  If you remove the hash symbol from in front of battery_charge_power_curve:  and   battery_discharge_power_curve:

Then on both lines, after the semi-colon, type auto.  Like this:

# Some batteries tail off their charge rate at high soc%
  # enter the charging curve here as a % of the max charge rate for each soc pecentage.
  # the default is 1.0 (full power)
  # The example below is from GE 9.5kwh battery with latest firmware and gen1 inverter
  # Predbat can compute this curve automatically if you have enough data, restart the add-on and look in the logfile for the data
  # once set here Predbat will no longer re-compute the curve.
  # Can also be set to 'auto' to just use the calculation curve, not recommended if you are using low power charging mode.
  battery_charge_power_curve: auto
  #  91 : 0.91
  #  92 : 0.81
  #  93 : 0.71
  #  94 : 0.62
  #  95 : 0.52
  #  96 : 0.43
  #  97 : 0.33
  #  98 : 0.24
  #  99 : 0.24
  #  100 : 0.24
  battery_charge_power_curve_discharge: auto

RobinCu, 2024-04-15:
I really should work out how to paste code into GitHub, so it doesn't appear in a large bold text!
[image]

gcoan, 2024-04-15:
So you don't have anything configured.

You can uncomment those lines as @RobinCu suggests or try manually forcing it to create the curve, but it's trying to do that and failing so not sure that'd make much difference 

battery_charge_power_curve:
  auto

johnleeshore, 2024-04-15:
I'll have a go now. Presume will have to wait for data?

johnleeshore, 2024-04-15:
[image]

gcoan, 2024-04-15:
> [image]
> 
> 

I have the auto on a separate line indented with two spaces, this might be ok. 

Does that clear the error in the log file?

johnleeshore, 2024-04-15:
`# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  # geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today:
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today:
    - sensor.givtcp_{geserial}_export_energy_today_kwh
  pv_today:
    - sensor.givtcp_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
  #
  ge_cloud_data: False
  ge_cloud_serial: '{geserial}'
  ge_cloud_key: 'xxxx'
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  #
  # Run balance inverters every N seconds (0=disabled) - only for multi-inverter
  balance_inverters_seconds: 60
[... 326 lines cut ...]
  # Gas rates for comparison
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  watch_list:
    - '{octopus_intelligent_slot}'
    - '{octopus_ready_time}'
    - '{octopus_charge_limit}'
    - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
`

RobinCu, 2024-04-15:
> > [image]
> > 
> > 
> 
> I have the auto on a separate line indented with two spaces, this might be ok. 
> 
> Does that clear the error in the log file?

That's interesting. I queried it with Trefor, when I couldn't get the syntax correct and he said same line.  It must have more flexibility than first thought.

johnleeshore, 2024-04-15:
[image]

johnleeshore, 2024-04-15:
This is my apps.yaml file

```
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  # geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today:
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today:
    - sensor.givtcp_{geserial}_export_energy_today_kwh
  pv_today:
    - sensor.givtcp_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
  #
  ge_cloud_data: False
  ge_cloud_serial: '{geserial}'
  ge_cloud_key: 'xxxx'
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
[... 330 lines cut ...]
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  watch_list:
    - '{octopus_intelligent_slot}'
    - '{octopus_ready_time}'
    - '{octopus_charge_limit}'
    - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'

```

gcoan, 2024-04-15:
This all looks fine. I'm guessing it is lack of history that's causing the problem.  

So for now remove the auto lines and uncomment the curves

Ie

```yaml

battery_charge_power_curve: 
  91 : 0.91
  92 : 0.81
  93 : 0.71
  94 : 0.62
  95 : 0.52
  96 : 0.43
  97 : 0.33
  98 : 0.24
  99 : 0.24
  100 : 0.24

battery_discharge_power_curve: 
  4 : 1.0
```

johnleeshore, 2024-04-15:
Cheers for that. I'll wait

johnleeshore, 2024-04-15:
Just noticed I have a corrupted database saved in home assistant and so only have info for last 36 hours!
[image]
I might have to wait a bit longer....

gcoan, 2024-04-15:
Ah, that might explain the lack of history.

Be interesting to find out how much history predbat needs for this to work properly. I can then clarify this in the documentation for others

Thanks

johnleeshore, 2024-04-15:
Seen that corrupted sqlite dbs can be a problem and so might change to Maria dB

gcoan, 2024-04-15:
> Seen that corrupted sqlite dbs can be a problem and so might change to Maria dB

Don't !

There's plenty of videos on YouTube of how to convert to MariaDB,  all from about 2 years ago. It used to be that HA didn't handle large volumes of data in the database very well, and many people switched to MariaDB as a result.

Since then there's been a lot of changes to the way HA handles the database and in particular the use of statistics dramatically reduces the volume of long term history data held.  The advice from the Home Assistant team is that of course you can change if you want to, but if you do, you're on your own in terms of support.

Personally I never wanted to be a DBA when I was working, much less now ...

I had a db corruption about a month after I had HA, possibly as a result of me mucking about with the db directly and having too much history.  Since then I never edit the db directly, I only keep 14 days history and its been fine for me for 5+ months.  I think you were probably unlucky.

johnleeshore, 2024-04-15:
Ok I won't try Maria - thanks for the advice

closed 2024-09-07 by johnleeshore