springfall2008/batpred #960: Predbat not taking advantage of available plunge price periods for car charging

njhcarroll, 2024-04-15:
**Describe the bug**
When there are plunge price periods on Octopus agile, I've noticed that the car charging slots selected by predbat can include non-plunge price periods when there are still available slots resulting in the charging cost being higher than is necessary.

**Expected behavior**
Predbat would use all available plunge price slots (within the time by which charging should be complete) before going on to include the cheapest non-plunge price slots

**Predbat version**

v7.16.15

**Environment details**

not relevant for this.

**Screenshots**
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)
[appdaemon.log](https://github.com/springfall2008/batpred/files/14983787/appdaemon.log)

njhcarroll, 2024-04-15:
Should add that target car charging complete is set for 9am

mpartington, 2024-04-16:
Saw the same behaviour yesterday on the -0.64 slot
[image]

springfall2008, 2024-04-20:
Maybe fixed on main now with this PR: https://github.com/springfall2008/batpred/pull/977

closed 2024-04-21 by springfall2008