springfall2008/batpred #962: Solis Hybrid inverter and forced discharge not working due to Energy Mode switch

spetrie13, 2024-04-16:
**Describe the bug**
Justed moved from Go to Octopus Flux, but discharge did not take place as planned during 1600 to 1900 time slot.
Charging overnight seems to work, but not discharge. Plan shows correct charge and discharge plan.

**"Energy Storage Control Switch"** 

- During no discharge or no charge period the "Energy Storage Control Switch" indicates "Self-Use - No Grid Charging"
- During charge it changes from "Self-Use - No Grid Charging" to "Self-Use"
- During discharge it remains as "Self-Use - No Grid Charging"
- If I set Predbat to readonly and manually set "Energy Storage Control Switch"  to  "Self-Use" then discharge starts as per the plan with the discharge times set.

**Expected behavior**
Predbat indicates correct discharge in the correct period in the plan, but the "Energy Storage Control Switch" has the incorrect setting preventing the discharge.
On discharge the  "Energy Storage Control Switch"  should be set to  "Self-Use" 

For the solis inverter you only need to set the "Energy Storage Control Switch"  to  "Self-Use" and set the charge and discharge timers to enable charge and discharge.

**Predbat version**
V7.16.15

**Environment details**

- Inverter and battery setup:  Solis Hybrid inverter
- Solax Modbus integration
- Standard HAOS installer

**Screenshots**
"Energy Storage Control Switch"  to  "Self-Use":
[image]
[image]
[image]

Enable Predbat and it changes the "Energy Storage Control Switch" 
[image]
Battery returns to just discharge to house
[image]

spetrie13, 2024-04-17:
Screenshot today
[image]

springfall2008, 2024-04-17:
Are you able to test this fix from a branch 'https://github.com/springfall2008/batpred/tree/solis'

you need to copy predbat.py into your apps directory (overwrite the existing one)

spetrie13, 2024-04-17:
> Are you able to test this fix from a branch 'https://github.com/springfall2008/batpred/tree/solis'
> 
> you need to copy predbat.py into your apps directory (overwrite the existing one)

Will try tomorrow.

spetrie13, 2024-04-18:
Getting error message with test fix:
08:35:16 INFO pred_bat: ERROR: Exception raised cannot access local variable 'new_mode' where it is not associated with a value
[image]

spetrie13, 2024-04-18:
Scheduled discharge did take place in spite of the error messages.

[image]

[_logs.zip](https://github.com/springfall2008/batpred/files/15027169/_logs.zip)

springfall2008, 2024-04-18:
Can you please update from the branch again, hopefully I fixed this issue

spetrie13, 2024-04-19:
Charge worked last night (0200 today), but discharge did not work.
[image]
Not sure why discharge time starts at 0000 and ends at 1901. I would have thought it should have been set to start 1600 and ends at 1900 as the plan indicates. I believe and an overlap of the charge and discharge times cause the issue? Should we set the discharge start time when setting discharge period?
If I manually set the discharge time to 1600 then discharge starts with no other changes.
[image]
[image]

springfall2008, 2024-04-19:
Hmm okay, the overlap has no impact on GE inverters so I'll see if I can fix that one for you

springfall2008, 2024-04-19:
I've updated the branch with a potential fix, can you test again please?

spetrie13, 2024-04-19:
Seems to work. Just restarted Appdemon and set discharge hours to 0

It set the start and end discharge times:

17:54:52 INFO pred_bat: Completed run status Discharging
--
17:54:52 INFO pred_bat: Solis Energy Control Switch setting 35 Self-Use unchanged for eco True
17:54:52 INFO pred_bat: Inverter 0 set force discharge to True
17:54:52 INFO pred_bat: Inverter 0 Wrote Timed Export to inverter_mode, successfully now Timed Export
17:54:20 INFO pred_bat: Sleeping (workaround) as start/end of discharge window was just adjusted
17:54:20 INFO pred_bat: Inverter 0 Turning on scheduled discharge
17:54:20 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to True successfully and got on
17:54:18 INFO pred_bat: WARN: Inverter 0 Trying to press button.solis_modbus_update_charge_discharge_times didn't complete
17:54:06 INFO pred_bat: Inverter 0 Wrote 1 to discharge_end_minute, successfully now 1
17:54:04 INFO pred_bat: Inverter 0 Wrote 19 to discharge_end_hour, successfully now 19
17:54:02 INFO pred_bat: Inverter 0 Wrote 19:01:00 to discharge_end_time, successfully now 19:01:00
17:54:00 INFO pred_bat: Inverter 0 Set new end time to 19:01:00 was 00:00:00
17:54:00 INFO pred_bat: Inverter 0 Wrote 30 to discharge_start_minute, successfully now 30
17:53:58 INFO pred_bat: Inverter 0 Wrote 17 to discharge_start_hour, successfully now 17
17:53:56 INFO pred_bat: Inverter 0 Wrote 17:30:00 to discharge_start_time, successfully now 17:30:00
17:53:54 INFO pred_bat: Inverter 0 set new start time to 17:30:00
17:53:54 INFO pred_bat: Adjust idle time computed idle is 19:01:00-23:59:00
17:53:54 INFO pred_bat: Adjust idle time, charge 00:00:00-00:00:00 discharge 17:30:00-19:01:00
17:53:54 INFO pred_bat: Inverter 0 Adjust force discharge to True, change times from 00:00:00 - 00:00:00 to 17:30:00 - 19:01:00
17:53:54 INFO pred_bat: Discharging now - current SOC 19.75 and target 13.75
17:53:54 INFO pred_bat: Next discharge window will be: 2024-04-19 17:30:00+01:00 - 2024-04-19 19:01:00+01:00 at reserve 55
17:53:54 INFO pred_bat: Not setting charging window yet as not within the window (now 04-19 17:50:00 target set_window_minutes 30 charge start time 04-20 02:00:00
17:53:54 INFO pred_bat: Charge window will be: 2024-04-20 02:00:00+01:00 - 2024-04-20 05:00:00+01:00 - current soc 79 target 61
17:53:54 INFO pred_bat: WARN: Inverter 0 Trying to press button.solis_modbus_update_charge_discharge_times didn't complete
17:53:42 INFO pred_bat: Inverter 0 No write needed for discharge_end_minute: 0 == 0
17:53:42 INFO pred_bat: Inverter 0 No write needed for discharge_end_hour: 0 == 0
17:53:42 INFO pred_bat: Inverter 0 Wrote 00:00:00 to discharge_end_time, successfully now 00:00:00
17:53:40 INFO pred_bat: Inverter 0 Set new end time to 00:00:00 was 23:59:00
17:53:40 INFO pred_bat: Inverter 0 No write needed for discharge_start_minute: 0 == 0
17:53:40 INFO pred_bat: Inverter 0 No write needed for discharge_start_hour: 0 == 0
17:53:40 INFO pred_bat: Inverter 0 Wrote 00:00:00 to discharge_start_time, successfully now 00:00:00

[image]

spetrie13, 2024-04-22:
Hi,

Looks like the latest version predbat is not setting the Solis discharge start time again. Set to 00:00:00 to 19:01:00 when it should have been 16:00:00 to 19:01:00. It overlaps with the charge time so does not discharge.

16:00:31 INFO pred_bat: Charge window will be: 2024-04-23 02:00:00+01:00 - 2024-04-23 05:00:00+01:00 - current soc 99 target 16
16:00:31 INFO pred_bat: Configuring charge window now (now 04-22 16:00:00 target set_window_minutes 30 charge start time 04-23 02:00:00
16:00:31 INFO pred_bat: Adjust idle time, charge 02:00:00-05:00:00 discharge 00:00:00-00:00:00
16:00:31 INFO pred_bat: Adjust idle time computed idle is 16:00:00-23:59:00
16:00:33 INFO pred_bat: Inverter 0 Wrote 02:00:00 to charge_start_time, successfully now 02:00:00
16:00:35 INFO pred_bat: Inverter 0 Wrote 2 to charge_start_hour, successfully now 2
16:00:36 INFO pred_bat: Inverter 0 No write needed for charge_start_minute: 0 == 0
16:00:38 INFO pred_bat: Inverter 0 Wrote 05:00:00 to charge_end_time, successfully now 05:00:00
16:00:40 INFO pred_bat: Inverter 0 Wrote 5 to charge_end_hour, successfully now 5
16:00:40 INFO pred_bat: Inverter 0 No write needed for charge_end_minute: 0 == 0
16:00:52 INFO pred_bat: WARN: Inverter 0 Trying to press button.solis_modbus_update_charge_discharge_times didn't complete
16:00:52 INFO pred_bat: Inverter 0 Updated start and end charge window to 02:00:00 - 05:00:00 (old 05:00:00 - 16:00:00)
16:00:52 INFO pred_bat: Include original discharge start 04-22 00:00:00 with our start which is 04-22 00:00:00
16:00:52 INFO pred_bat: Next discharge window will be: 2024-04-22 00:00:00+01:00 - 2024-04-22 19:01:00+01:00 at reserve 21
16:00:52 INFO pred_bat: Discharging now - current SOC 19.8 and target 4.2
16:00:52 INFO pred_bat: Inverter 0 current discharge rate is 0.0 and new target is 5000
16:00:54 INFO pred_bat: Inverter 0 Wrote 5000 to discharge_rate, successfully now 5000
16:00:56 INFO pred_bat: Inverter 0 Wrote 94.0 to timed_discharge_current, successfully now 94.0
16:00:56 INFO pred_bat: Inverter 0 Adjust force discharge to True, change times from 00:00:00 - 00:00:00 to 00:00:00 - 19:01:00
16:00:56 INFO pred_bat: Adjust idle time, charge 02:00:00-05:00:00 discharge 00:00:00-19:01:00
16:00:56 INFO pred_bat: Adjust idle time computed idle is 19:01:00-23:59:00
16:00:56 INFO pred_bat: Inverter 0 Set new end time to 19:01:00 was 00:00:00
16:00:58 INFO pred_bat: Inverter 0 Wrote 19:01:00 to discharge_end_time, successfully now 19:01:00
16:01:00 INFO pred_bat: Inverter 0 Wrote 19 to discharge_end_hour, successfully now 19
16:01:02 INFO pred_bat: Inverter 0 Wrote 1 to discharge_end_minute, successfully now 1
16:01:14 INFO pred_bat: WARN: Inverter 0 Trying to press button.solis_modbus_update_charge_discharge_times didn't complete
16:01:16 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to True successfully and got on
16:01:16 INFO pred_bat: Inverter 0 Turning on scheduled discharge
16:01:16 INFO pred_bat: Sleeping (workaround) as start/end of discharge window was just adjusted
16:01:48 INFO pred_bat: Inverter 0 Wrote Timed Export to inverter_mode, successfully now Timed Export
16:01:48 INFO pred_bat: Inverter 0 set force discharge to True
16:01:48 INFO pred_bat: Solis Energy Control Switch setting 35 Self-Use unchanged for eco True
16:01:48 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
16:01:48 INFO pred_bat: Completed run status Discharging

spetrie13, 2024-04-23:
> completed via 1f65a37

Hi Trefor,

Solis inverter discharge not working with latest update.
The discharge start time is not being set and results in an overlap with
the charge times and discharge does not take place. (Quirk of Solis not
allowing time overlaps) This was working in the solis test branch.

[image: image.png]
Regards,

Stuart