springfall2008/batpred #966: Sunsynk inverter: Hold charge allows discharge

bensebborn, 2024-04-18:
**Describe the bug**

I’ve integrated our Sunsynk inverter based on the advise for “other inverters”. Charge and discharge work fine

So I’ve probably missed something in how we get the battery to hold charge?

 when the plan is marked as “Hold charge” the battery discharges as normal. 

**Expected behavior**
I would expect the battery to be set to charge mode to ensure any load uses the grid, which would maintain the battery SOC

instead it seems the load is covered by the battery

our plan shows 100% at 23:00 then holding until 04:30. When I checked the SOC was at 43% at 0430

**Predbat version**

7.16.15

**Environment details**

- Inverter and battery setup
- Sunsynk 
- Standard HAOS installer or Docker
- HAOS
- Anything else?

A few lines of log show inverter is in discharge mode

-04-18 03:35:11.965088 INFO pred_bat: Not setting charging window yet as not within the window (now 04-18 03:35:00 target set_window_minutes 30 charge start time 04-18 13:00:00
2024-04-18 03:35:11.973860 INFO pred_bat: Inverter 0 Call service template charge_stop_service = {'service': 'switch.turn_on', 'entity_id': 'switch.sunsynk_inverter_use_timer'}
2024-04-18 03:35:11.985135 INFO pred_bat: Inverter 0 Call service switch/turn_on with data {'entity_id': 'switch.sunsynk_inverter_use_timer'}
2024-04-18 03:35:11.992066 INFO pred_bat: Inverter 0 current discharge rate is 9740.0 and new target is 8000
2024-04-18 03:35:14.017883 INFO pred_bat: Inverter 0 Wrote 8000 to discharge_rate, successfully now 8000
2024-04-18 03:35:14.022443 INFO pred_bat: Current SOC 53% is greater than Target SOC 0. Grid Charge disabled.
2024-04-18 03:35:14.053152 INFO pred_bat: Will recompute the plan as it is now 10.0 minutes old and will exceed the max age of 10 minutes before the next run

[image]
[image]

config file
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat
  
  soc_max:
   - 28.8
   
  battery_rate_max:
   - 8000

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

[... 403 lines cut ...]
  #iboost_energy_scaling: 1.0
  # Gas rates for comparison
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  futurerate_adjust_import: True
  futurerate_adjust_export: False
  futurerate_peak_start: "16:00:00"
  futurerate_peak_end: "19:00:00"
  futurerate_peak_premium_import: 14
  futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'

bensebborn, 2024-04-18:
Here’s sensors showing charge mode and battery SOC

[image]
[image]

bensebborn, 2024-04-18:
But more info…

This is currently integrated using the basic method suggested for other inverters

So just has commands for charge/discharge

to do that on Sunsynk I am turning the timer on/off. When off this forces the battery to stay at 100% so this is the charge command. 

When timer is on, it allows discharge down to min SOC

alternatively we could leave the timer on and set the SOC and “grid charge” mode. When grid mode is on it will charge to the target SOC. when grid mode is off, it will discharge to the target SOC. 

Maybe that would work better?

bensebborn, 2024-04-18:
Update: I have tried incorporating the SOC setting. 

Because this is only one field, it’s not working correctly. When I set force charge, I can see the SOC target is set to 100% but then as it’s the same field it then gets set to the reserve Soc and overwritten so doesn’t charge

I did try to disable the “has reserve soc@ which allowed it to charge, but then I discharge the SOC remained on 100% so didn’t discharge. Is there a way of only setting the SOVC to target on charge, and reserve on discharge?

springfall2008, 2024-04-18:
So you want the target SOC to be set to the discharge target during export?

bensebborn, 2024-04-18:
So:

during charge, send only the 'soc_max' (as on Sunsynk, the 'soc' field acts as an upper limit during charge)
during discharge, send only the 'reserve' (as on Sunsynk, the 'soc' field acts as an lower limit during discharge)

I think those are the correct fields.

Currently it seems to send to both 'soc_max' and then 'reserve', so one figure overwrites the other.

bensebborn, 2024-04-18:
What I observed, is the 'SOC' on the inverter was initially set to '100' for example (charge soc target) then set to 0 very quickly

Config was:

reserve:
     - number.sunsynk_inverter_capacity_point_6
charge_limit:
     - number.sunsynk_inverter_capacity_point_6
     
     So it seems it was setting 'charge_limit' to 100 then setting 'reserve' to 0.
     
     However, because these both point to the same sensor on Sunsynk, they were overwriting each other, leaving the inverter soc at 0%.
     
     If that's not expected behaviour i can set it back to that config and send over the logs

bensebborn, 2024-04-18:
Here you go:

```
2024-04-18 21:46:55.088160 INFO pred_bat: Inverter 0 Current charge limit is 15 % and new target is 100 %
2024-04-18 21:46:59.095197 INFO pred_bat: Inverter 0 Wrote 100 to charge_limit, successfully now 100
2024-04-18 21:46:59.098203 INFO pred_bat: Inverter 0 Current Reserve is 100.0 % and new target is 12.0 %
2024-04-18 21:47:03.105737 INFO pred_bat: Inverter 0 Wrote 12.0 to reserve, successfully now 12
2024-04-18 21:49:24.410232 INFO pred_bat: Inverter 0 Wrote 49 to charge_limit, successfully now 49
2024-04-18 21:49:30.432829 INFO pred_bat: Inverter 0 Wrote 12.0 to reserve, successfully now 12
```

And observed the same on the 'number.sunsynk_inverter_capacity_point_6' sensor.

Initial: 15
Then: 100%
Then: 12%
Then: 49%
Then: 12%

This is due to both the 'reserve' and the 'current_limit' pointing to the same sensor, and both being set during charging.

We only have one SOC sensor on the inverter, so need just the 'charge_limit' to be set during charge, and 'reserve' to just be set during discharge

Hope that helps

springfall2008, 2024-04-19:
Reserve is the minimum battery level before it shuts off, but charge_limit is the AC charge limit before stopping charging. 

I can probably have an option to make sure it does as you suggest, let me have a look

bensebborn, 2024-04-19:
Thank you. It would make things much better on a Sunsynk/Deye inverter.

bensebborn, 2024-04-19:
In the meantime, is there any way to stop the battery from discharging during 'HoldCharge. SOC: 100%'?

It seems to just discarge during this period. I'm presuming due to the 'reserve' not being set due the above. Any workaround?

springfall2008, 2024-04-19:
I've created a new branch now with some of this rolled back into the code:

1. I added inverter type SK to predbat.py
  - Enabled target SOC
3. I change the code so if you have a target SOC but no reserve then it changes the target soc % to the reserve during a discharge.
4. I added a sunsynk.yaml template onto the branch, its like yours but more generic now and using the settings built into Predbat.

Please copy predbat.py from: https://github.com/springfall2008/batpred/blob/sunsynk/apps/predbat/predbat.py and also look at my template https://github.com/springfall2008/batpred/blob/sunsynk/templates/sunsynk.yaml

One issue I notice is that 'charge_rate' and 'discharge_rate' are sensors and can't be modified, they are supposed to be number's which can be changed so Predbat can cut the discharge rate to 0 during a charge for example.

bensebborn, 2024-04-19:
Thanks for this. Will give that a go and report back

With regards the charge / discharge rate, the inverter only gives us access to Max Charge/Discharge Current (Amps) We can’t set this as Watts. . 

So I created a sensor which returned power by (current x voltage)

So it’s accurate in terms of the figure returned but you’re right, it can’t them be modified?

edit: Have found a “max power” editable figure ( not current power) but it’s one figure that controls both charge and discharge depending upon what it’s doing. Would that work?

bensebborn, 2024-04-19:
Ok will Gcsss this a proper go over right to check charge rates etc

I’ve done a quick test:

force charge:  Set SOC to 88% which was same as “Limit %” shown on plan so looks good

force discharge:  Set SOC to 0% so this allowed discharge correctly

force idle:  SOC remained and wasn’t set. So here it was 88% which prevented the battery draining. Presume this needs to be reset down to 0%

So just the idle to sort I believe. 

Rather than the SOC being set to 0% on idle/discharge, can it be set to the reserve?  

Ben

Meatballs1, 2024-04-20:
I was just having a go at setting this up and then saw this.... The way I control the inverter (AC Coupled with separate SolarEdge PV) is:

* Idle - Timer On, Work Mode: Limited to Home
* Charging - Timer Off
* Freeze Charging - Battery max discharge current: 0
* Hold Charging - As above I guess?
* No Charge - Battery max charge current: 0  Battery max grid charge current: 0 (If I hit my min level, my battery BMS requests a force charge and the inverter will ignore battery max charge current and start grid charging if this isn't set). 
* Discharging - Timer On, Work Mode: Selling First
* Freeze Discharging - Battery max charge current: 0

Where battery is set to 0, it needs reverting back to the previous limit afterwards. 

All timeslots set to 20% (or min DoD)

I don't think I have enough history for some of my template load settings yet to really check the data yet

jacekowski, 2024-04-20:
It would be really good if we could get some sort of "simple" entitiy with current command (start/end times are not really necessary) and any additional parameters (minimum/maximum soc). that way it would be a lot simpler to write some sort of interface using HA automations or node-red to convert that to inverter commands rather than predbat trying to be clever

bensebborn, 2024-04-20:
> I was just having a go at setting this up and then saw this.... The way I control the inverter (AC Coupled with separate SolarEdge PV) is:
> 
> * Idle - Timer On, Work Mode: Limited to Home
> * Charging - Timer Off
> * Freeze Charging - Battery max discharge current: 0
> * Hold Charging - As above I guess?
> * No Charge - Battery max charge current: 0  Battery max grid charge current: 0 (If I hit my min level, my battery BMS requests a force charge and the inverter will ignore battery max charge current and start grid charging if this isn't set).
> * Discharging - Timer On, Work Mode: Selling First
> * Freeze Discharging - Battery max charge current: 0
> 
> Where battery is set to 0, it needs reverting back to the previous limit afterwards (or setting to the right level with each mode)
> 
> All timeslots set to 20% (or min DoD)
> 
> I don't think I have enough history for some of my template load settings yet to really check the data yet.
> 
> I'm not sure if the battery `Low battery`setting could be used to set a reserve (I dont think its currently exposed in the esphome version I have but I'm sure could digout the modbus register)

I’m moving away from that setup for Charging as it doesn’t take into account SOC

now that’s fixed my plan is to set all the timers the same except slot 6 so I that will cover all day

Then when I want to charge, tick Grid Charge Slot 6
SOC is then set to target SOC

this seems to be working well the only minor issue is that SunSynk only lets you change the time slots in 30 mins so my slot 6 starts at 00:30

[image]

So now:

Charge:  Set Grid Charge on, set SOC
Discharge: Set grid charge off, Set SOC. Set selling first mode

Idle: Set grid charge off, Set SOC. Set load first mode

I’ll be testing that today to see if the charge adjustments work too. 

I haven’t yet tried freeze charge etc.

bensebborn, 2024-04-20:
This is working well. HoldCharge is working as the SOC is correctly set to 100%

Only thing I’ve had to do is reset SOC to 12% when cancelling charge/discharge (ie going to idle). Have done this via the automation for now but probably should be in the code.

springfall2008, 2024-04-20:
> This is working well. HoldCharge is working as the SOC is correctly set to 100%
> 
> Only thing I’ve had to do is reset SOC to 12% when cancelling charge/discharge (ie going to idle). Have done this via the automation for now but probably should be in the code.

Why the reset, what goes wrong if you leave it as-is?

bensebborn, 2024-04-20:
Because that SOC field works as a target during charge and a limit during discharge. 

So when we’re not charging (so idle or discharging) the inverter will only discharge if the battery current SOC is above the SOC field. 

So currently if it was charging it might be set high. Then it goes idle and because the battery is then above that figure, it won’t allow any discharge. 

So golden rule in this inverter:

set the SOC field to target or 100% when charging (✅)

Set the SOC field to dischsrge limit or reserve when discharging (✅ when limit it set. ❓ Haven’t tested if force discharge without limit)

Set the SOC field to reserve when idle (❌)

Meatballs1, 2024-04-20:
I'm normally a fan of setting device schedules in automations as they are resilient to outages, but the SunSynk schedule is clunky. Having a 23.5 hour time slot seems a bit of a pain too as one half hour is going to be broken? 

Predbat/HA knows the SoC, so can we not just signal a change of mode when the target/reserve SoC is reached? Would this be a problem if Predbat is only checking every 5 minutes? 

I confess I need to get my head around a bit more of how predbat controls things.

Meatballs1, 2024-04-20:
With the `output_charge_control` set to `current`, there is no charge_rate to supply so it uses the default of 2600W:

https://github.com/springfall2008/batpred/blob/6885144b156bfd0291a7f4420e57a79e9fcdbd79/apps/predbat/predbat.py#L2192

bensebborn, 2024-04-20:
> I'm normally a fan of setting device schedules in automations as they are resilient to outages, but the SunSynk schedule is clunky. Having a 23.5 hour time slot seems a bit of a pain too as one half [...]

I guess that would work. However as I have a 10k inverter a lot can change in 5/10 mins! Potentially could go 1.6kW over/under

my plan was to use an automation to apply the SOC to the other slots. A bit clunky but would keep them all the same then the other 30 min slot would be correct. Would never change the times btw, just using it as a way of controlling SOC.

jacekowski, 2024-04-20:
I'm using node red as interface between predbat and sunsynk. I'm not adusting times at all, just adjusting work mode and settings for current slot as needed (while also doing some other house keeping on the inverter - such as forcing discharge early (i have 10kW of panels on 5kW inverter, so if i avoid hitting 100% i can export 5kW and anything above that charges the battery))

springfall2008, 2024-04-20:
> With the `output_charge_control` set to `current`, there is no charge_rate to supply so it uses the default of 2600W:
> 
> ```
> 2024-04-20 11:00:48.838506 INFO pred_bat: Found 1 inverters totals: min reserve 3.28 current reserve 0.66 soc_max 16.4 soc 14.6 charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 11.0 export limit 11.0 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
> ```
> 
> https://github.com/springfall2008/batpred/blob/6885144b156bfd0291a7f4420e57a79e9fcdbd79/apps/predbat/predbat.py#L2192
> 
> Do we need to set `Invertor_Max_Bat_Rate` or `Invertor_Max_Rate`?

You need to set 'battery_rate_max' in the apps.yaml

springfall2008, 2024-04-20:
Assuming has change enable time is still false this may fix your target soc % issue when not charging:

https://github.com/springfall2008/batpred/blob/sunsynk/apps/predbat/predbat.py

bensebborn, 2024-04-21:
Have popped this code in so will keep an eye. Thank you!
[image]
[image]

I have noticed that our plan is showing battery discharge down to tomorrow morning. Yet we have reserve set to 12%

Not sure any of these changes would have caused that… Is this expected?

bensebborn, 2024-04-21:
Looos like a bug actually as the SOC drops very quickly with low load (consider this is a 28.8kWh battery)

[image]

bensebborn, 2024-04-21:
So this is working, but it's setting my soc back down to 4% on idle, which is too low.

I'm not sure where this 4% is coming from?

I can see in logs: 

`2024-04-21 12:56:00.752788 INFO pred_bat: Inverter 0 with soc_max 28.8 kWh nominal_capacity 28.8 kWh battery rate raw 9300.0 w charge rate 9.3 kW discharge rate 9.3 kW battery_rate_min 0.0 w ac limit 18.0 kW export limit 18.0 kW reserve 4.0 % current_reserve 4.0 %`

Could you let me know what I need to change the 4% to 15% please?

(This is also probably the reason for #985 )

Meatballs1, 2024-04-21:
input_number.predbat_set_reserve_min

bensebborn, 2024-04-21:
That was already set to 13% but didn't change anything

However, I've found the following which wasn't documented (from what I could see) but was in the templates:

```
  battery_min_soc: 
   - 14
```

This has done the trick 👍

bensebborn, 2024-04-21:
Hmmm... so now my plan has updated and looks strange.

The min SOC is indeed 14% which is better.

However it just seems to have drained the battery with no plan to charge before the peak time..?

As you can see here, there's quite a few slots this afternoon around 13-15p

Then at peak time it goes up to 18-25p

Should it not be scheduling a charge at these lower rates to keep enough battery to cover teh peak slots?

[image]
[Predbat – Home Assistant - Plan.pdf](https://github.com/springfall2008/batpred/files/15052850/Predbat.Home.Assistant.-.Plan.pdf)

I have Best SOC Keep 3.4 KWh (about 12% of total battery capacity)
Reserve Min: 14%
Best SOC Min/Max: 0

Sorry probably my configuration being wrong, any help appreciated!
Plan attached as PSF

Meatballs1, 2024-04-21:
Still getting to grips with it too, but have you set the Rate Low Threshold?

Mine is just at 0 on agile.

bensebborn, 2024-04-21:
Rate Low Threshold is 0p

springfall2008, 2024-04-21:
Do you have control charge enabed in the mode?

bensebborn, 2024-04-21:
> Do you have control charge enabed in the mode?

It’s set to Control Charge and Discharge

bensebborn, 2024-04-21:
Not sure if related but checked the status column and seeing

“ Warn - Inverter 0 write to charge_limit failed”

I think this is since changing to current control instead of power

might not be related though.

bensebborn, 2024-04-23:
Hey @springfall2008 

The above issue hasn't happened again, so hard to recreate now. Will let you know if it happens again.

Could we get the changes on this Sunsynk branch merged into main so we can take advantage of any new releases?

Thanks for your work on this

closed 2024-04-27 by bensebborn

Meatballs1, 2024-04-28:
@bensebborn could you show example of your final config? And did you hit issues with the charge_limit timezone being to 1dp as mine fails to write if its not an integer?

springfall2008, 2024-05-03:
> Hey @springfall2008
> 
> The above issue hasn't happened again, so hard to recreate now. Will let you know if it happens again.
> 
> Could we get the changes on this Sunsynk branch merged into main so we can take advantage of any new releases?
> 
> Thanks for your work on this

This was merged onto MAIN and released

Can you help me with the documentation on Sunsynk by sharing the steps to create a working install?