springfall2008/batpred #967: Plan optimising minimal charge vs minimal cost?

Nilogax, 2024-04-18:
I've noticed that the plan is sometimes preferring to retain more charge during the discharge window (Octopus Flux) so that minimal charge is needed in the following charge window, but setting a forced discharge will give a lower overall plan cost.  I'm aware that these sort of things are often down to configuration so any pointers on how to set predbat to give the lowest cost option?

Example from today - this is the initial discharge/charge phase, predbat.best_metric final state is -45.26

[image]
[image]

And then after setting a forced disharge this is the result, predbat.best_metric final state is now -75.58

[image]
[image]

Log file for these two runs:

[predbat.log](https://github.com/springfall2008/batpred/files/15021207/predbat.log)

And config

[appsyaml.txt](https://github.com/springfall2008/batpred/files/15021266/appsyaml.txt)

springfall2008, 2024-04-21:
This is because the retained energy at the end of the plan has some value too.

You can change this by adjusting input_number.predbat_metric_battery_value_scaling, if you set it to 0 (not recommended) then the world ends (so to speak) at the end of the plan so dump everything...

Nilogax, 2024-04-21:
Ah of course - so an end of plan cost of £0.10 with a full battery is better than an alternative of -£0.05 with the battery at its reserve. Thanks.

closed 2024-04-21 by Nilogax