springfall2008/batpred #969: Luxpower inverter support

nickjf20, 2024-04-18:
MQTT bridge: https://github.com/celsworth/lxp-bridge

Preliminary app.yaml: https://github.com/nickjf20/lxp-predbat/blob/main/app.yaml

Consumption is currently not calculated by lxp-bridge. Add a helper like this: https://github.com/celsworth/lxp-bridge/issues/232

nickjf20, 2024-04-24:
Trying to add charge control:
https://github.com/nickjf20/lxp-predbat/blob/main/mod
https://github.com/nickjf20/lxp-predbat/blob/main/app.yaml

LXP bridge allows AC charge time slots but in the format "00:00-00:00". (Start - End). I'm unsure how to integrate this.

rossetf, 2024-06-04:
I ended up using some helpers as attributes to hold the values (set via predbat) I then used an automation to update the charge slot when either or the helpers changed values, not the most performant but then it doesn't need to be.  I no longer use the same LXP integration as you, so my memory is a little cloudy although I have similar issues in the one I am using and am again using helpers to set the final lxp attributes

nickjf20, 2024-06-05:
Hi Rossetf, are you using Guy's integration? Unfortunately it doesn't work with my BH dongle. 

Could you share an example of your helpers and I will see if I can translate it to lxpbridge?

rossetf, 2024-06-05:
Hi Nickjf20,  I am, I will try to put an explanation together for you later tonight.

rossetf, 2024-06-05:
I am Using Guy's integration, and so far so good

My apps.yml looks something like

```
 inverter_type: "LP"

  inverter:
    name: "LuxPower"
    has_rest_api: False
    has_mqtt_api: False
    output_charge_control: "none"
    has_charge_enable_time: True
    has_discharge_enable_time: True
    has_service_api: False
    has_target_soc: True
    has_reserve_soc: False
    has_timed_pause: False
    charge_time_format: "HH:MM:SS"
    charge_time_entity_is_option: False
    soc_units: "%"
    num_load_entities: 1
    has_ge_inverter_mode: False
    time_button_press: False
    clock_time_format: "%H:%M:%S"
    write_and_poll_sleep: 10
    has_time_window: True
    support_charge_freeze: False
    support_discharge_freeze: False
    has_idle_time: False
    can_span_midnight: True
```

with the time related stuff  as below
```
 scheduled_charge_enable:
    - switch.lux_ac_charge_enable
  scheduled_discharge_enable:
    - switch.lux_force_discharge_enable
  charge_start_time:
   - input_text.charge_start_time
  charge_end_time:
   - input_text.charge_end_time
  discharge_start_time:
   - input_text.discharge_start_time
  discharge_end_time:
   - input_text.discharge_end_time
``` 

As can be seen the helpers listed in for example charge_start_time are just text based helpers which batpred is more than happy with.

I then use an automations so when the helpers change the correct lux value is set up (in Guys these are type time)

```
alias: Charge Time Change - Start
description: ""
trigger:
  - platform: state
    entity_id:
      - input_text.charge_start_time
     
condition: []
action:
  - service: time.set_value
    metadata: {}
    data:
      time: "{{ states('input_text.charge_start_time')}}"
    target:
      entity_id: time.lux_ac_charge_start1
    enabled: true
mode: single
```

Which for you I think will look something like 
```
alias: Charge Time Change - Start or End
description: ""
trigger:
  - platform: state
    entity_id:
      - input_text.charge_start_time
      - input_text.charge_end_time
condition: []
action:
- service: text.set_value
    metadata: {}
    data: 
      option: "{{ states('input_text.charge_start_time')}}-{{ states('input_text.charge_end_time')}}"
    target:
      entity_id: text.ac_charge_timeslot_1
mode: single
```

You would also need something based off this for discharge times.

Hope that helps.