springfall2008/batpred #972: Car charging plans to charge above manual SOC

bensebborn, 2024-04-18:
**Describe the bug**
If you enter a current SOC > target SOC, a charging plan is created

**Expected behavior**
Charging plan isn't created

**Predbat version**
7.16.15

**Example:**
Limit set to 80%
Current SOC set to 81%
77kWh battery:

```

Cars 1 charging from battery True planned [True], charging_now [False] smart [True], max_price [15.0], plan_time ['07:45:00'], battery size [77.0], limit [61.6], rate [10.8]

Current Car SOC kWh: [62.370000000000005]

Plan car 0 charging from 62.370000000000005 to 61.6 with slots [{'start': 1230, 'end': 1260, 'average': 18.74}, {'start': 1260, 'end': 1290, 'average': 19.85}, {'start': 1290, 'end': 1350, 'average': 14.93}, {'start': 1350, 'end': 1380, 'average': 13.03}, {'start': 1380, 'end': 1410, 'average': 11.47}, {'start': 1410, 'end': 1440, 'average': 7.94}, {'start': 1440, 'end': 1470, 'average': 10.08}, {'start': 1470, 'end': 1500, 'average': 8.82}, {'start': 1500, 'end': 1530, 'average': 6.57}, {'start': 1530, 'end': 1560, 'average': 8.66}, {'start': 1560, 'end': 1590, 'average': 7.28}, {'start': 1590, 'end': 1650, 'average': 5.91}, {'start': 1650, 'end': 1680, 'average': 6.15}, {'start': 1680, 'end': 1710, 'average': 7.5}, {'start': 1710, 'end': 1770, 'average': 5.58}, {'start': 1770, 'end': 1800, 'average': 6.62}, {'start': 1800, 'end': 1830, 'average': 9.74}, {'start': 1830, 'end': 1860, 'average': 15.52}, {'start': 1860, 'end': 1890, 'average': 11.69}, {'start': 1890, 'end': 1920, 'average': 16.92}, {'start': 1920, 'end': 1980, 'average': 14.33}, {'start': 1980, 'end': 2010, 'average': 14.09}, {'start': 2010, 'end': 2070, 'average': 10.02}, {'start': 2070, 'end': 2100, 'average': 7.76}, {'start': 2100, 'end': 2130, 'average': 9.46}, {'start': 2130, 'end': 2160, 'average': 8.18}, {'start': 2160, 'end': 2220, 'average': 6.65}, {'start': 2220, 'end': 2250, 'average': 5.73}, {'start': 2250, 'end': 2280, 'average': 4.69}, {'start': 2280, 'end': 2310, 'average': 6.17}, {'start': 2310, 'end': 2370, 'average': 5.11}, {'start': 2370, 'end': 2400, 'average': 6.62}, {'start': 2400, 'end': 2430, 'average': 20.13}, {'start': 2430, 'end': 2460, 'average': 26.68}, {'start': 2460, 'end': 2490, 'average': 23.87}, {'start': 2490, 'end': 2550, 'average': 26.93}, {'start': 2580, 'end': 2610, 'average': 19.18}, {'start': 2610, 'end': 2670, 'average': 19.94}, {'start': 2670, 'end': 2700, 'average': 19.08}, {'start': 2700, 'end': 2730, 'average': 20.46}, {'start': 2730, 'end': 2760, 'average': 17.64}, {'start': 2760, 'end': 2790, 'average': 17.0}, {'start': 2790, 'end': 2820, 'average': 14.55}, {'start': 2820, 'end': 2850, 'average': 11.47}, {'start': 2850, 'end': 2880, 'average': 7.94}, {'start': 2880, 'end': 2910, 'average': 10.08}, {'start': 2910, 'end': 2940, 'average': 8.82}, {'start': 2940, 'end': 2970, 'average': 6.57}, {'start': 2970, 'end': 3000, 'average': 8.66}, {'start': 3000, 'end': 3030, 'average': 7.28}, {'start': 3030, 'end': 3090, 'average': 5.91}, {'start': 3090, 'end': 3120, 'average': 6.15}, {'start': 3120, 'end': 3150, 'average': 7.5}, {'start': 3150, 'end': 3210, 'average': 5.58}, {'start': 3210, 'end': 3240, 'average': 6.62}, {'start': 3240, 'end': 3270, 'average': 9.74}, {'start': 3270, 'end': 3300, 'average': 15.52}, {'start': 3300, 'end': 3330, 'average': 11.69}, {'start': 3330, 'end': 3360, 'average': 16.92}, {'start': 3360, 'end': 3420, 'average': 14.33}, {'start': 3420, 'end': 3450, 'average': 14.09}, {'start': 3450, 'end': 3510, 'average': 10.02}, {'start': 3510, 'end': 3540, 'average': 7.76}, {'start': 3540, 'end': 3570, 'average': 9.46}, {'start': 3570, 'end': 3600, 'average': 8.18}, {'start': 3600, 'end': 3660, 'average': 6.65}, {'start': 3660, 'end': 3690, 'average': 5.73}, {'start': 3690, 'end': 3720, 'average': 4.69}, {'start': 3720, 'end': 3750, 'average': 6.17}, {'start': 3750, 'end': 3810, 'average': 5.11}, {'start': 3810, 'end': 3840, 'average': 6.62}, {'start': 3840, 'end': 3870, 'average': 20.13}, {'start': 3870, 'end': 3900, 'average': 26.68}, {'start': 3900, 'end': 3930, 'average': 23.87}, {'start': 3930, 'end': 3990, 'average': 26.93}, {'start': 4020, 'end': 4050, 'average': 19.18}, {'start': 4050, 'end': 4110, 'average': 19.94}, {'start': 4110, 'end': 4140, 'average': 19.08}] from soc 62.370000000000005 to 61.6 ready by 07:45:00
2024-04-18 20:50:22.303362 INFO pred_bat: 
```

bensebborn, 2024-04-18:
Confirmed it doesnt actually set the charging slots it just documents them - potentially this could be aborted earlier to save some caculations, if `current SOC > target SOC` ?

springfall2008, 2024-04-21:
I think it's fine it will do nothing but log the case?

bensebborn, 2024-04-21:
Cool just wondered if it would save some CPU

closed 2024-04-21 by bensebborn