springfall2008/batpred #975: Repeatedly charging and discharging the battery

gcoan, 2024-04-20:
**Describe the bug**
Noticed this morning that predbat is letting the battery solar charge then force discharge, then repeat.  This can't be efficient

7.16.16 and secondary order slots OFF
[image]

Thought it must be 7.16.16 causing it so reverted to 7.16.15 with secondary order slots OFF, much the same:
[image]

Tried turning secondary order slots ON with 7.16.15 and its still similar
[image]

**Expected behavior**
Either hold the battery level low or let it charge and stay full to the peak would involve less battery conversion losses 

Previous plans with secondary order slots off have been fine

springfall2008, 2024-04-20:
This is odd, can you try the updates on main also and see if it makes any difference?

gcoan, 2024-04-20:
I had put it back to 7.16.16 and still trying the charge and discharging
[image]

Eventually with a load of force idles I managed to stop it doing this 
![image](https://github.com/springfall2008/batpred/assets/142018870/b13f121d-841a-4776-85ae-193fc290200f

Will try main now

SwiftRR, 2024-04-20:
I am on 7.16.16 with secondary order slots ON
This is what I see, running on until sun sets.
[image]

Rob

gcoan, 2024-04-20:
Looks like @SwiftRR is having the same behaviour 

I tried main and cancelled the force idles , same thing 
[image]

springfall2008, 2024-04-20:
Do you have the conversion losses set correctly?

bensebborn, 2024-04-20:
I have lots of solar export today and mine is correctly holding charge and exporting the solar directly so doesn’t seem to be affecting me

SwiftRR, 2024-04-20:
I have an GE AC coupled inverter.
Hybrid inverter switch is OFF
Conversion losses are 4/4/4

On today’s snapshots, daytime charging of battery mainly by solar with ’popup from grid’.

I am happy to tweak settings to help test the issue.

Rob

Rob Ritchie
***@***.***
----------------------------

gcoan, 2024-04-20:
> Do you have the conversion losses set correctly?

My loss figures have been the same for ages, just checked and they are:
inverter 4%
battery charge 5.5%
battery discharge 5%

hybrid=true

battery metric cycle cost=0

metric min improvement=0

metric min improvement discharge=0.2

I don't think I have seen this behaviour from predbat before.

Tomorrow is doing similar, not quite as extreme but still planning repeated charge then discharge when a hold charge and export the restwould normally be more efficient:
[image]

Can provide the logfile if you would like to see it

SwiftRR, 2024-04-20:
For comparison with Geoffrey's times for Sunday, this is my predbat plan:
[image]

Tariff Agile import, 15p fixed export.

I am in Yorkshire region and my import rates are about 0.6p lower than Geoffrey's.

Conversion losses 4/4/4

hybrid OFF

Secondary order slots=ON
 
metric battery cycle cost=0

metric min improvement=0

metric min improvement discharge=0.1

Nordpool ON
------------------------
Rob

gcoan, 2024-04-20:
@springfall2008  think this is probably related to #924 and #861 both of which are about the behaviour of input_number.predbat_min_improvement_discharge that doesn't seem to be working properly

If I progressively increase the min_improvement_discharge I eventually get to a plan that "looks right" and isn't charging and discharging the battery to make a profit

0.1:
[image]

0.2:
[image]

0.5:
[image]

1:
[image]

3:
[image]

6:
[image]

gcoan, 2024-04-21:
Had another look at it this morning.  Using the version on b2663b8 on main

Secondary order slots off

Initially metric_min_discharge_improvement was 6p and the plan looked OK:
[image] 

changed back to 0.3p and a series of discharges were introduced:
[image]

and 1p, slightly less discharging but similar:
[image]

Looking at one of these in a bit more detail, discharge and export is at an effective rate of 13.68p vs just keeping the charge at 100% and exporting the solar.
The first (discharge and solar recharge) incurs the battery discharge loss (5%), the inverter loss (4%) and on recharge the battery charge loss (5.5%) and a second set of inverter loss (4%). Compare to just exporting solar straight away which will only incur the 4% inverter loss and should be more profitable.

Increase metric_min_discharge_improvement to 5p and then 6p:
[image]
[image]

In both of these predbat appears to be discharging and exporting at 13.68p effective rate to then import and charge at 15.77p effective rate?

Increasing metric_min_discharge_improvement to 8p stops the discharge and recharge which is the correct plan but is having to artificially constrain using the battery.  I would rather be using the battery if it were truly profitable to do so
[image]

SwiftRR, 2024-04-21:
Interesting. For comparison, here's mine with 
metric_min_discharge_improvement at default of 0.1 p  !!!!!!
calculate discharge on charge slots TRUE
combine charge slots TRUE
combine discharge slots  FALSE
calculate secondary order slots TRUE
... and running with losses, 4/4/4 on hybrid inverter FALSE
Looks much steadier.
[image]

Rob

gcoan, 2024-04-21:
> metric_min_discharge_improvement at default of 0.1 p !!!!!!
> calculate discharge on charge slots TRUE
> combine charge slots TRUE
> combine discharge slots FALSE
> calculate secondary order slots TRUE
> ... and running with losses, 4/4/4 on hybrid inverter FALSE

And my settings are:

metric_min_discharge_improvement at 8p to stop the discharging
calculate discharge on charge slots FALSE
combine charge slots FALSE
combine discharge slots FALSE
calculate secondary order slots FALSE
Losses, 5.5/5/4 on hybrid inverter TRUE

Also
Calculate full second pass (slower), Calculate plan faster (less accurate), Calculate tweak second pass all FALSE
Metric battery cycle cost 0p

SwiftRR, 2024-04-21:
I also have:
Calculate full second pass (slower), Calculate plan faster (less accurate), Calculate tweak second pass all FALSE
Metric battery cycle cost 0p

My plan seems to be leaving itself.
This one has rumbled on for a while now.
Is it 'Over to Trefor time'?

Rob

gcoan, 2024-04-21:
I think it is one for @springfall2008 

There's the other two issues also about min discharge improvement not seemingly working "quite right"