springfall2008/batpred #980: Allow for different types of optimisation

jm8-dev, 2024-04-20:
**Is your feature request related to a problem? Please describe.**
- Predbat is great - really been enjoying using it! I would like to request that predbat has the feature to optimise for different scenarios instead of just cost. I would like to request that predbat can optionally optimise for self-consumption of solar, and grid green-ness. 

**Describe the solution you'd like**
As we move into the summer, predbat wants to charge my battery up cheaply (usually in the early morning), and then export as much of my solar as possible. Although this usually minimises my effective energy cost (on Octopus Agile tariffs), I would like to self-consume my solar as much as possible - for example, letting the solar panels charge up the battery in the early morning. Even if this isn't the most cost effective idea, it is likely the most 'green' energy I can use.

It would be great to have a new 'select' input or 'boolean toggles' where I could choose between the following algorithim optimisations:
1) Optimise for self-consumption of solar and battery power
2) Optimise for buying grid energy when it is 'greenest'
3) Optimise for miniming cost of energy

If these were boolean toggles for the three options, I could then pick and choose my desired 'blend' of optimisations. If it is a select input that I choose between the three, then I was thinking that self-consumption of solar would be priority but the algorithm would still buy energy when it is greenest, and cheapest, whilst keeping the battery charged for the evening (i.e. a rainy summer day).

**Describe alternatives you've considered**
I've considered using expert mode to tune the algorithm for the above, but ideally this would be a feature of predbat that I can use to pick and choose some defaults. This would make it easier when moving from winter, into spring and summer. Please let me know if you have ideas on how I could use expert mode to achieve this though!

**Additional context**
N/A

springfall2008, 2024-04-21:
I like the idea :)

Do you know how you can determine 'greenest' on the grid inside HA?

jm8-dev, 2024-04-21:
Yes I thought about using Octopus' greenness forecast - currently I'm using the [National Grid's Carbon Intensity API ](https://carbonintensity.org.uk/)which publishes regional half-hourly carbon intensity estimates, which could work nicely with Agile.

I've integrated into HA using this guide: https://www.speaktothegeek.co.uk/2023/05/local-uk-grid-carbon-intensity-in-home-assistant/

springfall2008, 2024-04-21:
I found this integration that might help: https://github.com/jfparis/sensor.carbon_intensity_uk

mpartington, 2024-04-21:
> You can use the octopus greenness forecast for overnight periods and a longer term trend direction ![image](https://private-user-images.githubusercontent.com/142018870/324248396-9a73b5cd-3fb8-464a-bd46-a5aacc923304.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM3MzYxMTUsIm5iZiI6MTcxMzczNTgxNSwicGF0aCI6Ii8xNDIwMTg4NzAvMzI0MjQ4Mzk2LTlhNzNiNWNkLTNmYjgtNDY0YS1iZDQ2LWE1YWFjYzkyMzMwNC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjFUMjE0MzM1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODJkMTRjOTY5MzQyYmEyYTA5M2MwYjRlYTk3ZWRlZjRiY2ZlZTExNTg3MTAzOTM1MWE4MTEzNzJhYjMwYjk3NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.aCNx5rYAKYM6-I7mS_3ervNaBoaz65tSc6ek7fLx5yg)

Is it easy to plot this? I had a quick look at bottle cap Dave's readme and wasn't sure what to plot.

gcoan, 2024-04-22:
> Is it easy to plot this? I had a quick look at bottle cap Dave's readme and wasn't sure what to plot.

Apex chart for the greenness forecast:

```yaml
type: custom:apexcharts-card
experimental:
  color_threshold: true
header:
  show: true
  title: Greenness Forecast
graph_span: 7d
span:
  start: hour
apex_config:
  chart:
    height: 135
  stroke:
    width: 4
  xaxis:
    labels:
      format: ddd
series:
  - entity: sensor.octopus_energy_a_xxxx_greenness_forecast_current_index
    name: Forecast
    show:
      in_header: false
      legend_value: false
    color_threshold:
      - value: 0
        color: red
      - value: 40
        color: orange
      - value: 80
        color: green
    data_generator: |
      return entity.attributes.forecast.slice(0, 7).map((entry) => {
      return [new Date(entry.start), entry.greenness_score];
        });
```

springfall2008, 2024-04-22:
The forecast part is now on main using the HA integration for national grid.

Optimisation for carbon will be added soon