springfall2008/batpred #981: Factor in cost of cycles/battery wear and tear

Meatballs1, 2024-04-20:
Batteries have a finite lifetime, and generally guaranteed for x cycles. One example metric of the cost of charging/discharging a battery is:

`Cost / (capacity * DoD * max cycles * efficiency loss)` which works out at about 6p/kWh for my setup. 

There's probably other ways to estimate this, and I suspect the batteries will carry on for a while after their charge cycles just at reduced capacity, however, I'm not sure batpred takes this into account in the modelling?

**Describe the solution you'd like**
A configurable input value that adds a cost/kWh to charging.

RobinCu, 2024-04-20:
Yes, this feature exists in expert mode (input_number.predbat_metric_battery_cycle).

RobinCu, 2024-04-20:
More info here: https://springfall2008.github.io/batpred/customisation/

Meatballs1, 2024-04-20:
Perfect thank you

closed 2024-04-20 by Meatballs1