springfall2008/batpred #997: Error: Exception raised with new 7.17.0

njhcarroll, 2024-04-23:
**Describe the bug**
Have followed requirements to install new carbon add on into custom_components and installed integration.  All working with new sensors reporting values.  However, when restarting HA and appdaemon get an exception raised error
**Expected behavior**
Was expecting Probate to strat without problems

**Predbat version**

7.17.0

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
Appdaemon Log:  
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.2  (aarch64 / raspberrypi5-64)
 Home Assistant Core: 2024.4.4
 Home Assistant Supervisor: 2024.04.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[23:01:32] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-04-23 23:01:59.601958 WARNING pred_bat: ------------------------------------------------------------
2024-04-23 23:01:59.603753 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-04-23 23:01:59.604172 WARNING pred_bat: Worker Ags: {'id': '9dd5e8d31f7f4f068ce52970ab555bc8', 'name': 'pred_bat', 'objectid': 'e2fa85fc40fd465c85ef96d674a21572', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f83ead210>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-04-23 23:01:59.604477 WARNING pred_bat: ------------------------------------------------------------
2024-04-23 23:01:59.606179 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 13614, in update_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 13609, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12598, in update_pred
    self.fetch_sensor_data()
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11815, in fetch_sensor_data
    self.carbon_intensity, self.carbon_history = self.fetch_carbon_intensity(entity_id)
                                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11602, in fetch_carbon_intensity
    carbon_history = self.minute_data_import_export(self.now_utc, "predbat.carbon_now", required_unit="g/kWh", increment=False, smoothing=False)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4890, in minute_data_import_export
    raise ValueError
ValueError

2024-04-23 23:01:59.607276 WARNING pred_bat:

**Log file**
[appdaemon-2.log](https://github.com/springfall2008/batpred/files/15083546/appdaemon-2.log)

njhcarroll, 2024-04-23:
Saw that the exception was caused by a lack of history for carbon so switched it off.  That seemed to sort out the exception due to the carbon data missing.  However, am now getting a new exception preventing startup.  pred_bat: ERROR: Exception raised unsupported operand type(s) for +: 'NoneType' and 'float'

[appdaemon-3.log](https://github.com/springfall2008/batpred/files/15083691/appdaemon-3.log)

Speshman, 2024-04-23:
I had the same issue when I switched carbon on. Switched it off and am also getting the error njhcarroll mentioned of

ERROR: Exception raised unsupported operand type(s) for +: 'NoneType' and 'float'

Speshman, 2024-04-23:
[image]

njhcarroll, 2024-04-23:
Hi @Speshman - how do you revert to the previous version?

Speshman, 2024-04-23:
> Hi @Speshman - how do you revert to the previous version?

[image]

I use the toggle on the default dashboard

njhcarroll, 2024-04-23:
I'm missing those options on my loaded dashboard...

Speshman, 2024-04-23:
Try addng a new dashboard and set it to default dashboard. don't edit it and it should make a massive dashboard with absoloutely everything in it.  Probably a better way of doing it but I cant think of a quicker one

njhcarroll, 2024-04-23:
Thanks!  Will give it a go

njhcarroll, 2024-04-23:
Seemed to solve the problem by setting dashboard expert mode switch to off.  Everything suddenly working again.  Go figure!  Going to bed now before I do any more damage :)

DrJohnM61, 2024-04-23:
> Thanks! Will give it a go

And turn off auto update

mpartington, 2024-04-24:
Guessing likely same as in other logs, but these are the error I'm getting when turning carbon enable on

2024-04-24 07:18:40.125272 WARNING pred_bat: ------------------------------------------------------------
2024-04-24 07:18:40.127061 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-04-24 07:18:40.127736 WARNING pred_bat: Worker Ags: {'id': '789e91ba65cf43a184f459d01e679c38', 'name': 'pred_bat', 'objectid': 'f8215874904143ed9b5580c2a4365ceb', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7f9ec15150>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-04-24 07:18:40.128230 WARNING pred_bat: ------------------------------------------------------------
2024-04-24 07:18:40.140605 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 13614, in update_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 13609, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12598, in update_pred
    self.fetch_sensor_data()
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11815, in fetch_sensor_data
    self.carbon_intensity, self.carbon_history = self.fetch_carbon_intensity(entity_id)
                                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11602, in fetch_carbon_intensity
    carbon_history = self.minute_data_import_export(self.now_utc, "predbat.carbon_now", required_unit="g/kWh", increment=False, smoothing=False)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 4890, in minute_data_import_export
    raise ValueError
ValueError

2024-04-24 07:18:40.141148 WARNING pred_bat: ------------------------------------------------------------

Speshman, 2024-04-24:
> Fix now on main for this v17.17.1

Still getting this error

ERROR: Exception raised unsupported operand type(s) for +: 'NoneType' and 'float'

Says plan is invalid then stops with that error

[predbatlog.txt](https://github.com/springfall2008/batpred/files/15092453/predbatlog.txt)

Speshman, 2024-04-24:
If i turn off expert mode as noted by someone else then the error changes to

ERROR: Exception raised type NoneType doesn't define __round__ method

alfwro13, 2024-04-24:
same here on 7.17.1 and still getting that error:
ERROR: Exception raised unsupported operand type(s) for +: 'NoneType' and 'float'

Speshman, 2024-04-24:
7.17.2 seems to have fixed it for me

njhcarroll, 2024-04-24:
And for me as well.  Thank you