springfall2008/batpred #1000: Goodwe inverter support

thibaultmol, 2024-04-24:
Do note: the defaut Goodwe integration on Home Assistant does not support manually setting charge/discharge. Only the experimental version of that integration does https://github.com/mletenay/home-assistant-goodwe-inverter

springfall2008, 2025-01-01:
If you are willing to try it then it should be possible to configure

Brkie, 2025-02-06:
Also intressed in support for goodwe, tell me if I can somehow help to make it possible

gcoan, 2025-02-06:
> Also intressed in support for goodwe, tell me if I can somehow help to make it possible 

If you start from the template apps.yaml you'll need to work out what the input controls and services are to control your inverter/battery - to start it charging, stop charging, determine battery level, etc

see https://springfall2008.github.io/batpred/inverter-setup/#i-want-to-add-an-unsupported-inverter-to-predbat

If you get something working then we can incorporate it into the documentation.

gcoan, 2026-02-18:
> I've done that and I have a working setup right now! 🎉

haha, its a matter of knowing what to look for, even though you provided no logs and little information .....

let us know how you get on

I did notice you made a lot of use of secret config settings, is there a reason for that?  Normally we just code everything in apps.yaml and people can then adjust for their inverter if its different from the authors inverter.
Would prefer you do that when you do share the working config as it makes it easier for others to reuse

gcoan, 2026-02-23:
They are different parts that work together 

Reserve is an entry in apps.yaml with the inverter sensor name for the lower discharge Soc limit, to discharge down to 

Battery min Soc is an input number entity in HA you configure to the minimum value to be set in the reserve entity, or the lowest Soc level that your inverter can access. So I set my minimum to 4 as the lowest 4% isn't usable. For some inverters it's 10%

gcoan, 2026-02-26:
> Hmm, the reserve can also reference a number (writeable), right? And predbat will update it.

correct, that is how reserve works in predbat

> That got me thinking about the "Hold Charge" implementation. If my thinking is correct, then when predbat wants to enable "Hold Charge" then it will update this reserve value as the inverter would guarantee that the battery will not drain below that value, but it should not trigger charge_start_service as we are not charging and because that, in my case, causes the inverter to not discharge from the battery.
>
> What do you think? Am I missing something?

I'd have to check the code to be 100% sure but predbat expects the reserve to hold the battery level. Does your inverter have target SoC for charge-to?  I'm not 100% sure but I think predbat would issue a start charge though to charge to the hold charge target.
I don't get a lot of hold charges on my plan but when I do, lie this morning, I see the inverter continuing to charge above the target:
[image]

In my case I think this is the inverter not doing what predbat instructs, so I just live with it.
 
> I believe this is where hold charge is being handled, however unless I'm missing something it's not correctly handling the scenario above where my battery SoC was 97% but the hold charge was set to 86% as the difference from SoC was > 1%.

86% is the minimum level the battery should hold at as far as predbat is concerned (to meet the plan).

So what happened with your inverter when predbat issued a hold charge with reserve lower than current soc?

bpinto, 2026-02-26:
@gcoan This is what I had in mind, but I ran the test suite and a lot of tests fail with this, I have tried to understand the tests but there are a lot of abstractions that make it hard for newcomers to understand. 😅 

```diff
> diff --git a/apps/predbat/execute.py b/apps/predbat/execute.py
index e57baa6..a43d2ef 100644
--- a/apps/predbat/execute.py
+++ b/apps/predbat/execute.py
@@ -201,7 +201,7 @@ class Execute:
                                 status = "Hold charging"
                                 self.log("Inverter {} Hold charging as SoC {}% is above target SoC {}% set_discharge_during_charge {}".format(inverter.id, self.soc_percent, dp0(target_soc), self.set_discharge_during_charge))
 
-                                if (target_soc < 100.0) and (abs(inverter.soc_percent - target_soc) <= 1.0):
+                                if (target_soc < 100.0) and (inverter.soc_percent >= target_soc):
                                     # If we are within 1% of the target but not at 100% then we can hold charge
                                     # otherwise keep charging enabled
                                     if self.set_soc_enable and ((self.set_reserve_enable and self.set_reserve_hold and inverter.reserve_max >= inverter.soc_percent) or inverter.inv_has_timed_pause):
```

For what is worth, the logic used to be similar to what I'm proposing but it was changed in https://github.com/springfall2008/batpred/pull/1655

dmshimself, 2026-09-20:
Yes I'd be interested in finding out the current status for goodwe inverters.  I'm happy to have a go if there is a sensible config to start with?

dmacwalter, 2026-09-20:
> Yes I'd be interested in finding out the current status for goodwe inverters.  I'm happy to have a go if there is a sensible config to start with?

@dmshimself @RampouchCZ — following up my Feb comment about EMS support. I've now got a GoodWe ET hybrid under full Predbat control (charge, forced export, stop) running local-only, no cloud. Here's the whole working config to start from.

This is the EMS path, so it needs [mletenay/home-assistant-goodwe-inverter](https://github.com/mletenay/home-assistant-goodwe-inverter) rather than the core integration, and an inverter that exposes `select.goodwe_ems_mode` and `number.goodwe_ems_power_limit`. Different from @bpinto's setup above, which uses the simpler `eco_charge` / `general` operation modes — if your inverter has no EMS block, follow his instead.

Setup: GoodWe ET hybrid, 22.4 kWh battery, 10 kW inverter, 5.4 kW export limit, Predbat v9.0.3 as an HA add-on.

## Approach

Predbat writes the charge and discharge rates into two `input_number` helpers rather than to the hardware, and scripts do the actual register writes. That leaves exactly one writer on the EMS registers and avoids a race between Predbat and the scripts.

Create two helpers, 0 to your `battery_rate_max` in watts, step 100, box mode:

- `input_number.predbat_charge_rate`
- `input_number.predbat_discharge_rate`

The ceiling must be at least `battery_rate_max`. Mine was capped at 8000 with `battery_rate_max` at 9600 and Predbat's write failed silently.

## apps.yaml

```yaml
  inverter_type: GW
  inverter:
    name: GoodWe ET (custom)
    has_rest_api: false
    has_mqtt_api: false
    has_service_api: true
    output_charge_control: power
    has_charge_enable_time: false
    has_discharge_enable_time: false
    has_target_soc: false
    has_reserve_soc: true
    charge_time_format: S
    charge_time_entity_is_option: false
    soc_units: '%'
    num_load_entities: 1
    has_ge_inverter_mode: false
    has_fox_inverter_mode: false
    time_button_press: false
    clock_time_format: '%d-%m-%y %H:%M:%S'
    write_and_poll_sleep: 2
    has_time_window: false
    support_charge_freeze: true
    support_discharge_freeze: true

  charge_rate:
  - input_number.predbat_charge_rate
  discharge_rate:
  - input_number.predbat_discharge_rate

  charge_start_service:
    service: script.turn_on
    entity_id: script.predbat_charge
    repeat: true
  charge_stop_service:
    service: script.turn_on
    entity_id: script.predbat_stop
    repeat: false
  discharge_start_service:
    service: script.turn_on
    entity_id: script.predbat_export
    repeat: true
  discharge_stop_service:
    service: script.turn_on
    entity_id: script.predbat_stop
    repeat: false
```

`repeat: true` on the start services matters — Predbat re-calls them each cycle, which is how the export setpoint tracks changing PV.
[... 123 lines cut ...]
      value: 10000
  - action: select.select_option
    target:
      entity_id: select.goodwe_ems_mode
    data:
      option: auto
```

## EMS behaviour worth knowing

Write order matters. The power limit has to be written before the mode, not after — write the mode first and the setpoint does not stick.

The power limit is direction-agnostic magnitude, and the same mode means different things depending on it. At limit 0, `battery_standby` is a genuine hold. At limit 10000, it behaves as a charge request.

Put service keys at the top level, not under a nested `data:` block. Predbat flattens the keys under a service entry into the service data, so a nested block is passed through as a literal key named `data` and HA rejects it with `{'code': 'invalid_format', 'message': "not a valid option at 'data'"}`. The error names the unexpected key, not the value, which is misleading.

Quote every `HH:MM:SS` in apps.yaml. Unquoted times parse as sexagesimal integers under YAML 1.1, so `11:00:00` becomes 39600 and you get `Error: Exception raised 'int' object has no attribute 'count'` out of `basic_rates`. Values starting with a zero survive as strings because the leading zero excludes them from the int resolver, which is why the first `rates_import` entry looks fine and nothing else does.

## Not solved

Freeze export. The EMS mode list (auto, charge_pv, discharge_pv, import_ac, export_ac, conserve, off_grid, battery_standby, buy_power, sell_power, charge_battery, discharge_battery) has no mode meaning "don't charge but do discharge to cover house load", and GoodWe isn't a feed-in-first inverter, so a freeze clips surplus PV. If anyone has characterised `conserve` or `discharge_pv` on an ET I'd be glad to hear it.

`set_reserve_service` isn't wired, so Predbat can't move the reserve. The GoodWe's own on-grid and backup depth-of-discharge settings are the only floors on house discharge.

Happy to answer questions or test changes on real hardware.

dmshimself, 2026-09-20:
Many thanks for that.  I have an older ET hybrid (GW10KL-ET), so I don't think it will have EMS support.  But once I have some time, I'll start with bpinto's config as a starting point as you suggest.  I'm already using the HACS experimental integration, so I'm in pretty shape there too.  

Many thanks for the feedback and ideas.  Very much appreciated

dmacwalter, 2026-09-20:
Check for a firmware update via SolarGo. I only noticed EMS after a firmware update via SolarGo.

Craig-Burgess-UK, 2026-10-07:
We are currently specifying a new UK three-phase system and are considering the newer GoodWe 20 kW ET-LV platform (GW20K-ET-L-G10 / current UK equivalent) because its hardware is a very strong fit for a large battery/PV system.

The proposed system is roughly:
- 3 x 20 kW GoodWe ET-LV inverters
- ~79 kWp PV
- ~145 kWh 48 V battery storage
- dynamic tariffs / export optimisation through Predbat
- Home Assistant local control preferred; no dependency on cloud control

I can see from this thread that GoodWe can now be made to work with Predbat using custom apps.yaml/scripts and the newer EMS controls, which is encouraging.

Before we choose the inverter platform, could you clarify the likely direction of travel for GoodWe support in Predbat?

In particular:
1. Is there any plan for a more native/standard GoodWe configuration (similar in ease of setup to Solis/LuxPower/Sunsynk) rather than a bespoke unsupported-inverter configuration?
2. Is the newer ET-LV/EMS interface expected to make reliable charge/discharge/export control easier to support?
3. Is there any rough timescale or milestone for fuller GoodWe support?
4. Are there any known limitations that would make GoodWe a poor choice for a new system where Predbat control is a key design requirement?

We are not looking to install immediately, so if fuller support is likely within the next year that would materially affect the inverter decision.

Thanks — this is a very useful project and we are trying to choose the hardware around the control layer rather than retrofit the control strategy afterwards.

dmacwalter, 2026-10-09:
@Craig-Burgess-UK I'm a user rather than a maintainer, so the roadmap and timescale questions are for @gcoan / @springfall2008. Here's what I can say from running a GoodWe under Predbat control.

As of current main there's no GoodWe template in `templates/` and no native GoodWe inverter type. It's the unsupported-inverter route: you define the inverter capabilities yourself and drive it through `charge_start_service` / `discharge_start_service` scripts, as below.

The EMS interface does make control possible. `charge_battery` and `discharge_battery` with `ems_power_limit` hit setpoints within about 2% once you write them in the right order. But I wouldn't call it easy. The mode names don't mean what they say, a couple of modes ignore the grid export limit, and it took a few weeks of testing on real hardware to get a script I trust (details below).

Points I'd check before committing to your design:

- EMS needs firmware support and the custom (HACS) `mletenay` integration, not the core one. I only got EMS after a SolarGo firmware update. Confirm your exact model and firmware expose `select.goodwe_ems_mode` and `number.goodwe_ems_power_limit` before buying.
- I run a single inverter. I haven't tested EMS on a parallel group of three. Whether you write to each unit or to the master, and how the units share the setpoint, would need confirming, along with how it maps onto Predbat's multi-inverter config.
- Thresholds in my scripts are tuned to my site (5.4 kW export cap, ~10 kW AC, 9.6 kW battery rate). Nothing transfers unchanged to a 60 kW three-phase system.
- Not solved: there's no EMS mode for freeze export, and `sell_power` will exceed the export limit unless you guard it.

If Predbat control is a hard requirement and you want template-level setup, a natively supported inverter is the lower-risk choice today. GoodWe works if you're prepared to own the scripts and test on your own hardware.

---

Update on the GoodWe EMS config I posted earlier. It has been running Predbat in full control (charge, export, freeze) since late September, and a few things in that first post turned out to be wrong or incomplete. Current versions are below.

Hardware is unchanged: GoodWe GW9.99K-EHB, 22.4 kWh battery, 16.5 kWp PV, 5.4 kW export limit, ~10 kW inverter AC output. The thresholds in the export script are specific to those numbers, so adjust them for your site.

## What changed

1. Freeze charge now works. `battery_standby` with the power limit at 0, written before the mode, is a true hold: no charge, no discharge, and the house load is imported from the grid. With the limit left at 10000, the same mode behaves as a charge request and pulls from the grid. I point both `charge_freeze_service` and `discharge_freeze_service` at the freeze script below. For discharge freeze that's only an approximation, since it holds the battery rather than letting it cover house load (see Still not solved).

2. The export script needed much more than a clamp. On this inverter the EMS modes behave very differently during the day:
   - `discharge_battery` throttles PV, so it is only safe when PV is negligible (night/dusk).
   - `discharge_pv` respects the grid export limit and never charges the battery, but curtails PV to house load + export cap. In strong sun it hides the real surplus.
   - `sell_power` exports to the setpoint and sends the rest of the surplus to the battery, but it does NOT respect the grid export limit. I saw 8.5 kW export with the battery full.
   - `auto` respects the export limit and charges the battery first.
   - `battery_standby` at 0 curtailed PV and imported in full sun, so I don't use it for export.

   The script now picks the mode on each Predbat call from a first-match decision table: `discharge_battery` at night, `sell_power` when the surplus is clearly above the cap (or PV is being clipped), `auto` near full SOC or when the inverter is at its AC ceiling, and `discharge_pv` otherwise.

3. Because `sell_power` ignores the export limit, I also run a watchdog automation. It re-runs the export script and notifies me if export stays over 5.5 kW for 30 s while in `sell_power`.

4. The charge and stop scripts are unchanged from the first post. The write order (auto -> limit -> mode) is still the single most important thing. Write the mode first and the setpoint does not stick.

## apps.yaml additions

```yaml
  charge_freeze_service:
    service: script.turn_on
    entity_id: script.predbat_freeze
  discharge_freeze_service:
    service: script.turn_on
    entity_id: script.predbat_freeze
```

## Freeze script

```yaml
predbat_freeze:
  alias: Predbat - freeze battery
  mode: single
  sequence:
  - alias: Limit 0 BEFORE mode - this makes battery_standby a hold, not a charge request
    action: number.set_value
    target:
      entity_id: number.goodwe_ems_power_limit
    data:
      value: 0
  - action: select.select_option
    target:
      entity_id: select.goodwe_ems_mode
    data:
      option: battery_standby
```

(Mine has one extra branch that releases to auto while my EV battery-boost switch is on. That is site-specific, so I've left it out here.)
[... 88 lines cut ...]
    data:
      value: "{{ xset | int }}"
  - action: select.select_option
    target:
      entity_id: select.goodwe_ems_mode
    data:
      option: "{{ target_mode }}"
```

What the rules are for, in order:
1. Night: `discharge_battery`, which is setpoint-accurate, but only when PV is negligible because it throttles PV.
[... cut ...]
8. PV clipped, or surplus clearly above the cap: `sell_power`.
9. Otherwise: `discharge_pv`.

Known trade-off: decisions use instantaneous readings on each ~5 min Predbat call, so broken cloud can pick a sub-optimal mode for one cycle.

## Still not solved

Freeze export, as before. There is no EMS mode for "don't charge, but discharge to cover house load". Rate control in `charge_pv`, `import_ac` and `buy_power` also behaved unreliably in my testing, so I only use `charge_battery` for charging.

BingoBango80, 2026-10-10:
Hi all, just getting started - the thread is really helpful.

I've got a GoodWe GW6K-EHA-G20 inverter, 2 x GW8.3-BAT-D-G20 batteries (so 16kwh in total) and 15 x 450W panels (so 6750kw in total). GoodWe Experimental Integration is setup and has pulled through EMS mode https://github.com/mletenay/home-assistant-goodwe-inverter

I'll follow the thread - but to make it easier, has anyone (@dmacwalter) got a working apps.yaml file template they could link to? 

I'm happy to try to through it if not and setup a link - it looks like none of the links at the top of this thread exist any longer.

Thanks.