springfall2008/batpred #1003: v7.17.2 crashes predbat

SwiftRR, 2024-04-24:
**Describe the bug**
I have had predbat automatic update enabled.
At 18:00 pm 24th April, v7.17.2 installed.
This was followed by error messages and predbat freezing.
I spotted this at 7:30 pm but the time stamp at the top of my predbat plan was still on 18:00, with no updated lines since 18:00–18:30. Well and truly frozen.
I also tried restarting HA but this made no difference.
See status screenshot below together with predbat.log.

I had also been trying to install the carbon intensity integration, adding the necessary files to custom_components and add lines to my apps.yaml. I also enabled the carbon switch.
I never got this to work. I wasn't really sure about linking in to the National_UK data.

I have now unenabled automatic update. I tried to go back to pre carbon days with v7.16.7 but this didn't work. I couldn't unenable the carbon switch and it was still frozen.
So I then loaded v7.17.1 which has allowed me to get predbat unfrozen. I have now been able to unenable the carbon switch and I have commented out the carbon intensity line from my apps.yaml file.
Otherwise, I would have had to go the backup route.

**Expected behavior**
Predbat working smoothly as it usually does.

**Predbat version**
v7.17.2

**Environment details**
GE AC3.0  coupled inverter
Two 8.2 kWh batteries
- Standard HA install on Raspberry pi4 (HA on SSD drive)
- Very stable set up running 24/7

**Screenshots**
[image]

**Log file**
[predbat.log](https://github.com/springfall2008/batpred/files/15099030/predbat.log)

I have never had such as issue with predbat before, been running non-stop since 30th Jan.

Rob

Speshman, 2024-04-24:
https://github.com/springfall2008/batpred/issues/997

I'm getting similar issues as noted in that issue (which is marked as fixed but the issue persists for me)

gcoan, 2024-04-24:
> I tried to go back to pre carbon days with v7.16.7 but this didn't work. I couldn't unenable the carbon switch and it was still frozen.

The carbon switch was only introduced in predbat 7.17.0 so if you went back to a prior version then it wouldn't know anything about the switch and that's probably why you couldn't change it. 
Not sure why predbat was still frozen though. If you reloaded a version from HACS then you have to remember to restart appdaemon. 

I'm using the latest predbat with no issues but haven't tried the carbon integration. Looks to be fiddly to have to manually copy files, not a integration available in hacs which is a pain

gcoan, 2024-04-24:
> I'm getting similar issues as noted in that issue (which is marked as fixed but the issue persists for me)
> 
> I get the "round" issue with expert mode off and a different one with it on

If you don't add anything to apps.yaml (or leave it commented out) it works fine for me

Speshman, 2024-04-24:
ok so the new 17.2 version seems to work ok for me

SwiftRR, 2024-04-24:
In v7.17.1 I have commented out the carbon line in apps yaml.
I also turned the carbon switch OFF.

In my predbat log, I had the following error 18 times:
ERROR: Exception raised type NoneType doesn't define __round__ method

This NoneType/round error seems to be the issue, although we seem to be seeing this in different varieties.

I have now upgraded to v7.17.2 and predbat has behaved itself and from the timestamp, my plan is updating every 5 minutes as it should do..

My issue seems to have started after I added the carbon line to apps.yaml and tuned the carbon switch on.

I still have all the files added to custom_components but they should do no harm now (I hope).

For now, I am switching OFF the automatic upgrade switch.

Rob

springfall2008, 2024-04-25:
I think the issue has been fix in the latest release.

BTW: The downgrade would have been fine, and the carbon switch would not have done anything in the older version so could have been ignored.

LupoLoopy, 2024-04-25:
I'm running 7.17.3 and I'm still seeing the following error:

ERROR: Exception raised type NoneType doesn't define __round__ method

I'm running a SolarEdge install.

springfall2008, 2024-04-27:
Added a fix for that one on 'main' now

closed 2024-05-07 by springfall2008