springfall2008/batpred #1004: Rate thresholds keep changing

naltsta, 2024-04-24:
**Describe the bug**
High and low rate thresholds keep changing

**Expected behavior**
Thresholds only change occasionally

**Predbat version**

7.17.2

**Environment details**

Gen1 hybrid
[image]Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2024-04-27:
I'm not sure if this is really unexpected or not, maybe worth trying the latest release also

MalcolmSpencer, 2024-06-25:
I am seeing this alot, each time predbat runs it seems to change the rate but it seems to go in spells. I am on Octopus flux so the rates are stable. yesterday the thresholds didn't change all day, this morning the changed a lot and this afternoon they were stable.
[image]

Logs attached
[predbat.log](https://github.com/user-attachments/files/15975110/predbat.log)

closed 2025-01-01 by springfall2008