springfall2008/batpred #1005: Find optimum continuous car charging slots

arundalep, 2024-04-24:
My EV charger does not support pause or stop once charging begins. It has created a few errors by repeatedly cycling the charge requests when the slots have been split.

This means once batpred has started its first car charging slot, the charger will remain on until the car has finished charging even once the charging slot has finished. 

It would be really handy if batpred had the option for continuous car charging slots instead of splitting them up, much like the Octopus integration “target rates”.

If this is not an option is there a way for batpred to use the Octopus target slots for its EV charging prediction (it will still have battery SOC)?

springfall2008, 2024-04-25:
If you turn off car charging smart it will just charge whenever the home battery charges.

It's certainly a possible feature to have a single charge start time instead...

springfall2008, 2026-10-29:
Closing as stale. There has been no follow-up since April 2024.

The workaround from the thread still applies: turn off `switch.predbat_car_charging_plan_smart` and the car charges in the same window as the home battery rather than in split slots. If a single contiguous car charging window is still needed for chargers that cannot pause, please open a new issue saying which charger you have.

closed 2026-10-29 by springfall2008