springfall2008/batpred #998: values in predbat entities do not match values in configured entities

jacekowski, 2024-04-24:
I have the following configuration:

```
  inverter_type: MINE
...
  scheduled_charge_enable:
    - switch.schedule_charge
  scheduled_discharge_enable:
    - switch.schedule_discharge

```

values in  switch.schedule_charge does not match the value in sensor.predbat_mine_0_scheduled_charge_enable (in fact, it seems to be stuck on ON all the time).

Similiar with discharge entities.

springfall2008, 2024-04-24:
These will only work if you define the inverter to have these options:

has_charge_enable_time
has_discharge_enable_time

In the inverter settings must be on

jacekowski, 2024-04-24:
I've got this (i've tried it with both set to true and false and i didn't really notice any difference)

```
        has_charge_enable_time: True
        has_discharge_enable_time: True
        has_target_soc: True
        has_reserve_soc: True
        charge_time_format: "%Y-%m-%d %H:%M:%S"
        charge_time_entity_is_option: False

```
It does look like just using predbat_mine_0_* entities works for my purposes anyway.

springfall2008, 2024-11-16:
Closing this old issue, feel free to open a new one if still an issue

closed 2024-11-16 by springfall2008