springfall2008/batpred #1010: Battery run to min Soc even over peak prices

bensebborn, 2024-04-25:
Noticed this a few times now. The plan looks fine then it recalculates and switches to a very strange plan where the battery just discharges to min Soc and stays there for hours 

there’s an expensive period upcoming which is 8p more than the current price, but no charge is scheduled prior. 

sometimes it also shows car charging in slots where the battery is discharging, which is disabled 

 If I turn expert mode off, the plan returns to normal. So guessing it must be an advanced option but unsure which one would do this (haven’t changed many advanced options, mainly just those recommended for agile tarrif)

**Expected behavior**

Battery charges when there’s a clear difference between current price and future peak price. 

**Predbat version**

7.16.16

**Environment details**

Sunsynk 10k inverter
28.8kWh battery
EV

**Screenshots**

[image]
[image]

**Log file**
Can provide if needed. Currently on phone so not easy to access.

bensebborn, 2024-04-25:
[image]
[image]

bensebborn, 2024-04-26:
Have noticed this again today. Switching to/from expert mode mates no difference now

I had a decent plan which was estimating ending on approx £9.  Then refreshed and the plan is £24!

Will attach logs shortly

gcoan, 2024-04-26:
Yes that plan looks odd

Can you try upgrading to the latest version 7.17.3 and see if this improves things?

bensebborn, 2024-04-26:
I’m using the Sunsynk branch so unfortunately can’t upgrade until those changes are merged in

bensebborn, 2024-04-27:
Something I’ve noticed when this happens is all the prices are shown as red. So it doesn’t seem to be able to differentiate the cheaper slots. Maybe that’s why it’s not trying to charge?

[image]

springfall2008, 2024-04-27:
Now merged to 'main', can you test there please?

bensebborn, 2024-04-27:
Thank you, will do

bensebborn, 2024-04-27:
Thanks seems a lot better at first glance. I shall keep an eye as previously the plan did change quite drastically after a few runs, not sure why as the rates were the same and it was only about an hour later. 

Screenshot for the same period:

[image]

Still a couple of cheaper slots that I would have expected a charge in (eg Sunday 21:30 + 22:30 @ 15p, with all slots 16-19p after? But it might well be doing something more intelligent than I can spot right now

[image]

Will keep you posted

closed 2025-01-01 by springfall2008