springfall2008/batpred #1014: Getting ERROR: Exception raised type NoneType doesn't define __round__ method when Carbon Enable switched on

tractor137, 2024-04-26:
**Describe the bug**
Added the carbon entity which is working, added in the line to apps.yaml but selecting switch to on causes the error.

**Expected behavior**
HTML plan would show carbon usage, new apex card would populate. 

**Predbat version**

v7.17.3

**Environment details**

Givenergy hybrid inverter and battery
Standard install

**Screenshots**

**Log file**
2024-04-26 20:10:00.020703 INFO pred_bat: --------------- PredBat - update at 2024-04-26 20:10:00+01:00 with clock skew 0 minutes, minutes now 1210
2024-04-26 20:10:00.027437 INFO pred_bat: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 7.2 minutes
2024-04-26 20:10:00.030126 INFO pred_bat: Predbat /homeassistant/appdaemon/apps/batpred/predbat.py version v7.17.3 currently running, latest version is v7.17.3 latest beta v7.17.3
2024-04-26 20:10:00.033060 INFO pred_bat: Autoupdate: Currently up to date
2024-04-26 20:10:00.077860 INFO pred_bat: Configuration: forecast_hours 48 max_windows 96 num_cars 1 debug enable is False calculate_plan_every 5 calculate_fast_plan False
2024-04-26 20:10:00.082866 INFO pred_bat: Cars 1 charging from battery False planned [False], charging_now [False] smart [False], max_price [0.0], plan_time ['07:00:00'], battery size [100.0], limit [100.0], rate [2.3]
2024-04-26 20:10:00.086961 INFO pred_bat: Predbat mode is set to Control charge & discharge
2024-04-26 20:10:01.909861 INFO pred_bat: Found 11521 load_today datapoints going back 8 days
2024-04-26 20:10:04.906350 INFO pred_bat: Current data so far today: load 16.94 kWh import 16.8 kWh export 13.3 kWh pv 17.0 kWh
2024-04-26 20:10:04.909485 INFO pred_bat: Fetching futurerate data from None
2024-04-26 20:10:04.942654 INFO pred_bat: Fetching carbon intensity data from sensor.carbon_intensity_uk
2024-04-26 20:10:04.948816 INFO pred_bat: Warn: No carbon intensity history in sensor predbat.carbon_now
2024-04-26 20:10:04.969916 INFO pred_bat: Car 0 using Octopus Intelligent, charging limit 80.0, ready time 08:00:00 - battery size 100.0
2024-04-26 20:10:04.974237 INFO pred_bat: Current Car SOC kWh: [0.0]
2024-04-26 20:10:05.017044 INFO pred_bat: Standing charge is set to 65.121 p
2024-04-26 20:10:05.039875 INFO pred_bat: Octopus Intelligent slot at 04-26 01:00:00-04-26 01:30:00 assumed price 7.5
2024-04-26 20:10:05.044402 INFO pred_bat: Octopus Intelligent slot at 04-26 02:00:00-04-26 02:30:00 assumed price 7.5
2024-04-26 20:10:05.048851 INFO pred_bat: Octopus Intelligent slot at 04-26 03:00:00-04-26 03:30:00 assumed price 7.5
2024-04-26 20:10:05.053174 INFO pred_bat: Octopus Intelligent slot at 04-26 02:30:00-04-26 03:00:00 assumed price 7.5
2024-04-26 20:10:05.057428 INFO pred_bat: Octopus Intelligent slot at 04-26 03:30:00-04-26 04:00:00 assumed price 7.5
2024-04-26 20:10:05.061753 INFO pred_bat: Octopus Intelligent slot at 04-27 04:30:00-04-27 05:30:00 assumed price 7.5
2024-04-26 20:10:05.066269 INFO pred_bat: Octopus Intelligent slot at 04-27 05:30:00-04-27 06:00:00 assumed price 7.5
2024-04-26 20:10:05.071337 INFO pred_bat: Octopus Intelligent slot at 04-27 06:00:00-04-27 06:30:00 assumed price 7.5
2024-04-26 20:10:05.075619 INFO pred_bat: Octopus Intelligent slot at 04-27 07:00:00-04-27 07:30:00 assumed price 7.5
2024-04-26 20:10:05.874199 INFO pred_bat: Rate min forward looking: now 7.5 at end of forecast 7.5
2024-04-26 20:10:05.878618 INFO pred_bat: Import rates min 7.5 max 26.78 average 21.36
2024-04-26 20:10:05.899256 INFO pred_bat: Export rates min 15.0 max 15.0 average 15.0
2024-04-26 20:10:05.903470 INFO pred_bat: Rate thresholds (for charge/discharge) are import 26.28p (0.0) export 15.0p (0.0)
2024-04-26 20:10:05.915426 INFO pred_bat: High export rate found rates in range 15.0 to 15.0
2024-04-26 20:10:05.922723 INFO pred_bat: Low Import rate found rates in range 7.5 to 7.5
2024-04-26 20:10:05.927105 INFO pred_bat: Car 0 is using Octopus intelligent schedule
2024-04-26 20:10:05.931919 INFO pred_bat: Car 0 charging plan is: [{'start': 1710, 'end': 1770, 'kwh': 2.2, 'average': 7.5, 'cost': 16.5}, {'start': 1770, 'end': 1800, 'kwh': 0.76, 'average': 7.5, 'cost': 5.7}, {'start': 1800, 'end': 1830, 'kwh': 1.1, 'average': 7.5, 'cost': 8.25}, {'start': 1860, 'end': 1890, 'kwh': 1.1, 'average': 7.5, 'cost': 8.25}]
2024-04-26 20:10:06.075646 INFO pred_bat: Todays energy import 16.8 kWh export 13.3 kWh cost 1.26 p import 200.76 p export -199.5 p carbon 0.0 kg
2024-04-26 20:10:06.092361 INFO pred_bat: PV Data for pv_forecast_today total 20.71 kWh
2024-04-26 20:10:06.103833 INFO pred_bat: PV Data for pv_forecast_tomorrow total 15.71 kWh
2024-04-26 20:10:06.114655 INFO pred_bat: PV Data for pv_forecast_d3 total 17.23 kWh
2024-04-26 20:10:06.130757 INFO pred_bat: PV Data for pv_forecast_d4 total 31.2 kWh
2024-04-26 20:10:06.251172 INFO pred_bat: Historical data totals for days [1, 2, 3, 4, 5, 6, 7] are [11.63, 17.23, 15.76, 11.6, 12.61, 14.4, 20.03] - min 11.6
2024-04-26 20:10:06.282745 INFO pred_bat: Model filter enabled - Discarding day 4 as it is the lowest of the 7 datapoints
2024-04-26 20:10:06.304634 INFO pred_bat: WARN: Historical day 7 has 10 minutes of gap in the data, filled from 20.03 kWh to make new average 20.17 kWh (percent 99%)
2024-04-26 20:10:06.315694 INFO pred_bat: Historical days now [1, 2, 3, 5, 6, 7] weight [1, 1, 1, 1, 1, 1]
2024-04-26 20:10:06.427144 INFO pred_bat: Today's load divergence 131.53 % in-day adjustment 129.96 % damping 0.95x, Predicted so far 5.64 kWh with 12.6 kWh car excluded and 3.98 kWh import ignored and 0 forecast extra, Actual so far 8.8 kWh with 6.84 kWh car excluded and 1.3 kWh import ignored
2024-04-26 20:10:06.536475 INFO pred_bat: Inverter 0: Type GE GivEnergy
2024-04-26 20:10:06.539855 INFO pred_bat: Inverter 0 using Rest API http://homeassistant.local:6345
2024-04-26 20:10:07.799056 INFO pred_bat: Invertor time 2024-04-26 20:09:51+01:00 AppDaemon time 2024-04-26 20:10:07.794752+01:00 difference -0.28 minutes
2024-04-26 20:10:07.803458 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2024-04-26 20:10:07.808539 INFO pred_bat: Inverter 0 with soc_max 9.52 kWh nominal_capacity 9.52 kWh battery rate raw 2600 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 5.0 kW export limit 5.0 kW reserve 4.0 % current_reserve 4.0 %
2024-04-26 20:10:09.070447 INFO pred_bat: Inverter 0 SOC: 7.81 kW 82 % Current charge rate 2600.0 w Current discharge rate 2600.0 w Current power 2150.0 w Current voltage 52.0
2024-04-26 20:10:09.074683 INFO pred_bat: Inverter 0 scheduled charge enable is False
2024-04-26 20:10:09.079266 INFO pred_bat: Inverter 0 charge windows currently []
2024-04-26 20:10:09.083704 INFO pred_bat: Inverter 0 Charge settings: timed charged is disabled, power 2.6 kW
2024-04-26 20:10:09.093427 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
[... 76 lines cut ...]
2024-04-26 20:15:09.005592 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-04-26 20:15:09.010375 INFO pred_bat: Inverter 0 sensor.predbat_GE_0_scheduled_discharge_enable set to False
2024-04-26 20:15:09.015072 INFO pred_bat: Inverter 0 scheduled discharge enable is False
2024-04-26 20:15:09.019479 INFO pred_bat: Inverter 0 discharge windows currently [{'start': 0, 'end': 0, 'average': 0}, {'start': 1440, 'end': 1440, 'average': 0}]
2024-04-26 20:15:09.023054 INFO pred_bat: Found 1 inverters totals: min reserve 0.38 current reserve 0.38 soc_max 9.52 soc 7.71 charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 5.0 export limit 5.0 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2024-04-26 20:15:09.116458 INFO pred_bat: Base charge    window [  ]
2024-04-26 20:15:09.119936 INFO pred_bat: Base discharge window [ 26-04 00:00:00 - 26-04 00:00:00 @ 0p 100.0%, 27-04 00:00:00 - 27-04 00:00:00 @ 0p 100.0% ]
2024-04-26 20:15:09.122396 INFO pred_bat: Plan was last updated on 2024-04-26 20:00:00+01:00 and is now 15.0 minutes old
2024-04-26 20:15:09.126002 INFO pred_bat: Best charge    window [ 26-04 23:30:00 - 27-04 02:20:00 @ 7.5p 84%, 27-04 02:30:00 - 27-04 03:00:00 @ 7.5p 92%, 27-04 04:30:00 - 27-04 06:30:00 @ 7.5p 100%, 27-04 07:00:00 - 27-04 07:30:00 @ 7.5p 98%, 27-04 23:30:00 - 28-04 02:50:00 @ 7.5p 97%, 28-04 03:00:00 - 28-04 03:30:00 @ 7.5p 100%, 28-04 04:00:00 - 28-04 04:30:00 @ 7.5p 100%, 28-04 05:00:00 - 28-04 05:30:00 @ 7.5p 98% ]
[... cut ...]
2024-04-26 20:15:09.590856 INFO pred_bat:  METRIC: [   1.26,    1.26,    1.26,    1.26,    67.2,    67.2,    67.2,    67.2,    67.2,    80.9,   95.57,  103.89,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  108.13,  173.25,  173.25,  173.25,  173.25,  173.25,  173.25,  173.25,  173.25,  173.25,  173.25,  173.25,  173.25,   173.2,  172.76,  171.68,  167.52,  153.58,  142.84,  141.68,  141.64]
2024-04-26 20:15:10.164387 INFO pred_bat: Battery has 48.0 hours left - now at 7.71
2024-04-26 20:15:10.265354 INFO pred_bat: ERROR: Exception raised type NoneType doesn't define __round__ method

springfall2008, 2024-04-27:
Can you test 'main' to see if its resolved?

tractor137, 2024-04-27:
No still showing the error I'm afraid after updating to 7.17.4

tractor137, 2024-04-28:
Seems to be fixed but have a new problem now which I will raise as a separate issue

closed 2024-04-28 by tractor137