springfall2008/batpred #1015: Discharge Low Power setting

campbellcopeland, 2024-04-26:
**Is your feature request related to a problem? Please describe.**
Recently moved to Flux and have a 3 hour evening discharge window which is combined (via combine discharge switch) but only discharges during 2nd half of period at full rate. Can this discharge at a lower rate? Feeling this may be better for long term battery health/calibration?

**Describe the solution you'd like**
When there is a three hour Flux window, the discharge rate would decrease from AIO 6kW to a rate that allowed consistent rate across all three hours to achieve target SOC

**Describe alternatives you've considered**
There was an automation listed on Facebook group from Willy Reckitt called “Adjustable Force Export Single” but looking for a PredBat solution

**Additional context**
Feels like a discharge version of “predbat_set_charge_low_power”. Suspect it will help battery calibration to discharge at 3.6kW for 3 hours rather than 6kW for 1.5 hours.

mpartington, 2024-04-27:
Turning "Set Charge Low Power Mode" switch to on does this for you. 

Not sure if you need to select expert mode first to expose it, but set up is covered in the guide here under "Inverter control options":
https://springfall2008.github.io/batpred/customisation/#expert-mode

campbellcopeland, 2024-04-27:
I use the Charge Low Power setting already - it’s an equivalent for discharge that I’m interested in.

mpartington, 2024-04-27:
> I use the Charge Low Power setting already - it’s an equivalent for discharge that I’m interested in.

Apologies, I'll grab another coffee 🙂. Sounds like a good proposal

campbellcopeland, 2024-04-27:
To be fair, I’m not sure if there are advantages to a reduced rate of discharge or not. Unsure if this is beneficial for the BMS to calibrate SOC compared to maximum rate discharge and then just covering house load.

bensebborn, 2024-04-27:
Our batteries benefit from some time at 100% to rebalance. So if this was implemented maybe it still needs to sit at 100% for a specific time period?

springfall2008, 2024-05-06:
I've been reluctant to change the discharge power is it would prevent the battery from serving the home should you turn something power hungry on during that time period.