springfall2008/batpred #1020: Plan around midnight

Meatballs1, 2024-04-27:
The Plan seems to struggle around midnight and shows the wrong days/forecasted rates. I think it fixes itself after 1am (BST). Timezone issue or some confusion about what day it is around midnight? 

[image]

Meatballs1, 2024-04-28:
Plan recalulated after 1am:

[image]

Meatballs1, 2024-05-05:
Set a manual discharge for 00:30 but it didn't trigger:

[image]

Meatballs1, 2024-05-05:
Setting the discharge to 23:30 on the manual settings did work.

Meatballs1, 2024-05-06:
So despite appdaemon being set to `Europe/London` my host and the appdaemon docker were both running in UTC. now() gave UTC rather than BST despite running within appdaemon. 🤦

springfall2008, 2024-05-06:
I think you need to check that the correct timezone is set in:

appdeamon.yaml
apps.yaml
In Home Assistant
on your host.

Meatballs1, 2024-05-06:
I believe setting the TZ in docker should fix it, but will check.

Few things to note:

* Appdaemon recommends it's own time functions self.get_time()
* now in update_time is non tZ aware and so was just set to UTC
* utcnow in update_time is TZ aware and is local time +0100. Would expect it to be +0000

springfall2008, 2025-04-19:
Closing this issue as its stale, please create a new one if you still have a problem

closed 2025-04-19 by springfall2008