springfall2008/batpred #1021: Car charging way over max charge

dandwhelan, 2024-04-28:
**Describe the bug**
I've a manual input to select the max change I want on the car. I've moved from setting the soc of the car manually to an integration. The max charge percentage set isn't being followed. 

**Expected behavior**
Charge to the percentage I want the car to charge to.
**Predbat version**

7.17.4

**Environment details**

- giv 8.2kw inverter hy gen 2 5kw
- Standard HAOS installer 
- Anything else?
[apps (1).log](https://github.com/springfall2008/batpred/files/15141135/apps.1.log)
[appdaemon (2).log](https://github.com/springfall2008/batpred/files/15141136/appdaemon.2.log)

springfall2008, 2024-04-28:
Do you have the car charging limit set to the right figure as a % e.g. 75?

dandwhelan, 2024-04-28:
Yeah, I had it set to 55% overnight.

[image: image.png]

mpartington, 2024-04-29:
Have you set the battery size correctly in apps.yaml?

car_charging_battery_size:
    - 61.7
From what I can see in the logs, your car was set to be ready at 8am, so assume it finished then.  The last line in the log is unfortunately at 07:36, so just misses the finish

At 2024-04-28 07:36:44.432295 INFO pred_bat: Current Car SOC kWh: [31.152330000000003]
2024-04-28 07:36:45.735228 INFO pred_bat: Plan car 0 charging from 31.152330000000003 to 33.935 
  
So at 8am, Predbat was expecting the battery to be 33.935 kWh and at 07:36 it was 31.15kWh, which seems on track to hit it.
This seems to match your settings  55%*61.7 = 33.935 kWh

Another scenario I guess is if the car stopped reporting the SoC, in which case Predbat assumes the SoC is zero and may have continued to charge until it reported again (by which time it would have overcharged), but would need the later logs to confirm.

Of course Trefor may spot something else entirely

dandwhelan, 2024-04-30:
> Have you set the battery size correctly in apps.yaml?
>
> car_charging_battery_size: - 61.7 From what I can see in the logs, your car was set to be ready at 8am, so assume it finished then. The last [...]

Yeah the battery is set in kwh total capacity in the configuration.
I see where your coming from with 7.30am but that would be because I turned predbat to use manual soc as I couldn't get it to stop charging the car.

dandwhelan, 2024-05-02:
Ok, I've been diving into the data. 
At 4:30am this morning Predbat thought my car SOC was... 
[image]
[image]
Yet the entity above was at 53kwh at 4:30 a.m.
[image]

It's not looking at this SOC of the entity and adding the numbers onto something else...

[appdaemon1.txt](https://github.com/springfall2008/batpred/files/15192469/appdaemon1.txt)
[appdaemon2.txt](https://github.com/springfall2008/batpred/files/15192470/appdaemon2.txt)

Also I've no idea what is going on, on this predbat entity. 

[image]

[image]

dandwhelan, 2024-05-02:
Does the Car SOC work off % when using none manual soc mode?

mpartington, 2024-05-02:
Car_charging_soc must be in % whichever mode you are in.

It's not actually used in manual mode, as that mode assumes.your car doesn't report a soc via home assistant (so this doesn't exist).  For manual mode.you need to enter the staring soc (in kWh) and Predbat works out the final soc based on the max charge entity (input number in %) and the battery capacity in apps.yaml. It will then calculate the charge time needed.to deliver that.

[image]

dandwhelan, 2024-05-04:
> I'll update the doc to make it more explicit that Soc is a percentage

It's odd because it used manual soc uses the number and automation uses
percentage.

Anyway it's working now. Thanks

closed 2024-05-27 by gcoan