springfall2008/batpred #1022: Getting ERROR: Exception raised at randonly with v7.17.4

Surb109, 2024-04-28:
**Describe the bug**
Using predbat v7.17.4 will randomly will get ERROR: Exception raised, at which point the Predbat HTML plan schedule will not display. Randomly after some time, the error will clear and the html plan will show again

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**
7.17.4

**Environment details**

- Growatt Min8000tl-x
- Givenergy All-In-One

**Screenshots**
[image]

**Log file**
[appdaemon copy.log](https://github.com/springfall2008/batpred/files/15141820/appdaemon.copy.log)

springfall2008, 2024-04-28:
Your issue here is Home Assistant disconnecting from AppDaemon, leading to fetching bad data.

However I should probably try to catch some of those errors.

Surb109, 2024-04-28:
Appreciate the the update. As long as this isn't going an issue, happy to wait for this to be addressed in a future update

closed 2024-05-07 by springfall2008