springfall2008/batpred #1027: Octopus Intelligent - Unusual plan created when car charging

nickgee31, 2024-04-29:
**Describe the bug**
Last night I plugged my car in which is on Octopus Intelligent Go. Predbat recognised the car charging slots however it didn't charge my FoxESS battery during those times. At times it states the battery is idle and it will somehow hold the SOC however this didn't happen and the battery ended up discharging to the car at times. It is entirely possible that I am just missing a setting here but thought I'd post in here in case. I. had also tried downgrading to V7.17.0 to see if that helped but a similar plan was created.

**Expected behavior**
I would expect that when the car is charging that the battery is also charging. In the screenshot below, I would expect 23:30 to be charging, 00:00 to discharge and then 00:30 to 01:30 to be charging again, and so on.

**Predbat version**

V7.17.6

**Environment details**

- Inverter and battery setup - FoxESS - 
[apps.yaml.log](https://github.com/springfall2008/batpred/files/15146291/apps.yaml.log)
- Standard HAOS installer or Docker - Standard HAOS install
- Anything else?

**Screenshots**
[image]

**Log file**
[predbat.log](https://github.com/springfall2008/batpred/files/15146262/predbat.log)

nickgee31, 2024-04-30:
Just another example of a similar thing from last night in case it helps: 
[predbat.log](https://github.com/springfall2008/batpred/files/15160089/predbat.log)

[image]

nickgee31, 2024-04-30:
Out of interest I went back to an even earlier version (V7.16.17) as it mentioned about car planning changes. Things seem a lot better on V7.16.17:
[image]

Compared to the latest version, V7.17.6: 

[image]

So it does seem like something changed in an update to make the plan worse. I'll stay on the older version for now but happy to test any improvements.

nickgee31, 2024-05-01:
For what it's worth I've been through all the versions and the version I can get the 'normal' plan on is way back on V7.16.16 
[image]

I've been able to replicate this odd behaviour on two different Predbat set ups, with both a GE and a FoxESS inverter so don't think it's limited to the inverter type. I'll stick on this version and happy to test any changes if it's able to be spotted what changed.

phiebs, 2024-05-03:
What's you're Metric Battery Cycle Cost set to? I've had this before and it was set a bit too low (1p/kWh) and I had to tweak this up to 2.4p/kWh to get it to stop trying to discharge to the grid/car. YMMV!

tractor137, 2024-05-04:
I've been having the same issue and have just downgraded from 7.17.10 to 7.16.16 which seems to have solved the problem for now

springfall2008, 2024-05-06:
I think the issue here is that your inverter isn't supporting the mode where discharge is disabled while the car charges and hence the plan is sensible (holding the battery during car charging) but it doesn't happen.

As you have Fox I think we should look at how its configured, and then add it to the documentation once correct. Can you share how you set it up?

springfall2008, 2024-05-06:
I've just made a fix on main which I will release soon which will prevent any type of forced discharge during a car charging slot if 'allow car charging from battery' is off.

nickgee31, 2024-05-06:
Hi @springfall2008,

Just tested this on my GivEnergy AC3 inverter.

On version V7.16.16 this is the plan: 
[image]

And on V7.18.0 this is the plan: 
[image]

I can confirm that I have 'allow car charging from battery' to off. Here is the predbat.log if it is any help: 
[predbat.log](https://github.com/springfall2008/batpred/files/15225019/predbat.log)

tractor137, 2024-05-10:
I'm still experiencing the same problem - I've been trying each of the new releases but having to keep reverting to 7.16.16. The plan it creates in new versions is fine until I plug the car in, then It recalculates and throws in either unused charge slots or freeze charges.

pacemaker82, 2024-05-11:
+1 here for me too, seeing odd behaviour where the plan intends to keep the charge of the battery during car charge, but then decides to change that plan when the slot opens, setting it to Idle, then discharging the battery to the car. (GivEnergy AC 3 w/ 9.5 kWh battery)

nickgee31, 2024-05-13:
Hopefully this is helpful @springfall2008.

I've been through all the versions and the tipping point is V7.16.4 and it looks like the only change is https://github.com/springfall2008/batpred/pull/1013. Going back to V7.16.3 results in a normal plan.

springfall2008, 2024-05-14:
Interesting, it maybe there is a mistake in this change around the freeze, let me remove that bit...

springfall2008, 2024-05-14:
There's a potential fix on 'main' for testing

nickgee31, 2024-05-14:
Just tested main and all looks good now! 

[image]

Thank you for your help @springfall2008

closed 2024-05-28 by nickgee31