springfall2008/batpred #1028: Predbat stuck in FreezeDischarge folloiwng v7.17.6 update

nbullus, 2024-04-29:
**Describe the bug**
When the plan switched current slot from a "Freeze Discharge" state to a "Discharge" state, the battery did not discharge and it remained as if it was still in a "Freeze Discharge" state. As a result, surplus solar continued to increase the SOC at the minimum charge rate of around 30w, and the remainder was exported to the grid.

Other strange behaviour was also noted earlier which may be related such as a Combined Charge/Discharge slot changing to Discharge but battery remained in a Charge state. This will be reported separately once sufficient evidence has been taken.

**Expected behaviour**
Surplus solar & battery should have discharged to the grid, but it seemed to be stuck in the Freeze Discharge state (see first screenshot). 

The only way I could get out of this stuck "Discharge Freeze" state was to do the following:

1. Revert back to v7.17.4 (pre-Pause battery function introduced)
2. Disable GivTCP Battery Pause
3. Revert inverter settings to clear slot times
4. Restart PredBat

As a result the DIscharge is now working again as expected (see second screenshot)

**Predbat version**

7.17.6

**Environment details**

- GivEnergy AIO & Solis string inverter
- Standard HAOS installer

**Screenshots**

V7.17.6 - Stuck In Freeze Discharge
[image]

V.7.17.4 - Expected Behaviour
[image]

**Log file**
[Predbat Log.txt](https://github.com/springfall2008/batpred/files/15147233/Predbat.Log.txt)

gcoan, 2024-04-29:
Have a look at your givtcp log.  It could well be that predbat issued the commands to unfreeze your battery but these commands never got to the inverter - e.g. a comms failure between givtcp and the inverter.

You should be able to change any predbat activity much easier than you did do.

One option is to simply put predbat into read only then change it back again.  The switch to read only will put the inverter into Eco mode and then when you switch read only off, it'll re-execute the plan commands.

Or, even simpler, you can always directly manipulate the givtcp controls for your inverter.  I have a small dashboard with the controls on I can change and its worth you understanding how these work.  The usual ones you need are mode, battery reserve, charge rate and discharge rate, but for completeness you can add the start and end charge and discharge times and battery pause controls

nbullus, 2024-04-29:
GivTCP Log below. Errors are apparent but this is when I reset/restarted everything. I can see no errors around the time it changed state to discharging (09:45) and remained in the FreezeDischarge state. I see no setting of discharge time or time export mode being set like you see earlier in the log:

2024-04-29 08:36:28,194 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-04-29 08:36:36,471 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-04-29 08:36:41,953 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-04-29 08:50:10,381 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 09:01
2024-04-29 08:50:13,257 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-29 08:50:19,180 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 09:01
2024-04-29 08:50:21,866 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-29 08:50:27,461 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 09:01
2024-04-29 08:50:30,349 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-29 08:50:36,611 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export
2024-04-29 08:50:40,598 - Inv1 - write       -  [INFO    ] - Setting export mode was a success
2024-04-29 08:50:42,344 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-04-29 08:50:48,456 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 08:50:49,995 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 08:50:57,869 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 08:50:59,416 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 08:51:05,877 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: PauseCharge
2024-04-29 08:51:07,424 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
2024-04-29 09:00:11,287 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-04-29 09:00:12,091 - Inv1 - write       -  [INFO    ] - Setting dynamic mode was a success
2024-04-29 09:00:13,857 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-04-29 09:00:30,195 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 00:00
2024-04-29 09:00:33,093 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-29 09:26:27,902 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 6000 (50)
2024-04-29 09:26:29,454 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-29 09:26:35,527 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: Disabled
2024-04-29 09:26:37,073 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-04-29 09:30:10,824 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 09:30:12,367 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 09:30:18,458 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: PauseCharge
2024-04-29 09:30:20,197 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
2024-04-29 09:51:57,664 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: Disabled
2024-04-29 09:51:59,284 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-04-29 09:52:24,629 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 6000 (50)
2024-04-29 09:52:26,186 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-29 09:52:32,284 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 09:52:33,826 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 09:52:38,825 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 09:52:40,366 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 10:01:57,581 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 6000 (50)
2024-04-29 10:01:59,321 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-29 10:02:05,419 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 10:02:07,161 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 10:11:23,899 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 12:01
2024-04-29 10:11:26,781 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-29 10:11:33,033 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export
2024-04-29 10:11:37,189 - Inv1 - write       -  [INFO    ] - Setting export mode was a success
2024-04-29 10:11:38,734 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-04-29 10:12:10,261 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.129, 8899) failed: timed out
2024-04-29 10:12:12,765 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.129, 8899) failed: timed out
[... cut ...]
2024-04-29 10:46:45,706 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-04-29 10:46:51,753 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 10:46:53,298 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 10:51:30,833 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 10:30 - 11:01
2024-04-29 10:51:35,539 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success

nbullus, 2024-04-29:
Now seeing same behaviour at v7.17.4 in that its stuck in Freeze Discharge yet plan states it should be Discharging.

[image]

GivTCP Log:

2024-04-29 10:40:10,772 - Inv1 - write       -  [INFO    ] - Disabling Charge Schedule
2024-04-29 10:40:12,512 - Inv1 - write       -  [INFO    ] - Disabling Charge was a success
2024-04-29 10:40:18,616 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
2024-04-29 10:40:20,502 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
2024-04-29 10:40:26,641 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 16
2024-04-29 10:40:28,202 - Inv1 - write       -  [INFO    ] - Setting shallow charge 16 was a success
2024-04-29 10:45:10,548 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-04-29 10:45:12,089 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-04-29 10:46:30,431 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 10:30 - 12:01
2024-04-29 10:46:33,502 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-29 10:46:39,797 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export
2024-04-29 10:46:43,960 - Inv1 - write       -  [INFO    ] - Setting export mode was a success
2024-04-29 10:46:45,706 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-04-29 10:46:51,753 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-29 10:46:53,298 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-29 10:51:30,833 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 10:30 - 11:01
2024-04-29 10:51:35,539 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-29 11:00:10,141 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-04-29 11:00:10,967 - Inv1 - write       -  [INFO    ] - Setting dynamic mode was a success
2024-04-29 11:00:12,532 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-04-29 11:00:28,608 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 00:00
2024-04-29 11:00:31,308 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success

Predbat Log 2 v7.17.4 
[PredBat2 Log v7.17.4.txt](https://github.com/springfall2008/batpred/files/15147971/PredBat2.Log.v7.17.4.txt)
attached

nbullus, 2024-04-29:
I have put it into 'read only' and back again. This got it discharging correctly again until 11:30. 

At 11:33 it went into 'Freeze Discharge' and this was OK and behaved as expected.

[image]

At 11:36 the slot was changed back to 'Discharge', but remains in 'Freeze Charge' behaviour as before:

[image]

gcoan, 2024-04-29:
Thanks for the GivTCP log,  rules out a comms issue

Looking at the recent screenshots, you say "stuck in freeze discharge when plan says discharge" - I'm guessing you are saying that based on the predbat status = freeze discharge but the plan = discharge?

Asking this because I first thought you were saying that the inverter wasn't responding to predbat plan commands.

There may not actually be a problem here.  Predbat plans its activity in 5 minute slots and the current activity is shown in predbat status

However, the predbat plan shows the plan for the 30 minute period and if the individual slots are slightly different (e.g. freeze discharge, discharge, freeze discharge, discharge) then when Predbat displays the plan it has to summarise this,  and so it could be that the overall status is Discharging but the individual 5 minute slots beneath (that the predbat.status shows) could be slightly different.
There isn't a lot of difference between freeze discharge and discharge when the battery is almost empty, the F-D just holds the SoC but allows discharge based on house load whereas the Discharge will force export the battery contents to the grid.

nbullus, 2024-04-29:
Yes that's correct. The Predbat state = "Discharge Freeze", plan state = "Discharge". 

I could not see any events in the logs that shows Predbat attempting to set the discharge slot start/end times or the state when the plan moves it from 'Freeze Discharge' to 'Discharge' and this may be the reason why the state never changes. It certainly does not look like Predbat is attempting to change the state and the inverter failing to accept it, its more the inverter hasn't been told to do anything.

The end result is the battery is behaving as per the Predbat state but the Predbat state does not change to what is planned. The state of the slot in the plan many change multiple times within the 30 min plan slot (and often does), but the state should reflect what it wants the battery to do. In the last example, I have had a slot remain in a 'Discharge' state for the full 30 mins and PredBat has it in a state of  'Freeze Discharge' for the same 30 mins which is incorrect. It was not a case of the state changing multiple times in the 30 mins.

The key difference is that the plan is to export the battery and reduce the SOC, but instead charges the battery (albeit at minimum charge) and increases the SOC. The plan then re-evaluates itself and has to keep changing what it planned as its not sticking to it's own plan.

nbullus, 2024-04-29:
I have been monitoring what Predbat plans to do v/s what it actually does and checking what is in the GivTCP log. Runs fine for some time and then strange events occurred at 16:00 and 16:30 (see attached).

Irrespective of why the behaviour occurs, what is consistent is that the state label in the plan does not change to reflect what it is actually doing. At the 16:00 slot you will see it changed its mind 5 times and at no time did it modify the label from 'Discharging' when there were two occasions when it decided to change to a 'Charge'.

I am going to turn 'off' of the following settings to eliminate possibilities and try and make the plan simpler to see if this is having any effect and causing it to change it's mind:

- Calculate discharge on charge slots
- Calculate secondary order slots

Note that I have also moved forward again to latest version of Predbat as I don't think the recent upgrades have any effect on this.

[GivTCP Log for Discharge.docx](https://github.com/springfall2008/batpred/files/15151678/GivTCP.Log.for.Discharge.docx)

nbullus, 2024-05-01:
Yes I am also seeing similar errors reported repeatedly in my HA logs as described in #1032:

2024-05-01 12:39:36.637 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on /addons/46f69597_appdaemon-predbat/changelog request 0, message='Attempt to decode JSON with unexpected mimetype: application/octet-stream', url=URL('http://172.30.32.2/addons/46f69597_appdaemon-predbat/changelog')
2024-05-01 12:39:36.637 WARNING (MainThread) [homeassistant.components.hassio.data] Could not fetch changelog for 46f69597_appdaemon-predbat:

SwiftRR, 2024-05-02:
Many of us are observing the same behaviour, as shown in my plan this evening:
[image]
It doesn't seem to be right that the grid is being used rather than the battery, which would be cheaper.
I observed the same early this morning when import was 18p and my battery SOC was 100. The status was freeze charge, the grid being used rather than my battery. 
I have **input_number.predbat_metric_battery_cycle** set as 0p kWh..

Some are contemplating unenabling the freeze charging switch.
I don't want to do this as the freeze charging option can be useful.

The situation seems to arise when 

- the battery is full/nearly full and 
- the import price is close to the export price.

Rob

nbullus, 2024-05-02:
> Many of us are observing the same behaviour, as shown in my plan this evening: <img alt="image" width="599" src="https://private-user-images.githubusercontent.com/68816364/327582757-ef8341b7-3442-4e86-b946-b710ef6e3a8a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODM2MDYsIm5iZiI6MTcxNDY4MzMwNiwicGF0aCI6Ii82ODgxNjM2NC8zMjc1ODI3NTctZWY4MzQxYjctMzQ0Mi00ZTg2LWI5NDYtYjcxMGVmNmUzYThhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAyVDIwNTUwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY5N2UwNGUyNzU1YjRjMTc4MTcwZmMyM2RjOTQxNDQ4MGVhMzcxOTVkNTM3YzljMzhjYWVjZmQ1ZWE3NjBjZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.z_hLw8iIxnZILTtZnobTMPbWhxgnZswAwbnmiPtUkH4"> It doesn't seem to be right that the grid is being used rather than the battery, which would be cheaper. I observed the same early this morning when import was 18p and my battery SOC was 100. The status was freeze charge, the grid being used rather than my battery. I have **input_number.predbat_metric_battery_cycle** set as 0p kWh..
> 
> Some are contemplating unenabling the freeze charging switch. I don't want to do this as the freeze charging option can be useful.
> 
> The situation seems to arise when
> 
> * the battery is full/nearly full and
> * the import price is close to the export price.
> 
> Rob

The issue you are describing is not the same issue that I have raised here. Your issue is why is is it going into Freeze/Hold in the first place, but my issue is that once it's in Freeze, it cannot get out of it when its planned too change, e.g. Freeze Discharge -> Discharge. To avoid confusing two separate issues, I would remove and raise in the general Predbat chat to get views, and if required, raised as a separate issue.

gcoan, 2024-05-02:
Move the comment to #897 @SwiftRR ?

SwiftRR, 2024-05-02:
Following feedback, I have moved my comment to a new issue #1035.
See also related issue #897. 

Rob

nbullus, 2024-05-04:
Another example with different states. Predbat and Plan state running 'Idle' and battery should therefore be charged from excess solar. Charge window all setup as expected, yet battery continues to discharge and excess solar is being exported.

[image]

GivTCP log confirms it also wants to charge battery:

[image]

gcoan, 2024-05-04:
can you do a screen shot of what the givtcp mode, charge and discharge rates, battery reserve and pause mode were around this time - should from this be able to see what predbat was instructing the inverter to do

nbullus, 2024-05-04:
The screenshots above show the GivTCP mode, charge and discharge rates (under the Battery card the Battery Status card).

nbullus, 2024-05-05:
Additional logs for further issue requested by Trefor. There seems no obvious pattern to this issues. Sometimes it wants to charge but is stuck in Discharge, sometimes its stuck in Charge and cant get into Discharge. Sometimes switching to 'ReadOnly' and back gets it back on track, but sometime the inverter settings need resetting to get it going again and doing what is planned. 

[image]

Logs below:

[GivTCP_log.txt](https://github.com/springfall2008/batpred/files/15213436/GivTCP_log.txt)
[PredBat_log.txt](https://github.com/springfall2008/batpred/files/15213437/PredBat_log.txt)

gcoan, 2024-05-09:
> The 'Enable Charge Schedule' setting was also turned 'on' which will prevent the battery from charging again until 10:00. However, this conflicts with what the plan is currently trying to do at 9:00 which is to charge the battery up to 100%.
> 
> This means the 'Enable Charge Schedule' setting is being turned 'on' early

Can you see in Home Assistant when the enable charge schedule entity was set to on, and tie this to the predbat logs around that time?

nbullus, 2024-05-09:
> Can you see in Home Assistant when the enable charge schedule entity was set to on, and tie this to the predbat logs around that time?

Will do. I have just done some HA updates and restarted home assistant. This has reset the 'Enable Charge Schedule' back to off and there is a charge timeslot set which is a good start. I've also set an automation up to notify me the next time the 'Enable Charge Schedule' is turned on so I can dig the log out quickly. I suspect it will turn on at the required time, but remain on when it sets up the next charge window after that.

nbullus, 2024-05-10:
> Can you see in Home Assistant when the enable charge schedule entity was set to on, and tie this to the predbat logs around that time?

This screenshot shows that 'Enable Charge Schedule' is set (10:30 - 13:00) and this was actioned over 4 hours before the schedule starts. The below GivTCP Log below confirms this was enabled at 02:15, yet when it was first restarted it stated in the logs it wouldn't set this until within 30 mins of the schedule start. This means that no further charges can now happen till 10:30, yet charging is already planned between 08:30 - 09:30. The Predbat log for the 2:20 period is attached.

[image]

**GivTCP log:-**

2024-05-10 02:00:19,558 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of night rate tariff (Dynamic)
2024-05-10 02:15:10,090 - Inv1 - write       -  [INFO    ] - Disabling Charge Schedule
2024-05-10 02:15:11,657 - Inv1 - write       -  [INFO    ] - Disabling Charge was a success
2024-05-10 02:15:17,711 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
2024-05-10 02:15:19,397 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
2024-05-10 02:15:25,423 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 88
2024-05-10 02:15:30,819 - Inv1 - write       -  [INFO    ] - Setting shallow charge 88 was a success
2024-05-10 02:20:34,233 - Inv1 - write       -  [INFO    ] - Enabling Charge Schedule
2024-05-10 02:20:35,777 - Inv1 - write       -  [INFO    ] - Setting Charge Enable was a success
2024-05-10 02:20:41,857 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-05-10 02:20:43,597 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-05-10 05:00:01,600 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)
2024-05-10 06:00:09,151 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 10:30 - 13:00
2024-05-10 06:00:12,037 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2024-05-10 06:00:18,141 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: Disabled
2024-05-10 06:00:19,680 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success

[Predbat Log 5-10-2024.txt](https://github.com/springfall2008/batpred/files/15271477/Predbat.Log.5-10-2024.txt)

nbullus, 2024-05-10:
Further to the above, just spotted the below schedule setup and enabled for 01:30 - 06:00 tomorrow morning when there is clearly no charging planned n this period. I think there is a general issue with enabling/disabling of the charge schedule. It should really be enabled when it gets to the start time and then disabled when its finished. It should never be left enabled as its going to cause problems with old schedules that may be planned but then the plan changes before it gets there.

[image]

gcoan, 2024-05-11:
> I think there is a general issue with enabling/disabling of the charge schedule. It should really be enabled when it gets to the start time and then disabled when its finished. It should never be [...]

In of itself there shouldn't be an issue with setting the charge schedule ahead of time and the advantage of doing so in advance is that if there's any issues with communicating from Predbat to the inverter, the schedule is already set. If the schedule is not set until it's needed it needs a perfect comms link, any timeouts or disconnects will mean the charging is delayed. 

For this to be ok though it does require that if predbat changes the plan that the future charging schedule is changed as well. Perhaps this is where the problem is?

nbullus, 2024-05-11:
I see no advantage to enabling a charge schedule in advance. if the plan changes you can end up being in same situation but in the reverse whereby you can’t disable it due to comms issue and it carries on stuck in charge mode. 

I don’t see any problem in setting the charge start/end date in advance, but not if you also enable it in advance as well. I have seen a lot of times when the charge schedule is many hours away and the schedule has already been enabled and this stops any other charge from happening.  When you first Predbat and it sets up the first future schedule,  it doesn’t actually enable the charge schedule immediately but states in the log that it won’t enable it until it’s within 30 mins of the schedule start. So I think it’s a combination of two things:

1. Not Enabling a schedule more than 30 mins away.
2. Not disabling the schedule if the plan changes, e.g start time pushed further out or charge cancelled. 

It may be point (2) that is the crux of the issue in that  it’s not disabling the schedule when needed as you have pointed out, and may cure the problem.

nbullus, 2024-12-16:
I think this can be closed now as there have been so many versions released since this was raised and I also think the issue got resolved when then 'switch.predbat_inverter_set_charge_before' switch was introduced and not seen any further occurrences. Only thing is to add it to the documentation so say what it does for others to use.