springfall2008/batpred #1029: Car charging does not use the new pause method implement in #360 when holding battery

mpartington, 2024-04-29:
**Describe the bug**
As discussed, it appears that car charging has not implemented the new pause methodology introduced in #360 .

This means there is a higher lever of battery leakage during car charging if the battery is not charging.

**Expected behavior**
Set the slot to freeze charge if the car is charging and option to charge from battery is false

**Predbat version**
V7.17.6

- Standard HAOS installer or Docker

**Screenshots**
[image]
[image]
[image]

**Log file**
Not required, cause is known

mpartington, 2024-05-02:
Another bug spotted yesterday, that may also be car charging related, but perhaps a more general error.

Both the car and battery have been charging, but as the battery hits 100%, Predbat switches enable charge off and discharge freeze on. This allows the battery to supply the car (and house) despite being in a car charging slot and the rates being low.

It does this a few times
[appdaemon.log](https://github.com/springfall2008/batpred/files/15184331/appdaemon.log)

[image]

I assume it's Predbat making the change, I can't find any automation triggering at those times
[image]

mpartington, 2024-05-02:
Edit.. It is Predbat driving this behaviour. I have no automations that change the battery discharge rates

[image]

mpartington, 2024-05-02:
So looking at the logs, is there a timing (and possible logic) issue - does Predbat know the car is charging at the start of the run?

The issue seems to be the order in which Predbat does stuff, and probably the speed my RPI4 processes commands.  

When SoC reaches 100% Predbat selects PauseDischarge (04:27:41), which allows the battery to discharge.  It then realises the car is charging and it needs to hold the battery.  So it subsequently adjusts (04:28:06) the battery reserve level to 100% (which is absolutely fine on my inverter)

This means there is a 30 second burst of discharge from the battery. 

Next run, Predbat realises the SoC is not at 100%, so Disables pause and lets the battery charge back up. Rinse and repeat....

What is maybe not clear is why this behaviour stops after 4.30, but I suspected this might be due to the change in rates at that point, but appears not - 4 til am 5 is 10.1p

2024-05-02 04:27:31.569481 INFO pred_bat: Inverter 0 Turning off scheduled charge
2024-05-02 04:27:41.635158 INFO pred_bat: Inverter 0 Wrote inverter_mode to PauseDischarge successfully
2024-05-02 04:27:41.641274 INFO pred_bat: **Inverter 0 set pause mode to PauseDischarge**
2024-05-02 04:36:08.301358 INFO pred_bat: Disabling battery discharge while the car 0 is charging

2024-05-02 04:27:51.690505 INFO pred_bat: Car charging from battery is off, next slot for car 0 is 05-02 04:25:00 - 05-02 05:00:00

2024-05-02 04:28:06.465867 INFO pred_bat: Disabling battery discharge while the car 0 is charging
2024-05-02 04:28:06.528985 INFO pred_bat: Resetting charging SOC as we are not within the window or charge is disabled and inverter_soc_reset is enabled (now 05-02 04:25:00 target set_soc_minutes 30 charge start time 05-02 04:00:00)
2024-05-02 04:28:06.552062 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target
2024-05-02 04:28:06.575660 INFO pred_bat: Adjust reserve to target charge 100 % (set_reserve_enable is true)
2024-05-02 04:28:06.584686 INFO pred_bat: Inverter 0 Current Reserve is 4.0 % and new target is 100 %
2024-05-02 04:28:17.754618 INFO pred_bat: Set inverter 0 reserve 100 via REST successful on retry 0
2024-05-02 04:28:17.864858 INFO pred_bat: Completed run status Hold charging, Hold for car

springfall2008, 2024-05-03:
I've released a fix

mpartington, 2024-05-04:
possible issue, did not hold, but turned on home battery charging despite home battery at 100%. Had same effect though, and battery did not discharge to car

charge started 15:48 - 16:00 

[appdaemon.log](https://github.com/springfall2008/batpred/files/15210160/appdaemon.log)

[image]

springfall2008, 2024-05-04:
For reference, the battery was charging it started at 97% so I think it worked correctly:

>Charge window will be: 2024-05-04 15:30:00+01:00 - 2024-05-04 16:00:00+01:00 - current soc 97 target 100

mpartington, 2024-05-04:
Agree, user error! I forgot I bumped the charge soon as I plugged in, so
the battery drained a few %

springfall2008, 2024-05-04:
Hopefully fixed, please re-open if not

closed 2024-05-04 by springfall2008