springfall2008/batpred #1030: Error in log file for appdaemon-predbat addin

mauriceatkinson, 2024-04-29:
Trying to install predbat but seeing errors within the appdaemon-predbat addin log file.

I have Solis inverter and a GivEnergy gateway and GivEnergy AllinOne battery.

I am using the Solis template but the predbat_dashboard.yaml never gets created so I am assuming this error is causing me to not be able to install correctly.

`^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 5092, in minute_data_load
    raise ValueError
ValueError

2024-04-29 17:10:00.141900 WARNING pred_bat: ------------------------------------------------------------
2024-04-29 17:15:00.223788 WARNING pred_bat: ------------------------------------------------------------
2024-04-29 17:15:00.225171 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-04-29 17:15:00.226290 WARNING pred_bat: Worker Ags: {'id': '28c7234339764a1baf6da732e5b27e18', 'name': 'pred_bat', 'objectid': '405734e9f49d4fbd97a33f2f050ff76b', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f04a5543d10>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-04-29 17:15:00.226535 WARNING pred_bat: ------------------------------------------------------------
2024-04-29 17:15:00.227784 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 14032, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 14028, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12994, in update_pred
    self.fetch_sensor_data()
  File "/config/apps/predbat.py", line 12123, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 5092, in minute_data_load
    raise ValueError
ValueError

2024-04-29 17:15:00.228083 WARNING pred_bat: ------------------------------------------------------------`

gcoan, 2024-04-29:
Try using the GivEnergy template for your all in one as its your battery inverter that predbat controls, not your Solis solar inverter (which is what I assume you have).

mauriceatkinson, 2024-04-29:
Thanks I tried the givenergy sensor but I still see the error:
`s6-rc: info: service legacy-services: starting
[19:07:50] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-04-29 19:08:16.134256 WARNING pred_bat: ------------------------------------------------------------
2024-04-29 19:08:16.134449 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-04-29 19:08:16.134583 WARNING pred_bat: Worker Ags: {'id': '91128f2b289849b5ad7acc6a36383232', 'name': 'pred_bat', 'objectid': '31adb8d522c04dd2ac64aa57e6b41f9e', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7ff070345b90>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-04-29 19:08:16.134716 WARNING pred_bat: ------------------------------------------------------------
2024-04-29 19:08:16.135582 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 14010, in update_time_loop
    raise e
  File "/config/apps/predbat.py", line 14005, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12994, in update_pred
    self.fetch_sensor_data()
  File "/config/apps/predbat.py", line 12181, in fetch_sensor_data
    raise ValueError
ValueError

2024-04-29 19:08:16.135707 WARNING pred_bat: ------------------------------------------------------------`

closed 2026-08-28 by chalfontchubby