springfall2008/batpred #1031: Simpler & more effective cross charging prevention

gcoan, 2024-04-29:
**Is your feature request related to a problem? 
I have balance inverters enable turned on and balance inverters for cross charging on.
Have tried balance_inverter_seconds set to 60 and now 30, but predbat does not prevent my hybrid inverters from cross charging.

[image]

G SOC dropping, H SOC falling then rising as G cross charges into H. H battery power negative (charging) and following the same shape of activity as G battery power which is positive (discharging). 

[image]

Looking in more detail from 22:00 to 22:10, can see charge and discharge for inverter 0 (G) remains at 2600, Predbat tries to stop the cross charging by reducing both charging and discharging rate for inverter 1 (H):

[image]
[image]

**Describe the solution you'd like**
Predbat prevents cross charging!

I don't understand why the discharge rate for H (inverter 1) is being varied when the problem is that G (inverter 0) is discharging and H is charging.

Propose a much simpler solution to prevent this evening/night-time cross charging when predbat is in Idle/Eco mode is to set the battery charge rates to zero in the period between sunset and sunrise.

Obviously would need to set the charge rate appropriately when predbat initiates grid charging overnight.

**Describe alternatives you've considered**
Shadow automation to keep setting inverter discharge rates based on analysing predbat status and time of day.  Would end up less effective as predbat will keep on resetting the charge rate every time it runs.

**Additional context**
See also #829 for other cross-charging function that appears not to be working (during charging)

gcoan, 2024-05-08:
@springfall2008 I have been trying to see if I can use what predbat already has in its toolbox to resolve the cross charging issue I see every night, e.g.
[image]

I have the cross-charging balance inverters switched on:
[image]

And predbat tries, but it just doesn't prevent cross-charging as shown above.

Seeing if I could implement my own automation to turn off inverter charging when there's no solar generation. I know this will need more refinement for when predbat plans grid charging overnight, but trying this as a first go.

Created an input_number helper, predbat_battery_charge_rate

And in apps.yaml:
```yaml
  inverter_limit_charge:
    - input_number.predbat_battery_charge_rate
    - input_number.predbat_battery_charge_rate
```

But get the following error:
```
2024-05-08 23:52:54.034551 INFO pred_bat: WARN: Out of range index 1 within item inverter_limit_charge value ['input_number.predbat_']
2024-05-08 23:52:54.036057 INFO pred_bat: WARN: Return bad int value None from inverter_limit_charge using default 2600
2024-05-08 23:52:54.070933 INFO pred_bat: WARN: Out of range index 1 within item inverter_battery_rate_min value 0
```

So doesn't work.

I'd prefer it if predbat cross charging prevention did work, but it doesn't, so hence this thinking of just setting charge rate to zero when there is no PV and its not planning to grid charge

springfall2008, 2024-10-06:
Can you explain your proposed solution more, how would setting the charge rate to 0 after dark help?

gcoan, 2024-10-06:
> Can you explain your proposed solution more, how would setting the charge rate to 0 after dark help?

It's not a perfect solution to resolve cross charging but it definitely would resolve the most common cross charging I see.

After dark, no solar charging and both batteries are fairly full.  Evening meal cooked, house loads are higher than a single inverter output so both inverters discharge to meet the load, leaving both batteries partially empty.

What happens then is that inverter 0 ALWAYS cross charges into inverter 1.  0 discharges and 1 charges.

This will happen repeatedly throughout the night when 1 loses a bit of SoC then 0 will discharge so 1 charges and fills back up to full again.

e.g. graph of overnight SoC on the two inverters/batteries, the ringed areas are where they cross charge.  See how H (inverter 1) bobbles around 100% full as well
[image]

It happens a bit during the day but if Predbat could put the battery into charge rate zero / Pause Charge when there is no chance of solar charging then it would stop the overnight cross charging which is something I observe every night and clearly wastes electricity with the dual conversion losses.
I've proven this manually by putting predbat into read only and then pause charging inverter 1.  Usually that's all that is required to stop the cross charging as it occurs from inverter 0->1, very rarely from 1->0.

I have been doing this with an automation through most of the summer but obviously I have to stop my automation and turn predbat read only off at the appropriate time so Predbat can charge the batteries overnight.   Having Predbat automatically stop the batteries cross charging would resolve this for me.

springfall2008, 2024-10-06:
Interesting idea, let me think on that

gcoan, 2025-11-28:
Introduced new switch.predbat_set_freeze_export_during_demand to override demand mode with freeze export