springfall2008/batpred #1032: Error Message in Homessistant Logs since update to Supervosr 2024.04.4

BondBug, 2024-04-30:
These errows started after the Homeassisant supervisor update to 2024.04.4

Appdaemon - Batpred 7.17.6

These are the Homeassistant Log Errors no Errors where reported in the Appdaemon Log File

Logger: homeassistant.components.hassio.handler
Source: components/hassio/handler.py:612
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 23:13:29 (3 occurrences)
Last logged: 23:23:29

Client error on /addons/46f69597_appdaemon-predbat/changelog request 0, message='Attempt to decode JSON with unexpected mimetype: application/octet-stream', url=URL('http://172.30.32.2/addons/46f69597_appdaemon-predbat/changelog')

Logger: homeassistant.components.hassio.data
Source: components/hassio/data.py:497
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 23:13:29 (3 occurrences)
Last logged: 23:23:29

Could not fetch changelog for 46f69597_appdaemon-predbat:

gcoan, 2024-04-30:
I'm seeing similar errors in my HA Core log on 2924.4.4 but for a different add on so maybe it's a HA issue?

```
Logger: homeassistant.components.hassio.handler
Source: components/hassio/handler.py:612
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 17:39:04 (75 occurrences)
Last logged: 23:46:48

Client error on /addons/9cf1ea8f_mqtt_explorer/changelog request 0, message='Attempt to decode JSON with unexpected mimetype: application/octet-stream', url=URL('http://172.30.32.2/addons/9cf1ea8f_mqtt_explorer/changelog')
```

SwiftRR, 2024-05-01:
This is from my HA log from this morning (01/05/2024). (v7.17.6) HA 2024.3.3 on Pi4 (SSD)
See also #1033 and #1028.
I have a GE AC coupled inverter and no Pause settings have appeared

Rob
------------------------------------------------------
 Could not fetch changelog for 46f69597_appdaemon-predbat: 
 11:32:50 – (WARNING) Home Assistant Supervisor - message first occurred at 30 April 2024 at 22:12:50 and shows up 322 times
 Client error on /addons/46f69597_appdaemon-predbat/changelog request 0, message='Attempt to decode JSON with unexpected mimetype: application/octet-stream', url=URL('http://172.30.32.2/addons/46f69597_appdaemon-predbat/changelog') 
 11:32:50 – (ERROR) Home Assistant Supervisor - message first occurred at 30 April 2024 at 22:12:50 and shows up 322 times
 State attributes for predbat.plan_html exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored 
 11:30:11 – (WARNING) Recorder - message first occurred at 24 April 2024 at 19:32:18 and shows up 2783 times

springfall2008, 2024-05-03:
Not sure it's Predbat related it does it on multiple add ons:

https://github.com/home-assistant/supervisor/issues/5053

closed 2025-01-01 by springfall2008