springfall2008/batpred #1033: Predbat stuck in Freeze charge

SwiftRR, 2024-05-01:
**Describe the bug**
Predbat stuck in Freeze charge. Seems to have frozen and crashed.
The only way I could nudge predbat back was to restart appdaemon-predat addon.
I spotted this between 7:00 am and 7:30 am this morning (01-05-2124). Could there ne a link with moving from a 30 day month?

**Expected behavior**
Normal behaviour!
I am thinking on downgrading down step by step to 7.16.7 and removing automatic upgrade. This is the 2nd crash I have experienced with v7.17.X

**Predbat version**
v7.17.6 with automatic update enabled

**Environment details**
GE AC coupled 3.0 daisy chained to two 8.2 kWh batteries 
- Standard HAOS on SSD on Raspberry Pi

****Screenshots**
Appdaemon-predbat log:**
  File "/config/apps/predbat.py", line 13007, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11330, in calculate_plan
    self.optimise_all_windows(metric, metric_keep)
  File "/config/apps/predbat.py", line 10527, in optimise_all_windows
    best_soc, best_metric, best_cost, soc_min, soc_min_minute, best_keep = self.optimise_charge_limit(
                                                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 9609, in optimise_charge_limit
    resultmid[try_soc] = hanres.get()
                         ^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/pool.py", line 774, in get
    raise self._value
  File "/usr/lib/python3.11/multiprocessing/pool.py", line 540, in _handle_tasks
    put(task)
  File "/usr/lib/python3.11/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/reduction.py", line 51, in dumps
    cls(buf, protocol).dump(obj)
_pickle.PicklingError: Can't pickle <function wrapped_run_prediction_charge at 0x7f9fd6e660>: it's not the same object as predbat.wrapped_run_prediction_charge

2024-04-28 21:11:45.117931 WARNING pred_bat: ------------------------------------------------------------
Update complete
Update complete
Update complete
2024-04-30 22:10:00.724276 WARNING pred_bat: ------------------------------------------------------------
2024-04-30 22:10:00.725023 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-04-30 22:10:00.725674 WARNING pred_bat: Worker Ags: {'id': '8ff1ebcd14284ecc87cf242c0235e109', 'name': 'pred_bat', 'objectid': 'ef8d049a24314ca8bd60dfcdd4f4ce64', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7fa1580b90>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-04-30 22:10:00.726132 WARNING pred_bat: ------------------------------------------------------------
2024-04-30 22:10:00.729132 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 14032, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 14028, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12994, in update_pred
    self.fetch_sensor_data()
  File "/config/apps/predbat.py", line 12123, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 5092, in minute_data_load
    raise ValueError
ValueError

2024-04-30 22:10:00.729612 WARNING pred_bat: ------------------------------------------------------------
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service appdaemon: stopping
[07:17:27] INFO: Service AppDaemon exited with code 0 (by signal 0)
s6-rc: info: service appdaemon successfully stopped
s6-rc: info: service init-appdaemon: stopping
s6-rc: info: service init-appdaemon successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: appdaemon-predbat
 Predbat pre-install in AppDaemon
-----------------------------------------------------------
 Add-on version: 1.0.10
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.5  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.3.3
 Home Assistant Supervisor: 2024.04.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[07:17:34] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started

**Log file**
Predbat.log
[predbat-2.log](https://github.com/springfall2008/batpred/files/15174442/predbat-2.log)

Rob

springfall2008, 2024-05-04:
I'm not sure the above crash relates to the logfile which goes on until 7:20am correctly.

Can you explain the actual issue, do you think the freeze charging was wrong?

SwiftRR, 2024-05-04:
Trefor,
This Predbat crash was 3 days ago. I resuscitated predbat by restarting the appdaemon–predbat addon.

I do not think that the freeze charging status was responsible. Many of us have been getting to grips with this status which seems to be necessary in the plan when slot rates are close to 15p (for those of us with 15p outgoing) and with SOC at 100 for most slots.
 
I suggest that we close this issue. It may resurface if I get a repetition of predbat crashing.

Rob

springfall2008, 2024-11-16:
Closing this old issue, feel free to open a new one if still an issue

closed 2024-11-16 by springfall2008