springfall2008/batpred #1037: FreezeCharge used over night instead of charging during Intelligent window to 100%

Sycamore967, 2024-05-03:
**Describe the bug**
I've include a screenshot, because I don't have access to HA at the moment for any other logs, but it looks to me as though BatPred didn't use the Intelligent window to charge the battery to 100% last night but for some reason decided to only partially charge during my intelligent charge window of 23:30 - 05:30.  The result is that the car load was serviced by the battery, with BatPred ultimately falling way short of charging the batteries enough to get through the day.

Solcast has always forecast that there would be hardly any PV today, so any plan would have had to have known PV wasn't going to fill the void.

**Expected behavior**
Charge too 100% during all intelligent charge windows.

**Predbat version**

latest I can't get the number right now

**Environment details**

2x 9.5 batteries
3.6 Hybrid inverter

**Screenshots**
[image]

**Log file**
Can't get them at the moment, but later if rrequired.

nickgee31, 2024-05-03:
This looks very similar to an issue that I raised - https://github.com/springfall2008/batpred/issues/1027

I reverted back to V7.16.16 which fixed it for me.

Sycamore967, 2024-05-03:
Yeah I read you post a couple of days ago.  Very interesting. It’s not the first time it’s done it for me so is there an underlying issue here?

Sycamore967, 2024-05-07:
On the latest version 7.18.1 now and there are still issues.  This is tonight's plan

[image]

You'll see I'm not discharging down to my discharge limit,  even thought there are slots available.  I could be making more money.
It will also be Freezing Charge for 1.5hrs despite that fact that the car will be charging on intelligent during those slots and will drain the house battery.

[image]

I've included my config and log.

[PredBat_Files.zip](https://github.com/springfall2008/batpred/files/15239078/PredBat_Files.zip)

Sycamore967, 2024-05-13:
V7.18.6 - Doesn't solve the problem, I'm afraid

In fact it's worse - No discharge to minimums and FreezeCharge over night, for no reason?

I have Allow Car to Charge from Battery  set to no.

[image]

This is tonight's plan:

[image]

springfall2008, 2024-05-14:
Poteital fix on main for testing

Sycamore967, 2024-05-14:
Brill thanks Trefor - I'll install when I get back tonight and let you know how it goes.

Sycamore967, 2024-05-17:
I've been running v7.19.0 for 3 days now.  The weather has been awful so not much PV, however today is looking good.  So Predbat has coped with managing the batteries and today a lot more PV.  So far, much much better.  I've not seen any weird plan being generated and no FreezeCharge during the Intelligent window.

Sycamore967, 2024-06-03:
I'm not seeing this as an issue anymore, in any of the later Predbat versions