springfall2008/batpred #1040: Predbat charging at expensive rate

jacekowski, 2024-05-04:
**Describe the bug**
For some reason, predbat decided to charge today at relatively expensive rate (20p, well above export rate) on a day with 50kWh solar forecast.

**Expected behavior**

Do nothing.

**Predbat version**

latest 7.17.7

**Screenshots**
[image]

RobinCu, 2024-05-04:
Can you check if there was any significant draw from the grid?  The limit was set to 48%, SOC already at 47%, PV was forecast to exceed load and the cost for the 30-minute period decreased slightly.  It might have just been a blip in the calculation run.

jacekowski, 2024-05-04:
It has charged from the grid (not much, but it did)

jacekowski, 2024-05-05:
I've got it doing another stupid thing, it's preventing battery from discharging now. 
[image]

mpartington, 2024-05-05:
What are your values for these:
[image]
[image]
[image]

I tried to explain why holding charge is sensible in issue #1035, however it may be your inputs above are a little high and are forcing this at higher rates than you would like.

Ultimately if you don't freeze, then you will recharge your battery on lost solar, which is valued at 15p/kwh. You will also have round trip losses if you discharge the battery + whatever value you've used in the battery metric.

jacekowski, 2024-05-05:
0.04 for all loss values and i don't have battery cycle cost entity.

Accounting for losses, it should use battery for anything above 17p, but it is doing freeze charge even at 19p+ rates.

mpartington, 2024-05-05:
Strange, it was doing this for me at the very high 25p+ peak rates, but Trefor put a fix in that seems to have worked. Might be worth uploading a log filewith a few cycles in debug mode next time you catch it

mpartington, 2024-05-05:
Looks like.mine.has also decided to do similar..Doesn't make sense, as can top up on solar for less cost. Slot force to Idle as was actually charging
[image]

mpartington, 2024-05-05:
logs.. 

[appdaemon.log](https://github.com/springfall2008/batpred/files/15214283/appdaemon.log)
[appdaemon1.log](https://github.com/springfall2008/batpred/files/15214286/appdaemon1.log)

At 19:35 , the plan goes nuts and starts to use grid at high rates and also plan on charging at high rates.  moving back a few versions, then reinstalling v7.7.12 restored the plan to something sensible
[image]

and back to normal

[image]

mpartington, 2024-05-05:
With debug for the latest change 21:25 is the relevant run

[appdaemon.log](https://github.com/springfall2008/batpred/files/15214493/appdaemon.log)

springfall2008, 2024-05-06:
The issue seems to relate to the use of metric battery cycle when its non-zero, testing a fix now

springfall2008, 2024-05-06:
Fix on main for testing

mpartington, 2024-05-06:
@springfall2008  Logs for the 0.5p and 2p battery metric.  
The 2p seems to be planning to hold at very high rates around19:00 (>21p)
[appdaemon1.log](https://github.com/springfall2008/batpred/files/15219277/appdaemon1.log)

Plans run around 10:30, so across 2 log files unfortunately

jacekowski, 2024-05-07:
I've changed my settings slightly to limit battery charge rate (to start exporting early and not waste any solar), by making battery_rate_max an entity and using node-red to adjust it. Now that i have it set to 2000, predbat comes up with plan to force charge.

[image]

closed 2024-05-27 by springfall2008