springfall2008/batpred #1048: Predbat did not switch off Predbat car charging slot at correct time

mpartington, 2024-05-05:
**Describe the bug**
Predbat scheduled 3 slots last night 05:00 to 05:30 , 06:00 to 06:30 and 08:00 to 08:30.  However the second slot incorrectly continued until 07:00.

A bit of background to the set up... I use Car Manual SoC as sometimes the car api stops reporting the SoC, so find this a safer approach. As you know Trefor, in this mode it will assume the amount of kWh delivered based on the charge rate.

You can see at 05:30, Predbat stops accumulating the kWh, and the logs confirm it is at target.    However the charge slot was not switched off and the plan also did not show an active slot planned.  It seems like parts of Predbat realised it should not be charging, but the state of 'binary_sensor.predbat_car_charging_slot' was not changed to off as expected

As the difference in cost for this unplanned slot was negligible to the one that was planned later, I let it continue. At 06:00, Predbat finally switched off  

**Expected behavior**
binary_sensor.predbat_car_charging_slot to turn off at the end of a planned slot

I think it may be due to this rounding:
2024-05-05 06:31:09.184482 INFO pred_bat: Car charging Manual SOC current is 32.26 next is 32.260999999999996

It will never meet the target (as it stops accumulating soc_kwh), so keeps adding another 5 min slot

at 7am, these values somehow align
2024-05-05 07:00:24.039204 INFO pred_bat: Car charging Manual SOC current is 32.26 next is 32.26

**Predbat version**

v7.17.10

**Environment details**

- Standard HAOS 

**Screenshots**
.
[image]

[image]

[image]

**Log file**
[appdaemon.log](https://github.com/springfall2008/batpred/files/15211971/appdaemon.log)

springfall2008, 2024-05-05:
Potential fix on main

springfall2008, 2024-05-07:
Now released

mpartington, 2024-05-14:
Still an issue with rounding.  Car should have switched off at 03:00, but slot remained active.  I looked in Predbat log and same issue :

2024-05-14 03:02:54.984694 INFO pred_bat: Car charging Manual SOC current is 48.48 next is **48.485**
[appdaemon1.log](https://github.com/springfall2008/batpred/files/15302661/appdaemon1.log)

Running the beta version on main (v7.18.7)

springfall2008, 2024-05-14:
Further fix on main

closed 2024-05-15 by springfall2008