springfall2008/batpred #1049: Save/Restore settings not working

BondBug, 2024-05-05:
I am not sure is it is my system at fault but when I select Save having made alterations to my settings nothing is saved in the prebat_save location. I have removed Predbat and then rebooted and reinstalled both appdaemon and Predbat and still have the same problem.
It also appears that having tried to save my settings and nothing happens, I am then unable to adjust any settings (input variable and Switches ) as they then keep reverting back to to the previouse setting and the only way to get my settings control back is to stop start Appdaemon.
Not sure how long this has been like this as have not tried to save before.

My System is Raspberry Pi running the latest 
Homeassitant OS core 2025.5.1
Suppervisor 2024.04.4

Appdaemon standalone
Predbat APP HACS Automation

**Predbat version** V 7.17.10

xxxx

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

gcoan, 2024-05-05:
Its broken on mine as well.  I have previous save sessions but the select is detected but predbat doesn't do anything 

[image]

[image]

Looks like Trefor is fixing it already

gcoan, 2024-05-05:
Hmm.  Maybe don't use save/restore until Trefor has fixed it.

After I tried doing a save of my settings, predbat crashed AppDaemon and it didn't do anything for 3 hours.  Only spotted because the predbat plan still said 11:00 at 14:30

BondBug, 2024-05-05:
Have updated to V7.17.12 and my Save/Restore issue seems to be fixed also I do a manual update and that also worked so all appears to be good
Thank you for working and resolving the issue

closed 2024-05-07 by springfall2008