springfall2008/batpred #1054: Async errors when upgrading from v7.17.11

SwiftRR, 2024-05-05:
**Describe the bug**
Async errors showing up in predbat status reports and in predbat.log.
I don't know whether this has been fixed already but best to document.

I then turned OFF automatic upgrade switch and downgraded back to v7.17.10.
I then tried manually upgrading to v7.17.11 and this appears to have been successful.

**Expected behavior**
No errors
**Predbat version**
v7.17.11 when upgrading to v7.17.10

**Environment details**
- Inverter and battery setup
  GE AC coupled inverter daisychained to two GE 8.2 kWh batteries 
- Standard HAOS installer or Docker
  HAOS 2024.03.03 on Raspberry Pi4

**Screenshots**
Status reports when all this happened.
[image]
**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
Predbat.log attached. 
[predbat.log](https://github.com/springfall2008/batpred/files/15212720/predbat.log)
All this happened from 12:00–12:30 05/05/2024

gcoan, 2024-05-05:
Ditto.

You have to do a manual upgrade using HACS to get around this

springfall2008, 2024-05-05:
Sorry, indeed its fixed now but you have to manually fix it as per the other thread.

closed 2024-05-07 by springfall2008