springfall2008/batpred #1056: Trouble updating leading to the realisation Appdaemon is not running but Prdbat seems to be.

Emmoore103, 2024-05-05:
**Describe the bug**
I installed version v7.17.10 (or it might have been v7.17.9 and noticed the next morning Predbat had not operated the battery as expected and the graphs had not updated. I noticed Appdaemon had crashed. I tried to get it going but it kept crashing. After some fiddling I managed to get Predbat up and running but I still notice Appdaemon is showing as not running. So I am confused why this is and worried it might error again.

I have managed to update to v7.17.11

This is the Appdaemon log:
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[12:54:16] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
ERROR Error loading secrets file: /config/secrets.yaml
Traceback (most recent call last):
  File "/usr/bin/appdaemon", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/__main__.py", line 417, in main
    admain.main()
  File "/usr/lib/python3.11/site-packages/appdaemon/__main__.py", line 276, in main
    if "appdaemon" not in config:
       ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable
[12:54:21] INFO: Service AppDaemon exited with code 1 (by signal 0)
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service appdaemon: stopping
s6-rc: info: service appdaemon successfully stopped
s6-rc: info: service init-appdaemon: stopping
s6-rc: info: service init-appdaemon successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.2  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.5.1
 Home Assistant Supervisor: 2024.04.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[12:54:27] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
ERROR Error loading secrets file: /config/secrets.yaml
Traceback (most recent call last):
  File "/usr/bin/appdaemon", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/__main__.py", line 417, in main
    admain.main()
  File "/usr/lib/python3.11/site-packages/appdaemon/__main__.py", line 276, in main
    if "appdaemon" not in config:
       ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable
[12:54:32] INFO: Service AppDaemon exited with code 1 (by signal 0)
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service appdaemon: stopping
s6-rc: info: service appdaemon successfully stopped
s6-rc: info: service init-appdaemon: stopping
s6-rc: info: service init-appdaemon successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
  

**Expected behavior**
For Appdaemon  to be working as well as Predbat.

**Predbat version**
Now on v7.17.11

**Environment details**

- Givenergy Gen 2 inverter and 2x9.5KWh batteries
- Not sure how I installed the software. It was only just withing my technical abilities as I only started using HA for this purpose.

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2024-05-05:
Remove the reference to secrets.yaml in appdeamon.yaml and it should be fixed

Emmoore103, 2024-05-05:
Thank you.

Eric

Emmoore103, 2024-05-05:
It seems to have worked. Thank you very much. Have a good evening.

Kind regards
Eric

closed 2024-05-06 by springfall2008