springfall2008/batpred #1060: Sunsynk Documentation

Meatballs1, 2024-05-05:
I have the following modifications to predbat.py:

* Set SoC target% to an int
* Set power to current to round 0
*  "support_charge_freeze": True,
*   "support_discharge_freeze": True,
*   "output_charge_control": "current",

I currently have the following configuration:

```yaml
  load_today:
    - sensor.household_load_energy_today # Custom sensor as AC coupled
  import_today:
    - sensor.sunsynk_daily_grid_import
  export_today:
    - sensor.sunsynk_daily_grid_export
  pv_today:
    - sensor.solaredge_pv_generation_today # Solaredge PV inverter

  num_inverters: 1
  inverter_type: SK

  charge_start_service:
    service: select.select_option
    entity_id: "select.sunsynk_work_mode"
    option: "Limited to Home"
  charge_stop_service:
    service: select.select_option
    entity_id: "select.sunsynk_work_mode"
    option: "Limited to Home"
  discharge_start_service:
    service: select.select_option
    entity_id: "select.sunsynk_work_mode"
    option: "Selling First"	

  timed_charge_current:
    - number.sunsynk_battery_max_charge_current
  timed_discharge_current:
    - number.sunsynk_battery_max_discharge_current
  battery_voltage:
    - sensor.sunsynk_battery_voltage
  charge_limit:
    - number.sunsynk_set_soc_timezone1 # Needs an automation to set across all 6 timezones

  # Customer sensors to get the BMS charge/discharge limit, based on inverter size (5kW, BMS reported limit, and inverter set limit)
  # Does this need to be an input value?
  charge_rate:
     - sensor.max_battery_charge_rate # {{ [5000,[states('number.sunsynk_battery_max_charge_current')|int,states('sensor.sunsynk_battery_charge_bms_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
  discharge_rate:
     - sensor.max_battery_discharge_rate # {{ [5000,[states('number.sunsynk_battery_max_discharge_current')|int, states('sensor.sunsynk_battery_discharge_bms_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}

 battery_rate_max: 5000
 inverter_limit:
   - 11000

  battery_power:
    - sensor.sunsynk_battery_output_power
  pv_power:
    - sensor.solaredge_ac_power
  load_power:
    - sensor.household_load # {{ states('sensor.sunsynk_grid_power')|float + states('sensor.solaredge_ac_power')|float + states('sensor.sunsynk_inverter_output_power')|float }}
  soc_kw:
    - sensor.sunsynk_battery_energy # {{ states('input_number.sunsynk_battery_size')|float * states('sensor.sunsynk_battery_soc')|float / 100 |float(0) }}
  soc_max:
    - input_number.sunsynk_battery_size # Set to 20.5kWh 
  battery_min_soc:
    - sensor.sunsynk_battery_min_soc # Could just be hardcoded? My shutdown is set to 20% {{ states('sensor.sunsynk_battery_capacity_shutdown')|int * states('input_number.sunsynk_battery_size')|float / 100 }}

```

Automation

```yaml
alias: PredBat - Copy Charge Limit
description: ""
trigger:
  - platform: state
    entity_id:
      - number.sunsynk_set_soc_timezone1
    to: null
condition: []
action:
  - service: number.set_value
    data_template:
      entity_id:
        - number.sunsynk_set_soc_timezone2
        - number.sunsynk_set_soc_timezone3
        - number.sunsynk_set_soc_timezone4
        - number.sunsynk_set_soc_timezone5
        - number.sunsynk_set_soc_timezone6
      value: "{{ states('number.sunsynk_set_soc_timezone1')|int(20) }}"
mode: single
```

I dont have a charge curve, currently get the following:

```
2024-05-06 00:43:29.453000 INFO pred_bat: Find discharge curve with sensors sensor.sunsynk_battery_energy and sensor.predbat_SK_0_discharge_rate and predbat.status and sensor.sunsynk_battery_output_power
--
2024-05-06 00:43:29.441535 INFO pred_bat: Note: Can not find battery charge curve (missing history), one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
2024-05-06 00:43:15.144601 INFO pred_bat: Find charge curve with sensors sensor.sunsynk_battery_energy and sensor.predbat_SK_0_charge_rate and predbat.status and sensor.sunsynk_battery_output_power
```

I think I have 7 days history in all of those, but maybe battery hasn't been topped up to 100% fully. Not sure if needed.

Meatballs1, 2024-05-06:
Not sure if I need to set `switch.predbat_inverter_soc_reset`

bensebborn, 2024-05-06:
I use automations for the controls.

Force charge:

  * Change Work Mode to Zero export to CT
  * Turn on Charge Point 6 Grid
  * Turn on Use Timer
  
  Force discharge:

  * Change Work Mode to Selling First
  * Turn off Charge Point 6 Grid
  * Turn on Use Timer
  
  Cancel charge/discharge:
  
    * Change Work Mode to Zero export to CT
    * Turn off Charge Point 6 Grid
    * Turn on Use Timer

springfall2008, 2024-05-14:
Could someone have a go at writing the documentation paragraph to tell new users how to set this up?

Are any predbat.py changes pending or are they now all in the mainline?

Meatballs1, 2024-05-14:
I've not been up and running for a while with the flurry of updates recently, then hit:

[image]

[image]

Not had time to look into it

springfall2008, 2024-06-01:
> I've not been up and running for a while with the flurry of updates recently, then hit:
> 
> ![Screenshot_20240514-182734](https://private-user-images.githubusercontent.com/1854557/330535631-5c9c9614- time to look into it

I suspect this is fixed now?

Meatballs1, 2024-06-01:
Yes, need to update for solcast now 😅

jimleach, 2024-06-24:
> I use automations for the controls.
>
> Force charge:
>
> * Change Work Mode to Zero export to CT
> * Turn on Charge Point 6 Grid
> * Turn on Use Timer
>
> Force discharge:
>
> * Change Work Mode to Selling First
> * [...]
Well done on getting the Sunsynk to work - I managed to get the solar/battery data into PredBat but couldn't work out how to program the charge slots on the inverter.

Any chance you could post the automations up please?

gcoan, 2024-06-24:
I'm happy to update the documentation with the sensors & any recommended automations

springfall2008, 2024-07-14:
Please review the attached pull request with documentation and template updates:

https://github.com/springfall2008/batpred/pull/1305

gcoan, 2024-07-14:
@springfall2008 as per #1304 should make clear in the template apps.yaml that 'battery_min_soc' needs to be set to a value in % not a value in kWh

tmh88, 2024-07-15:
> I have the following modifications to predbat.py:
> 
> * Set SoC target% to an int
> * Set power to current to round 0
> *  "support_charge_freeze": True,
> *   "support_discharge_freeze": True,
> *   "output_charge_control": "current",
> 
> 
> I currently have the following configuration:
> 
> ```yaml
>   load_today:
>     - sensor.household_load_energy_today # Custom sensor as AC coupled
>   import_today:
>     - sensor.sunsynk_daily_grid_import
>   export_today:
>     - sensor.sunsynk_daily_grid_export
>   pv_today:
>     - sensor.solaredge_pv_generation_today # Solaredge PV inverter
> 
>   num_inverters: 1
>   inverter_type: SK
> 
> 
>   charge_start_service:
>     service: select.select_option
>     entity_id: "select.sunsynk_work_mode"
>     option: "Limited to Home"
>   charge_stop_service:
>     service: select.select_option
>     entity_id: "select.sunsynk_work_mode"
>     option: "Limited to Home"
>   discharge_start_service:
>     service: select.select_option
>     entity_id: "select.sunsynk_work_mode"
>     option: "Selling First"	
> 
>   timed_charge_current:
>     - number.sunsynk_battery_max_charge_current
>   timed_discharge_current:
>     - number.sunsynk_battery_max_discharge_current
>   battery_voltage:
>     - sensor.sunsynk_battery_voltage
>   charge_limit:
>     - number.sunsynk_set_soc_timezone1 # Needs an automation to set across all 6 timezones
> 
>   # Customer sensors to get the BMS charge/discharge limit, based on inverter size (5kW, BMS reported limit, and inverter set limit)
>   # Does this need to be an input value?
>   charge_rate:
>      - sensor.max_battery_charge_rate # {{ [5000,[states('number.sunsynk_battery_max_charge_current')|int,states('sensor.sunsynk_battery_charge_bms_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
>   discharge_rate:
>      - sensor.max_battery_discharge_rate # {{ [5000,[states('number.sunsynk_battery_max_discharge_current')|int, states('sensor.sunsynk_battery_discharge_bms_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
> 
>  battery_rate_max: 5000
>  inverter_limit:
>    - 11000
> 
> 
>   battery_power:
>     - sensor.sunsynk_battery_output_power
>   pv_power:
>     - sensor.solaredge_ac_power
>   load_power:
>     - sensor.household_load # {{ states('sensor.sunsynk_grid_power')|float + states('sensor.solaredge_ac_power')|float + states('sensor.sunsynk_inverter_output_power')|float }}
>   soc_kw:
>     - sensor.sunsynk_battery_energy # {{ states('input_number.sunsynk_battery_size')|float * states('sensor.sunsynk_battery_soc')|float / 100 |float(0) }}
>   soc_max:
>     - input_number.sunsynk_battery_size # Set to 20.5kWh 
>   battery_min_soc:
>     - sensor.sunsynk_battery_min_soc # Could just be hardcoded? My shutdown is set to 20% {{ states('sensor.sunsynk_battery_capacity_shutdown')|int * states('input_number.sunsynk_battery_size')|float / 100 }}
> 
> ```
> 
> Automation
> 
> ```yaml
> alias: PredBat - Copy Charge Limit
> description: ""
> trigger:
>   - platform: state
>     entity_id:
>       - number.sunsynk_set_soc_timezone1
>     to: null
> condition: []
> action:
>   - service: number.set_value
>     data_template:
>       entity_id:
>         - number.sunsynk_set_soc_timezone2
>         - number.sunsynk_set_soc_timezone3
>         - number.sunsynk_set_soc_timezone4
>         - number.sunsynk_set_soc_timezone5
>         - number.sunsynk_set_soc_timezone6
>       value: "{{ states('number.sunsynk_set_soc_timezone1')|int(20) }}"
> mode: single
> ```
> 
> I dont have a charge curve, currently get the following:
> 
> ```
> 2024-05-06 00:43:29.453000 INFO pred_bat: Find discharge curve with sensors sensor.sunsynk_battery_energy and sensor.predbat_SK_0_discharge_rate and predbat.status and sensor.sunsynk_battery_output_power
> --
> 2024-05-06 00:43:29.441535 INFO pred_bat: Note: Can not find battery charge curve (missing history), one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
> 2024-05-06 00:43:15.144601 INFO pred_bat: Find charge curve with sensors sensor.sunsynk_battery_energy and sensor.predbat_SK_0_charge_rate and predbat.status and sensor.sunsynk_battery_output_power
> ```
> 
> I think I have 7 days history in all of those, but maybe battery hasn't been topped up to 100% fully. Not sure if needed. Or maybe because my charge_rate is dynamic?

Hi, I'm trying to get predbat working with my sunsynk inverter. Currently I have it set to test entities. And I have followed your config.

Im planning on using the following to control the inverter. Battery SOC setpoint (copied to all time slots), battery max charge current SP, battery max discharge current SP, inverter work mode. Leaving grid charge ticked on all time slots.

How's your inverter continuing to charge the batteries via solar?

Currently when the grid charge finishes it drops the soc sp to 10% and the max charge current to 0A and sets the discharge current to 120A.
This will stop it charging from the grid assuming we are above the soc sp, however it stops the inverter charging the battery from solar.

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing as done: there is now a SunSynk section in the [inverter setup docs](https://springfall2008.github.io/batpred/inverter-setup/) and a `templates/sunsynk.yaml` template (#1305), which also marks `battery_min_soc` as a percentage.

closed 2026-10-19 by chalfontchubby