springfall2008/batpred #1069: IOG on 7.18.2 - still doesn't charge at 11.30pm

PianSom, 2024-05-10:
For the last few versions IOG plans have been discharging in the 11.30-midnight slots, when import costs are only 7.5p. 

Expected behaviour is that charging starts at 11.30pm. This happened until (I think) around the start of April.

I had hoped the IOG improvements in this version would address this issue, but sadly not.

springfall2008, 2024-05-14:
Can you give more details please, it's fine to discharge during these slots provided the time isn't needed for charging but 'calculate_discharge_oncharge' can be turned off to avoid this behaviour?

PianSom, 2024-05-14:
Yes, certainly.

Today is a good example. I have `calculate_discharge_on_charge` set to ON. I want this, because in my overnight IOG cheap rate I have enough time to change, partially discharge and charge again - giving me maximised income.  Moreover, I want the battery to discharge prior to the cheap slot starting.

Here is my current plan; this is pretty good, but I would prefer the evening discharge to be done by 11.30pm, so I can start my charging then, as soon as the 7.5p rate starts - 

[image]

If I turn the switch to OFF I get two adverse effects. Firstly, the discharge session is moved to earlier in the evening (losing any flexibility for late evening increased load). And secondly I, of course, lose my overnight discharge slot.

[image]

PianSom, 2024-05-21:
This behaviour continues in 7.19.7

PianSom, 2024-05-27:
Behaviour continues on 7.20.2

springfall2008, 2024-11-16:
Closing this old issue, feel free to open a new one if still an issue

closed 2024-11-16 by springfall2008