springfall2008/batpred #1075: Fatal Errors after last night's update

btxluke, 2024-05-11:
**Describe the bug**
s6-rc: info: service legacy-cont-init: stopping s6-rc: info: service legacy-cont-init successfully stopped s6-rc: info: service fix-attrs: stopping s6-rc: info: service base-addon-log-level: stopping s6-rc: info: service fix-attrs successfully stopped s6-rc: info: service base-addon-log-level successfully stopped s6-rc: info: service base-addon-banner: stopping s6-rc: info: service base-addon-banner successfully stopped s6-rc: info: service s6rc-oneshot-runner: stopping s6-rc: info: service s6rc-oneshot-runner successfully stopped s6-rc: info: service s6rc-oneshot-runner: starting s6-rc: info: service s6rc-oneshot-runner successfully started s6-rc: info: service base-addon-banner: starting ----------------------------------------------------------- Add-on: appdaemon-predbat Predbat pre-install in AppDaemon ----------------------------------------------------------- Add-on version: 1.0.10 You are running the latest version of this add-on. System: Home Assistant OS 12.3 (aarch64 / raspberrypi4-64) Home Assistant Core: 2024.5.3 Home Assistant Supervisor: 2024.05.1 ----------------------------------------------------------- Please, share the above information when looking for help or support in, e.g., GitHub, forums or the Discord chat. ----------------------------------------------------------- s6-rc: info: service base-addon-banner successfully started s6-rc: info: service fix-attrs: starting s6-rc: info: service base-addon-log-level: starting s6-rc: info: service fix-attrs successfully started s6-rc: info: service base-addon-log-level successfully started s6-rc: info: service legacy-cont-init: starting s6-rc: info: service legacy-cont-init successfully started s6-rc: info: service init-appdaemon: starting s6-rc: info: service init-appdaemon successfully started s6-rc: info: service appdaemon: starting s6-rc: info: service appdaemon successfully started s6-rc: info: service legacy-services: starting [05:49:49] INFO: Starting AppDaemon... s6-rc: info: service legacy-services successfully started 2024-05-11 05:50:33.841988 WARNING pred_bat: ------------------------------------------------------------ 2024-05-11 05:50:33.842629 WARNING pred_bat: Unexpected error in worker for App pred_bat: 2024-05-11 05:50:33.843065 WARNING pred_bat: Worker Ags: {'id': 'dcdd68f2be7c4f8e8e22c3dc2e3bcd6f', 'name': 'pred_bat', 'objectid': '50b7b018088e4366b03e0e0b2e67ba2b', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f8e7c8390>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}} 2024-05-11 05:50:33.843544 WARNING pred_bat: ------------------------------------------------------------ 2024-05-11 05:50:33.848306 WARNING pred_bat: Traceback (most recent call last): File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"])) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 14430, in run_time_loop raise e File "/config/apps/predbat.py", line 14426, in run_time_loop self.update_pred(scheduled=True) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 13364, in update_pred self.fetch_inverter_data() File "/config/apps/predbat.py", line 12898, in fetch_inverter_data curve = inverter.find_charge_curve(discharge=True) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 2529, in find_charge_curve battery_power_data = self.base.get_history_wrapper(entity_id=battery_power_sensor, days=self.base.max_days_previous) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 5029, in get_history_wrapper raise ValueError ValueError 2024-05-11 05:50:33.848988 WARNING pred_bat: ------------------------------------------------------------ 2024-05-11 05:55:19.619784 WARNING pred_bat: ------------------------------------------------------------ 2024-05-11 05:55:19.620382 WARNING pred_bat: Unexpected error in worker for App pred_bat: 2024-05-11 05:55:19.620980 WARNING pred_bat: Worker Ags: {'id': 'dcdd68f2be7c4f8e8e22c3dc2e3bcd6f', 'name': 'pred_bat', 'objectid': '50b7b018088e4366b03e0e0b2e67ba2b', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f8e7c8390>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}} 2024-05-11 05:55:19.621500 WARNING pred_bat: ------------------------------------------------------------ 2024-05-11 05:55:19.624572 WARNING pred_bat: Traceback (most recent call last): File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"])) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 14430, in run_time_loop raise e File "/config/apps/predbat.py", line 14426, in run_time_loop self.update_pred(scheduled=True) File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 13364, in update_pred self.fetch_inverter_data() File "/config/apps/predbat.py", line 12898, in fetch_inverter_data curve = inverter.find_charge_curve(discharge=True) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 2529, in find_charge_curve battery_power_data = self.base.get_history_wrapper(entity_id=battery_power_sensor, days=self.base.max_days_previous) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/config/apps/predbat.py", line 5029, in get_history_wrapper raise ValueError ValueError 2024-05-11 05:55:19.626390 WARNING pred_bat: ------------------------------------------------------------

**Environment details**

- Givenergy 
- Standard HAOS installer 
)

corautem, 2024-05-11:
Same here, went back to v.7.18.2, disabled auto update and it's working again.

pelowj, 2024-05-11:
`2024-05-11 09:32:31.137043 INFO AppDaemon: Reloading Module: /config/apps/predbat.py
2024-05-11 09:32:31.405459 INFO AppDaemon: Loading app pred_bat using class PredBat from module predbat
2024-05-11 09:32:31.407800 INFO AppDaemon: Calling initialize() for pred_bat
2024-05-11 09:32:31.409796 INFO pred_bat: Predbat: Startup predbat
2024-05-11 09:32:31.411534 INFO pred_bat: Sanity check:
2024-05-11 09:32:31.413025 INFO pred_bat: Sanity scan files in '/config' : ['dashboards', 'predbat.log.1', 'apps', 'predbat_dashboard.yaml', 'predbat.log.2', 'namespaces', 'compiled', 'www', 'appdaemon.yaml', 'predbat.log', 'predbat.log.3']
2024-05-11 09:32:31.417681 INFO pred_bat: Sanity: Got app_dir /config/apps
2024-05-11 09:32:31.419334 INFO pred_bat: Sanity: Scanning app_dirs: ['/config/apps']
2024-05-11 09:32:31.421025 INFO pred_bat: Sanity: Got predbat.py in location /config/apps/predbat.py
2024-05-11 09:32:31.422529 INFO pred_bat: Sanity: Got apps.yaml in location /config/apps/apps.yaml
2024-05-11 09:32:31.495917 INFO pred_bat: Sanity: /config/apps/apps.yaml is a valid pred_bat configuration
2024-05-11 09:32:31.497408 INFO pred_bat: Sanity: Confirmed correct version v7.18.3 is in predbat.py
2024-05-11 09:32:31.503227 INFO pred_bat: Sanity check has passed
2024-05-11 09:32:31.528832 INFO pred_bat: Regular expression argument geserial matched ^sensor.givtcp_(.+)_soc_kwh$ with ed2152r267
2024-05-11 09:32:31.530465 INFO pred_bat: Regular expression argument pv_forecast_today matched ^(sensor.(solcast_|)(pv_forecast_|)forecast_today)$ with sensor.solcast_pv_forecast_forecast_today
2024-05-11 09:32:31.531937 INFO pred_bat: Regular expression argument pv_forecast_tomorrow matched ^(sensor.(solcast_|)(pv_forecast_|)forecast_tomorrow)$ with sensor.solcast_pv_forecast_forecast_tomorrow
2024-05-11 09:32:31.533397 INFO pred_bat: Regular expression argument pv_forecast_d3 matched ^(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3))$ with sensor.solcast_pv_forecast_forecast_day_3
2024-05-11 09:32:31.535077 INFO pred_bat: Regular expression argument pv_forecast_d4 matched ^(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4))$ with sensor.solcast_pv_forecast_forecast_day_4
2024-05-11 09:32:31.536502 INFO pred_bat: Regular expression argument car_charging_energy matched ^(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy)$ with sensor.myenergi_zappi_10304626_charge_added_session
2024-05-11 09:32:31.537995 INFO pred_bat: Regular expression argument car_charging_planned matched ^(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status)$ with sensor.myenergi_zappi_10304626_plug_status
2024-05-11 09:32:31.539814 INFO pred_bat: Regular expression argument car_charging_limit Matched ^number.charge_limit$ with number.charge_limit
2024-05-11 09:32:31.541603 INFO pred_bat: Regular expression argument car_charging_soc Matched ^sensor.battery$ with sensor.battery
2024-05-11 09:32:31.789556 INFO pred_bat: Updating HA config expert_mode to True
2024-05-11 09:32:31.790693 WARNING pred_bat: pred_bat: Entity switch.predbat_expert_mode not found in namespace default
2024-05-11 09:32:31.791620 INFO AppDaemon: pred_bat: Entity switch.predbat_expert_mode created in namespace: default
2024-05-11 09:32:32.179807 INFO pred_bat: Updating HA config active to True
2024-05-11 09:32:32.181082 WARNING pred_bat: pred_bat: Entity switch.predbat_active not found in namespace default
2024-05-11 09:32:32.181941 INFO AppDaemon: pred_bat: Entity switch.predbat_active created in namespace: default
2024-05-11 09:32:32.272193 INFO pred_bat: Updating HA config pv_metric10_weight to 0.15
2024-05-11 09:32:32.273255 WARNING pred_bat: pred_bat: Entity input_number.predbat_pv_metric10_weight not found in namespace default
2024-05-11 09:32:32.274046 INFO AppDaemon: pred_bat: Entity input_number.predbat_pv_metric10_weight created in namespace: default
2024-05-11 09:32:32.312737 INFO pred_bat: Updating HA config pv_scaling to 1.0
2024-05-11 09:32:32.313674 WARNING pred_bat: pred_bat: Entity input_number.predbat_pv_scaling not found in namespace default
2024-05-11 09:32:32.314323 INFO AppDaemon: pred_bat: Entity input_number.predbat_pv_scaling created in namespace: default
2024-05-11 09:32:32.369755 INFO pred_bat: Updating HA config load_scaling to 1.0
2024-05-11 09:32:32.370713 WARNING pred_bat: pred_bat: Entity input_number.predbat_load_scaling not found in namespace default
2024-05-11 09:32:32.371396 INFO AppDaemon: pred_bat: Entity input_number.predbat_load_scaling created in namespace: default
2024-05-11 09:32:32.403556 INFO pred_bat: Updating HA config load_scaling10 to 1.1
2024-05-11 09:32:32.404488 WARNING pred_bat: pred_bat: Entity input_number.predbat_load_scaling10 not found in namespace default
2024-05-11 09:32:32.405150 INFO AppDaemon: pred_bat: Entity input_number.predbat_load_scaling10 created in namespace: default
2024-05-11 09:32:32.443397 INFO pred_bat: Updating HA config load_scaling_saving to 1.0
2024-05-11 09:32:32.444332 WARNING pred_bat: pred_bat: Entity input_number.predbat_load_scaling_saving not found in namespace default
2024-05-11 09:32:32.444993 INFO AppDaemon: pred_bat: Entity input_number.predbat_load_scaling_saving created in namespace: default
2024-05-11 09:32:32.483818 INFO pred_bat: Updating HA config battery_rate_max_scaling to 1.0
[... cut ...]
2024-05-11 09:32:46.255160 CRITICAL AppDaemon: Pinned apps were: ['pred_bat']
2024-05-11 09:32:46.255398 CRITICAL AppDaemon: Thread will be restarted
2024-05-11 09:32:46.255609 INFO AppDaemon: Adding thread 0
2024-05-11 09:32:47.260680 CRITICAL AppDaemon: Thread thread-0 has died
2024-05-11 09:32:47.261011 CRITICAL AppDaemon: Pinned apps were: ['pred_bat']
2024-05-11 09:32:47.261232 CRITICAL AppDaemon: Thread will be restarted
2024-05-11 09:32:47.261486 INFO AppDaemon: Adding thread 0`

The errors I'm getting after last night's update.

Dave-Parry, 2024-05-11:
I got this after updating.

[image]

corautem, 2024-05-11:
> I got this after updating.
> 
> [image]
> 
> Not sure how to get logs like the ones above.
> 

You can get them from AppDaemon logs

gcoan, 2024-05-11:
> I got this after updating.
> 
> 
> 
> [image]
> 
> 
> 
> Not sure how to get logs like the ones above.
> 
> 

Details of where to find your log file is in the output data part of the documentation https://springfall2008.github.io/batpred/output-data/#predbat-logfile

Sycamore967, 2024-05-11:
Same here - PredBat is dead - I'm tryhing to get back to a previous version but it looks like I'm going to have to re-install.  I've attached my log
[appdaemon.log.zip](https://github.com/springfall2008/batpred/files/15283010/appdaemon.log.zip)

Sycamore967, 2024-05-11:
This is what PredBat looks like right now - how can I get back to a working version please? Current on v7.18.3

[image]

Sycamore967, 2024-05-11:
I'm now on v7.18.4 still hasn't helped I'm afraid.  Although I do now get a status.

[image]

[appdaemon (18.4).zip](https://github.com/springfall2008/batpred/files/15283171/appdaemon.18.4.zip)

Dave-Parry, 2024-05-11:
home-assistant_2024-05-11T07-05-16.434Z.log

Dave-Parry, 2024-05-11:
Ok got the logs

2024-05-11 08:04:21.376 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration config_editor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-11 08:04:21.378 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration octopus_energy which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-11 08:04:21.380 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solcast_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-11 08:04:21.381 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-11 08:04:21.382 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration octopus_intelligent which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-11 08:04:21.383 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration octopusagile which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-11 08:04:21.384 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration localtuya which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-11 08:04:21.636 WARNING (MainThread) [homeassistant.helpers.translation] Failed to load integration for translation: Integration 'resources' not found.
2024-05-11 08:04:24.685 ERROR (MainThread) [homeassistant.setup] Setup failed for 'resources': Integration not found.
2024-05-11 08:04:25.192 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] CURRENT_HVAC_HEAT was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use HVACAction.HEATING instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.202 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] CURRENT_HVAC_IDLE was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use HVACAction.IDLE instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.213 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] HVAC_MODE_AUTO was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use HVACMode.AUTO instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.221 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] HVAC_MODE_HEAT was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use HVACMode.HEAT instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.228 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] HVAC_MODE_OFF was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use HVACMode.OFF instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.233 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] SUPPORT_PRESET_MODE was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use ClimateEntityFeature.PRESET_MODE instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.239 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] SUPPORT_TARGET_TEMPERATURE was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use ClimateEntityFeature.TARGET_TEMPERATURE instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.244 WARNING (ImportExecutor_0) [homeassistant.components.climate.const] SUPPORT_TARGET_TEMPERATURE_RANGE was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use ClimateEntityFeature.TARGET_TEMPERATURE_RANGE instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.250 WARNING (ImportExecutor_0) [homeassistant.const] TEMP_CELSIUS was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.CELSIUS instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.256 WARNING (ImportExecutor_0) [homeassistant.const] TEMP_FAHRENHEIT was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.FAHRENHEIT instead, please report it to the author of the 'localtuya' custom integration
2024-05-11 08:04:25.267 WARNING (ImportExecutor_0) [homeassistant.components.cover] SUPPORT_CLOSE was used from localtuya, this is a deprecated constant which will be removed in HA Core 2025.1. Use Co [... cut ...]
2024-05-11 08:04:41.542 WARNING (ImportExecutor_0) [homeassistant.helpers.typing] EventType was used from octopus_energy, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.Event instead, please report it to the author of the 'octopus_energy' custom integration
2024-05-11 08:04:41.554 WARNING (ImportExecutor_0) [homeassistant.helpers.typing] EventType was used from octopus_energy, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.Event instead, please report it to the author of the 'octopus_energy' custom integration

btxluke, 2024-05-11:
@all A couple of Add-on Restarts:

Settings > Addons > Appdaemon-predbat > Restart 

Have cleared the errors for me,  possibly a caching issue?

corautem, 2024-05-11:
Jumped from v. 7.18.2 to 7.18.4, restarted the AppDaemon and it's working fine now. Thank you, very much, Trefor for you time and effort. I can't afford much but I made a small payment for all your work. Thank you

alomstar, 2024-05-11:
Hi, Are you able to share how you got it to work? I've done the update and restarted but same issue.

corautem, 2024-05-11:
> Hi, Are you able to share how you got it to work? I've done the update and restarted but same issue. 

 Me personally I didn't do anything specific, just update directly from predbat and restarted the AppDaemon even though from my past experience it is not necessary if you update from the Predbat or auto update but only if you update from HACS it is necessary to restart the AppDaemon. I restarted just to make sure everything is being restarted. 
 I need to mention I have installed on my machine the initial  Predbat, the one you install Predbat after the AppDaemon, but I don't think it should matter. 
 Initially it displayed an error but it cleared after couple of minutes, probably it takes a while until everything it gets restarted.

Sycamore967, 2024-05-11:
Evening Treffor.  Thanks for all your efforts on this one.  I've got everything back by installing 7.18.6

springfall2008, 2024-05-14:
If you get time to try this all should be fixed on 'main' but feel free to wait for the release also

pelowj, 2024-05-14:
I decided to uninstall and reinstall. Sometimes a clean slate is the best remedy in Home Assistant!

Dave-Parry, 2024-05-14:
v7.18.6 fixed it for me. Thanks Trefor.

closed 2024-05-15 by springfall2008